The-OpenROAD-Project/OpenROAD #1947: Stacktrace in GRT for "Extra Run for hard benchmark"

proppy, 2022-06-09:
```
[INFO GRT-0103] Extra Run for hard benchmark.
Stack trace:
 0# 0x0000000000FD1B49 in openroad
 1# 0x00007F90ECD45400 in /lib64/libc.so.6
 2# grt::FastRouteCore::mazeRouteMSMD(int, int, float, int, int, bool, int, float, int, int, int) in openroad
 3# grt::FastRouteCore::run() in openroad
 4# grt::GlobalRouter::findRouting(std::vector<grt::Net*, std::allocator<grt::Net*> >&, int, int) in openroad
 5# grt::GlobalRouter::globalRoute() in openroad
 6# 0x00000000017ADF1F in openroad
 7# 0x00007F90EFCD8EB2 in /lib64/libtcl8.5.so
 8# 0x00007F90EFD1D36C in /lib64/libtcl8.5.so
 9# TclObjInterpProcCore in /lib64/libtcl8.5.so
10# 0x00007F90EFCD8EB2 in /lib64/libtcl8.5.so
11# 0x00007F90EFD1D36C in /lib64/libtcl8.5.so
12# 0x00007F90EFD25647 in /lib64/libtcl8.5.so
13# TclEvalObjEx in /lib64/libtcl8.5.so
14# 0x00007F90EFD6027F in /lib64/libtcl8.5.so
15# 0x00007F90EFCD8EB2 in /lib64/libtcl8.5.so
16# 0x00007F90EFD1D36C in /lib64/libtcl8.5.so
17# 0x00007F90EFD25647 in /lib64/libtcl8.5.so
18# TclEvalObjEx in /lib64/libtcl8.5.so
19# 0x00007F90EFCE01D0 in /lib64/libtcl8.5.so
20# 0x00007F90EFCD8EB2 in /lib64/libtcl8.5.so
21# 0x00007F90EFD1D36C in /lib64/libtcl8.5.so
22# 0x00007F90EFD25647 in /lib64/libtcl8.5.so
23# TclEvalObjEx in /lib64/libtcl8.5.so
24# 0x00007F90EFCE3F00 in /lib64/libtcl8.5.so
25# 0x00007F90EFCD8EB2 in /lib64/libtcl8.5.so
26# 0x00007F90EFD1D36C in /lib64/libtcl8.5.so
27# TclObjInterpProcCore in /lib64/libtcl8.5.so
28# 0x00007F90EFCD8EB2 in /lib64/libtcl8.5.so
29# 0x00007F90EFD1D36C in /lib64/libtcl8.5.so
30# TclObjInterpProcCore in /lib64/libtcl8.5.so
31# 0x00007F90EFCD8EB2 in /lib64/libtcl8.5.so
32# 0x00007F90EFCD9F1E in /lib64/libtcl8.5.so
33# Tcl_EvalEx in /lib64/libtcl8.5.so
34# Tcl_Eval in /lib64/libtcl8.5.so
35# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
36# ord::tclAppInit(Tcl_Interp*) in openroad
37# Tcl_Main in /lib64/libtcl8.5.so
38# main in openroad
39# __libc_start_main in /lib64/libc.so.6
40# 0x0000000000FCE417 in openroad
```
[Uploading openroad_issue_reproducible.zip…]()

vvbandeira, 2022-06-09:
@luis201420 can you take a look at this?

luis201420, 2022-06-10:
Yes, I can. 
I ran this but didn't get the same error, I got this output after running global route.
```
[INFO GRT-0197] Via related to pin nodes: 32206
[INFO GRT-0198] Via related Steiner nodes: 7037
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 107953
[INFO GRT-0112] Final usage 3D: 888974
[ERROR GRT-0118] Routing congestion too high.
Error: groute.tcl, 69 GRT-0118
```

vvbandeira, 2022-06-10:
Thanks @luis201420.

@proppy which version of `openroad` did you use?

Also, @luis201420 can you check to see if this is another instance of the problem with `nangate45/bp_fe_top` and the steiner tree difference we were talking about over email?

proppy, 2022-06-13:
@vvbandeira `OPENLANE_TAG=2022.02.23_02.50.41`.

vvbandeira, 2022-06-13:
@proppy could you try with the latest version of OL? @luis201420 can confirm, but I would assume he used the latest release of OR, thus looks like the issue has already been fixed upstream.

luis201420, 2022-06-13:
@vvbandeira I think this bug was fixed with #1742.

vvbandeira, 2022-07-14:
@proppy is this still an open issue? I also saw that Cherry made some improvements to the antenna checker code rencently.

vijayank88, 2023-06-19:
@proppy Can we close this?

closed 2025-08-30 by maliberty