The-OpenROAD-Project/OpenROAD #1962: RSZ: setup repair has issues with latch time borrowing

antonblanchard, 2022-06-16:
When building Microwatt using OpenSTA created liberty files, I see setup repair giving up almost immediately:

```
[DEBUG RSZ-0001] pass 1 worst_slack = 0.000 ^
[DEBUG RSZ-0001] pass 2 worst_slack = 0.000 v
[DEBUG RSZ-0001] pass 3 worst_slack = 0.000 v
[DEBUG RSZ-0001] pass 4 worst_slack = 0.000 v
[DEBUG RSZ-0001] pass 5 worst_slack = 0.000 v
[DEBUG RSZ-0001] pass 6 worst_slack = 0.000 v
[DEBUG RSZ-0001] pass 7 worst_slack = 0.000 v
[DEBUG RSZ-0001] pass 8 worst_slack = 0.000 v
[DEBUG RSZ-0001] pass 9 worst_slack = 0.000 v
[DEBUG RSZ-0001] pass 10 worst_slack = 0.000 v
[DEBUG RSZ-0001] decreasing slack for 9 passes. Restoring best slack 0.000
[INFO RSZ-0041] Resized 1 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
```

The path with zero setup slack appears to be:

```
  0.00   0.00   clock ext_clk (rise edge)
  2.96   2.96   clock network delay (propagated)
  0.00   2.96 ^ _116993_/CLK (sky130_fd_sc_hd__dfxtp_4)
  0.53   3.49 v _116993_/Q (sky130_fd_sc_hd__dfxtp_4)
  0.49   3.98 v _057503_/Y (sky130_fd_sc_hd__nor2b_1)
  0.30   4.28 ^ _057505_/Y (sky130_fd_sc_hd__o21ai_4)
  0.18   4.46 v _057719_/Y (sky130_fd_sc_hd__nand2_8)
  0.26   4.73 ^ _057759_/Y (sky130_fd_sc_hd__nor2_8)
  0.21   4.94 ^ repeater2983/X (sky130_fd_sc_hd__buf_16)
  0.19   5.13 ^ repeater2982/X (sky130_fd_sc_hd__buf_16)
  0.19   5.32 ^ repeater2981/X (sky130_fd_sc_hd__buf_16)
  0.16   5.48 v _059470_/Y (sky130_fd_sc_hd__a21oi_4)
  0.17   5.65 ^ _059471_/Y (sky130_fd_sc_hd__clkinv_16)
  0.28   5.93 ^ soc0.bram.bram0.ram_0.memory_0/A0[5] (RAM512)
```

Which makes sense, since it is a path with a latch (in DFFRAM), and it does report 0 slack, however it seems to have the ability to borrow more time:

```
Startpoint: _116993_ (rising edge-triggered flip-flop clocked by ext_clk)
Endpoint: soc0.bram.bram0.ram_0.memory_0
          (negative level-sensitive latch clocked by ext_clk)
Path Group: ext_clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock ext_clk (rise edge)
                          0.00    0.00   clock source latency
                  0.53    0.38    0.38 ^ ext_clk (in)
     1    0.12                           ext_clk (net)
                  0.53    0.00    0.38 ^ repeater15/A (sky130_fd_sc_hd__buf_16)
                  0.37    0.43    0.81 ^ repeater15/X (sky130_fd_sc_hd__buf_16)
     1    0.35                           net7133 (net)
                  0.37    0.01    0.82 ^ clkbuf_0_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.08    0.25    1.07 ^ clkbuf_0_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     2    0.03                           clknet_0_ext_clk (net)
                  0.08    0.00    1.07 ^ clkbuf_1_0_0_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.21    0.27    1.34 ^ clkbuf_1_0_0_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     2    0.12                           clknet_1_0_0_ext_clk (net)
                  0.21    0.00    1.35 ^ clkbuf_2_1_0_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.20    0.32    1.66 ^ clkbuf_2_1_0_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     2    0.12                           clknet_2_1_0_ext_clk (net)
                  0.20    0.00    1.66 ^ clkbuf_3_3_0_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.43    0.49    2.15 ^ clkbuf_3_3_0_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     8    0.26                           clknet_3_3_0_ext_clk (net)
                  0.43    0.00    2.15 ^ clkbuf_6_29_0_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.10    0.29    2.45 ^ clkbuf_6_29_0_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     2    0.05                           clknet_6_29_0_ext_clk (net)
[... 81 lines cut ...]
-----------------------------------------------------------------------------
                                  6.04   data required time
                                 -6.04   data arrival time
-----------------------------------------------------------------------------
                                  0.00   slack (MET)

   Time Borrowing Information
--------------------------------------------
ext_clk nominal pulse width            15.00
clock latency difference               -0.52
library setup time                     -1.80
--------------------------------------------
CRPR difference                        -0.03
max time borrow                        12.65
--------------------------------------------
actual time borrow                      2.95
open edge uncertainty                  -0.25
open edge CRPR                          0.10
--------------------------------------------
time given to startpoint                2.81
--------------------------------------------
                             
```

Test case: [microwatt-resizer-2.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8916142/microwatt-resizer-2.tar.gz)

antonblanchard, 2022-06-16:
Attaching the SPEF and gate level verilog for RAM512

[RAM512.spef.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8918233/RAM512.spef.gz)
[RAM512.v.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8918237/RAM512.v.gz)

We have A0[] -> clock gating cells -> latches -> A0[] -> output register -> Do0[]

I'm trying to work out where the above timing path is coming from. A0 feeds clock gating cells also fed by a negative edge of the clock, so there's nothing to steal from in that path.

The same address pins are used for output, so I'm wondering if this timing path is the latch to the output register. If so we could mark the path as invalid, since we only use the RAM for reading or writing in one cycle, not both.

antonblanchard, 2022-06-17:
@jjcherry56 I'm guessing it's `LibertyCell::inferLatchRoles()`.

I also think the liberty file is modelling a multi cycle path for A0 (input setup/hold referenced to CLK in the first cycle and a second cycle async path of A0 -> Do).

jjcherry56, 2022-06-21:
There is a lot going on here. The place to start is by verifying the timing model for the ram. As the doc says, it is not designed to work with latches (there are many issues). But in this case it may just skate by because there is only one latch between the input and outputs. Modeling borrowing across multiple latches is something liberty is not up to. But there has been no testing on my end with transparent latches. Or gated clocks. 

So what you need to do is compare timing with the ram liberty model in microwatt with using the ram verilog.
Use report_checks -through inputs or outputs and check input/output timing. Please don't do things like this:
report_checks -unconstrained -fields {slew cap input nets fanout} -format full_clock_expanded -group_count 1000
because it only generates volumes of incomprehensible data.

I do not think the resizer has been used on a design with transparent latches. But would never use microwatt to debugging/verify that scenario.

LibertyCell::inferLatchRoles is doing it's job. The liberty model extract does not try to include ff/latch groups and there are indeed liberty is modeling latches.

antonblanchard, 2022-06-22:
@jjcherry56 agree with simplifying the test case. I've built the smallest DFFRAM I can (8x8), and have a trivial wrapper around it with no logic. I then compare a hierarchical STA using spef files in OpenSTA vs the liberty file in OpenROAD.

Pins EN0, WE0 and A0 appear to match. We didn't infer the clock gated latch on Di0* however:

```
Startpoint: Di0[0] (input port clocked by CLK)
Endpoint: RAM8_0 (falling edge-triggered flip-flop clocked by CLK)
Path Group: CLK
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock CLK (rise edge)
   0.00    0.00   clock network delay (propagated)
   2.00    2.00 ^ input external delay
   0.08    2.08 ^ Di0[0] (in)
   0.00    2.08 ^ RAM8_0/Di0[0] (RAM8)
           2.08   data arrival time

   5.00    5.00   clock CLK (fall edge)
   0.05    5.05   clock network delay (propagated)
   0.00    5.05   clock reconvergence pessimism
           5.05 v RAM8_0/CLK (RAM8)
   0.40    5.45   library setup time
           5.45   data required time
---------------------------------------------------------
           5.45   data required time
          -2.08   data arrival time
---------------------------------------------------------
           3.37   slack (MET)
```

vs what comes out of the hierarchical STA:

```
Startpoint: Di0[0] (input port clocked by CLK)
Endpoint: RAM8_0/WORD[0].W.BYTE[0].B.BIT[0].genblk1.STORAGE
          (positive level-sensitive latch clocked by CLK')
Path Group: CLK
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock CLK (rise edge)
   0.00    0.00   clock network delay (propagated)
   2.00    2.00 ^ input external delay
   0.06    2.06 ^ Di0[0] (in)
   0.00    2.06 ^ RAM8_0/WORD[0].W.BYTE[0].B.BIT[0].genblk1.STORAGE/D (sky130_fd_sc_hd__dlxtp_1)
           2.06   data arrival time

   5.00    5.00   clock CLK' (rise edge)
   0.59    5.59   clock network delay (propagated)
   0.00    5.59   clock reconvergence pessimism
           5.59 ^ RAM8_0/WORD[0].W.BYTE[0].B.BIT[0].genblk1.STORAGE/GATE (sky130_fd_sc_hd__dlxtp_1)
   0.00    5.59   time borrowed from endpoint
           5.59   data required time
---------------------------------------------------------
           5.59   data required time
          -2.06   data arrival time
---------------------------------------------------------
           3.53   slack (MET)

Time Borrowing Information
--------------------------------------------
CLK' nominal pulse width                5.00
clock latency difference               -0.06
library setup time                     -0.19
--------------------------------------------
max time borrow                         4.75
actual time borrow                      0.00
--------------------------------------------
```

I'm unsure if it makes sense, but can we detect it by the combination of `hold_rising` and `setup_falling` constraints?

I see a separate issue with the output (Do0). With this small configuration, the Do0 outputs are a bit odd in that they are 8 tristate buffers feeding the output directly. This might be a case where an output buffer should be added. I bring it up regardless, just in case there is an easy fix.

The hierarchical spef timing shows 1.71 ns slack:

```
Startpoint: RAM8_0/WORD[6].W.BYTE[0].B.BIT[0].genblk1.STORAGE
            (positive level-sensitive latch clocked by CLK')
Endpoint: Do0[0] (output port clocked by CLK)
Path Group: CLK
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   5.00    5.00   clock CLK' (rise edge)
   0.59    5.59   clock network delay (propagated)
   0.00    5.59 ^ RAM8_0/WORD[6].W.BYTE[0].B.BIT[0].genblk1.STORAGE/GATE (sky130_fd_sc_hd__dlxtp_1)
   0.35    5.94 ^ RAM8_0/WORD[6].W.BYTE[0].B.BIT[0].genblk1.STORAGE/Q (sky130_fd_sc_hd__dlxtp_1)
   0.34    6.28 ^ RAM8_0/WORD[6].W.BYTE[0].B.BIT[0].OBUF0/Z (sky130_fd_sc_hd__ebufn_2)
   0.00    6.29 ^ Do0[0] (out)
           6.29   data arrival time

  10.00   10.00   clock CLK (rise edge)
   0.00   10.00   clock network delay (propagated)
   0.00   10.00   clock reconvergence pessimism
  -2.00    8.00   output external delay
           8.00   data required time
---------------------------------------------------------
           8.00   data required time
          -6.29   data arrival time
---------------------------------------------------------
           1.71   slack (MET)
```

But the liberty file STA shows -0.14ns slack:

```
Startpoint: RAM8_0 (falling edge-triggered flip-flop clocked by CLK)
Endpoint: Do0[0] (output port clocked by CLK)
Path Group: CLK
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   5.00    5.00   clock CLK (fall edge)
   0.05    5.05   clock network delay (propagated)
   0.00    5.05 v RAM8_0/CLK (RAM8)
   3.09    8.14 ^ RAM8_0/Do0[0] (RAM8)
   0.00    8.14 ^ Do0[0] (out)
           8.14   data arrival time

  10.00   10.00   clock CLK (rise edge)
   0.00   10.00   clock network delay (propagated)
   0.00   10.00   clock reconvergence pessimism
  -2.00    8.00   output external delay
           8.00   data required time
---------------------------------------------------------
           8.00   data required time
          -8.14   data arrival time
---------------------------------------------------------
          -0.14   slack (VIOLATED)
```

Test case is at: [RAM8.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8954236/RAM8.tar.gz)

Note that the original issue I reported does not show up in this configuration, I'll work on a test case for that next.

antonblanchard, 2022-06-22:
I verified that the output issue (Do0) goes away when I use a larger DFFRAM configuration (32x8) which registers the output (instead of connecting tristate buffers directly to the output). I think we could ignore that issue.

antonblanchard, 2022-06-22:
I just realised I was running with a modified OpenROAD (which disabled latch inference), sorry for the confusion. The problems I now see with the RAM8 test case:

- We still didn't infer a latch for Di0 (explained in a previous comment).

- We infer a latch for both EN0 and A0, where they are actually a clock gate cell, which we can't borrow timing from:

liberty:
```
Startpoint: EN0 (input port clocked by CLK)
Endpoint: RAM8_0 (negative level-sensitive latch clocked by CLK)
Path Group: CLK
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock CLK (rise edge)
   0.00    0.00   clock network delay (propagated)
   2.00    2.00 ^ input external delay
   0.02    2.02 ^ EN0 (in)
   0.00    2.02 ^ RAM8_0/EN0 (RAM8)
           2.02   data arrival time

   0.00    0.00   clock CLK (rise edge)
   0.08    0.08   clock network delay (propagated)
   0.00    0.08   clock reconvergence pessimism
           0.08 ^ RAM8_0/CLK (RAM8)
   1.93    2.02   time borrowed from endpoint
           2.02   data required time
---------------------------------------------------------
           2.02   data required time
          -2.02   data arrival time
---------------------------------------------------------
           0.00   slack (MET)

Time Borrowing Information
--------------------------------------------
CLK nominal pulse width                 5.00
clock latency difference               -0.03
library setup time                     -0.33
--------------------------------------------
max time borrow                         4.64
actual time borrow                      1.93
--------------------------------------------
```

spef:

```
Startpoint: EN0 (input port clocked by CLK)
Endpoint: RAM8_0/WORD[5].W.BYTE[0].B.genblk1.CG
          (rising clock gating-check end-point clocked by CLK')
Path Group: CLK
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock CLK (rise edge)
   0.00    0.00   clock network delay (propagated)
   2.00    2.00 v input external delay
   0.01    2.01 v EN0 (in)
   0.12    2.12 v RAM8_0/DEC0.ENBUF/X (sky130_fd_sc_hd__clkbuf_2)
   0.23    2.36 v RAM8_0/DEC0.AND5/X (sky130_fd_sc_hd__and4b_2)
   0.10    2.46 v RAM8_0/WORD[5].W.SEL0BUF/X (sky130_fd_sc_hd__clkbuf_2)
   0.12    2.58 v RAM8_0/WORD[5].W.BYTE[0].B.CGAND/X (sky130_fd_sc_hd__and2_1)
   0.00    2.58 v RAM8_0/WORD[5].W.BYTE[0].B.genblk1.CG/GATE (sky130_fd_sc_hd__dlclkp_1)
           2.58   data arrival time

   5.00    5.00   clock CLK' (rise edge)
   0.33    5.33   clock network delay (propagated)
   0.00    5.33   clock reconvergence pessimism
           5.33 ^ RAM8_0/WORD[5].W.BYTE[0].B.genblk1.CG/CLK (sky130_fd_sc_hd__dlclkp_1)
  -0.12    5.21   library setup time
           5.21   data required time
---------------------------------------------------------
           5.21   data required time
          -2.58   data arrival time
---------------------------------------------------------
           2.63   slack (MET)
```

antonblanchard, 2022-06-22:
I built a larger DFFRAM (256x8). This is a set of 2 smaller DFFRAMs each with FFs at the output, followed by a 2:1 mux.  The 2:1 mux creates a combinatorial dependency between A0[5-7] and Do0, and `inferLatchRoles` looks at Do0, finds the connection to A0 and calls `makeLatchEnable`.

Test case: [RAM256.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8957416/RAM256.tar.gz)

jjcherry56, 2022-06-24:
I've been looking at correlating the RAM8. I fixed on bug that affected the CLK->Do0 paths. Hierarchical parasitic annotation only works on the "interior" nets. So the using the RAM8 spef with the wrapper doesn't work correctly. The worse issue is that the stupid default wire load models step in and pollute the net. I suggest you comment out the default_wire_load_mode line in the sky130 liberty. I am going to try and remove default wire load support because it is so insidious but I won't know the fallout until I try it. Bottom line: remove that liberty line and update opensta and the results are pretty close.

antonblanchard, 2022-06-27:
@jjcherry56 thanks, I was able to replicate your results after making those changes.

Is there any hope of getting the latch inference right for the EN0, A0 and Di0 pins? If I use `read_liberty -no_latch_infer` that fixes the EN0 and A0 pins, so only Di0 is incorrectly identified as not being a latch. However, since Di0 has a fair bit of slack and a design likely won't need to borrow time on these pins, perhaps that is a good enough compromise.

I started looking through the liberty spec (terrifyingly large) and wondered if the tlatch attribute could be used on the Di0 pin.

antonblanchard, 2022-07-01:
Ok, will open a new issue.

antonblanchard, 2022-07-01:
Opened https://github.com/The-OpenROAD-Project/OpenROAD/issues/2013

closed 2022-07-01 by antonblanchard