The-OpenROAD-Project/OpenROAD #1995: RSZ: setup repair on asap7 dies with "unsupported path type" error

antonblanchard, 2022-06-24:
When testing a design with the ASAP7 PDK, I hit an error when doing setup repair:

```
Repair setup violations...
[CRITICAL STA-0259] unsupported path type
```

This also reproduces with the gcd test in ORFS. An issue tarball: [cts_gcd_asap7_base_2022-06-25_06-25.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8980613/cts_gcd_asap7_base_2022-06-25_06-25.tar.gz)

jjcherry56, 2022-06-25:
I ran scripts/cts.tcl with the latest OR and it does not reproduce this issue. I honestly don't see any way the code could get to that line since it is inside a debugging function that isn't called by any user commands.

antonblanchard, 2022-06-26:
@jjcherry56 sorry for that, I was sure I rebuilt from pristine upstream source but obviously one of my debug patches was still applied.

commit ae256af2 (2024-03-19): closes #1995
closed 2022-06-26 by antonblanchard