The-OpenROAD-Project/OpenROAD #2013: STA infers a latch from a liberty model with a negative edge triggered clock gate cell

antonblanchard, 2022-07-01:
I'm seeing issues with the following test case: [RAM8-new.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9028580/RAM8-new.tar.gz)

The issue is with the `EN0` and `A0*` inputs. Both feed into a clock gate cell, which is clocked by the inverted clock. A hierarchical STA shows it correctly: 

```
# sta spef.tcl

Startpoint: EN0 (input port clocked by CLK)
Endpoint: RAM8_0/WORD[5].W.BYTE[0].B.genblk1.CG
          (rising clock gating-check end-point clocked by CLK')
Path Group: CLK
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock CLK (rise edge)
   0.00    0.00   clock network delay (propagated)
   2.00    2.00 v input external delay
   0.01    2.01 v EN0 (in)
   0.12    2.12 v RAM8_0/DEC0.ENBUF/X (sky130_fd_sc_hd__clkbuf_2)
   0.23    2.36 v RAM8_0/DEC0.AND5/X (sky130_fd_sc_hd__and4b_2)
   0.10    2.46 v RAM8_0/WORD[5].W.SEL0BUF/X (sky130_fd_sc_hd__clkbuf_2)
   0.12    2.58 v RAM8_0/WORD[5].W.BYTE[0].B.CGAND/X (sky130_fd_sc_hd__and2_1)
   0.00    2.58 v RAM8_0/WORD[5].W.BYTE[0].B.genblk1.CG/GATE (sky130_fd_sc_hd__dlclkp_1)
           2.58   data arrival time

   5.00    5.00   clock CLK' (rise edge)
   0.33    5.33   clock network delay (propagated)
   0.00    5.33   clock reconvergence pessimism
           5.33 ^ RAM8_0/WORD[5].W.BYTE[0].B.genblk1.CG/CLK (sky130_fd_sc_hd__dlclkp_1)
  -0.12    5.21   library setup time
           5.21   data required time
---------------------------------------------------------
           5.21   data required time
          -2.58   data arrival time
---------------------------------------------------------
           2.63   slack (MET)
```

Using a generated liberty model, STA in OpenROAD thinks `EN0` feeds into a latch:

```
# openroad -exit liberty.tcl

Startpoint: EN0 (input port clocked by CLK)
Endpoint: RAM8_0 (negative level-sensitive latch clocked by CLK)
Path Group: CLK
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock CLK (rise edge)
   0.00    0.00   clock network delay (propagated)
   2.00    2.00 ^ input external delay
   0.02    2.02 ^ EN0 (in)
   0.00    2.02 ^ RAM8_0/EN0 (RAM8)
           2.02   data arrival time

   0.00    0.00   clock CLK (rise edge)
   0.08    0.08   clock network delay (propagated)
   0.00    0.08   clock reconvergence pessimism
           0.08 ^ RAM8_0/CLK (RAM8)
   1.93    2.02   time borrowed from endpoint
           2.02   data required time
---------------------------------------------------------
           2.02   data required time
          -2.02   data arrival time
---------------------------------------------------------
           0.00   slack (MET)

Time Borrowing Information
--------------------------------------------
CLK nominal pulse width                 5.00
clock latency difference               -0.03
library setup time                     -0.33
--------------------------------------------
max time borrow                         4.64
actual time borrow                      1.93
--------------------------------------------
```

jjcherry56, 2022-07-02:
I changed the latch inference to require unate D->Q arcs and it looks like it solves this issue. I did not update the OR submodule yet.

antonblanchard, 2022-07-03:
Thanks @jjcherry56 this does look to have fixed it. Closing.

closed 2022-07-03 by antonblanchard