The-OpenROAD-Project/OpenROAD #2027: PDN insufficient width alternative strategy for not enough core area

QuantamHD, 2022-07-06:
```
[WARNING IFP-0028] Core area lower left (30.000, 30.000) snapped to (30.360, 32.640).
[INFO IFP-0001] Added 4 rows of 31 sites.
[ERROR PDN-0185] Insufficient width to add straps on layer met4 in grid "grid".
Error: pdn_config.pdn, 23 PDN-0185
```

Ran into the following PDN error with a small core area of 30,30um. With the following grid strategy, which was directly translated from the original PDN file for Sky130.

```
####################################
# global connections
####################################
add_global_connection -defer_connection -net {VDD} -inst_pattern {.*} -pin_pattern {^VDD$} -power
add_global_connection -defer_connection -net {VDD} -inst_pattern {.*} -pin_pattern {^VDDPE$}
add_global_connection -defer_connection -net {VDD} -inst_pattern {.*} -pin_pattern {^VDDCE$}
add_global_connection -defer_connection -net {VDD} -inst_pattern {.*} -pin_pattern {VPWR}
add_global_connection -defer_connection -net {VDD} -inst_pattern {.*} -pin_pattern {VPB}
add_global_connection -defer_connection -net {VSS} -inst_pattern {.*} -pin_pattern {^VSS$} -ground
add_global_connection -defer_connection -net {VSS} -inst_pattern {.*} -pin_pattern {^VSSE$}
add_global_connection -defer_connection -net {VSS} -inst_pattern {.*} -pin_pattern {VGND}
add_global_connection -defer_connection -net {VSS} -inst_pattern {.*} -pin_pattern {VNB}
global_connect
####################################
# voltage domains
####################################
set_voltage_domain -name {CORE} -power {VDD} -ground {VSS}
####################################
# standard cell grid
####################################
define_pdn_grid -name {grid} -voltage_domains {CORE}
add_pdn_stripe -grid {grid} -layer {met1} -width {0.48} -pitch {5.44} -offset {0} -followpins
add_pdn_stripe -grid {grid} -layer {met4} -width {1.600} -pitch {27.140} -offset {13.570}
add_pdn_stripe -grid {grid} -layer {met5} -width {1.600} -pitch {27.200} -offset {13.600}
add_pdn_connect -grid {grid} -layers {met1 met4}
add_pdn_connect -grid {grid} -layers {met4 met5}
####################################
# macro grids
####################################
####################################
# grid for: CORE_macro_grid_1
####################################
define_pdn_grid -name {CORE_macro_grid_1} -voltage_domains {CORE} -macro -orient {R0 R180 MX MY} -halo {2.0 2.0 2.0 2.0} -default -grid_over_boundary
add_pdn_connect -grid {CORE_macro_grid_1} -layers {met4 met5}
####################################
# grid for: CORE_macro_grid_2
####################################
define_pdn_grid -name {CORE_macro_grid_2} -voltage_domains {CORE} -macro -orient {R90 R270 MXR90 MYR90} -halo {2.0 2.0 2.0 2.0} -default -grid_over_boundary
```

Given that this did pass before is there an alternative strategy we could take in regards to small core area blocks? Perhaps by centering the strap insteading of heeding the offset value? @maliberty @arlpetergadfort

arlpetergadfort, 2022-07-06:
@QuantamHD with an area of 30 x 30, you are lucky to get 1 set of straps in on met4 and met5 (you should consider lowing those or just use a ring with a single strap in the middle).
In this instance, the user is specifying you want the straps offset that much with a given pitch. I don't think PDNgen should override the given parameters unless the there is a compelling reason to.

maliberty, 2022-07-06:
Do you have a screen shot of the PDN formerly generated?

vijayank88, 2023-06-19:
@QuantamHD 
Is that need further support?

gadfort, 2024-10-18:
@QuantamHD Is this still an issue?