The-OpenROAD-Project/OpenROAD #2031: How to resolve large differences between placement parasitics and global route parasitics

antonblanchard, 2022-07-06:
I'm trying to improve the frequency of my gate level multiplier when taping out on ASAP7. The slowest path after global routing was not the slowest path before global routing, and this path stands out because of how much worse it got.

Looking into this path, `net38` stands out since it got significantly worse (both in delay and slew):

STA before global routing (ie placement parasitics):
```
Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (propagated)
                        100.00  100.00 v input external delay
                  0.00    0.00  100.00 v a[43] (in)
     1    2.82                           a[43] (net)
                  0.24    0.08  100.08 v input38/A (BUFx24_ASAP7_75t_R)
                 46.15   25.90  125.97 v input38/Y (BUFx24_ASAP7_75t_R)
    72  157.91                           net38 (net)                                     <-----
                 67.34   12.93  138.90 v U$$2880/A (AND2x2_ASAP7_75t_R)
                  8.33   33.18  172.08 v U$$2880/Y (AND2x2_ASAP7_75t_R)
     1    0.69                           t$10037 (net)
                  8.33    0.02  172.11 v U$$2881/B1 (AO32x2_ASAP7_75t_R)
                 27.00   36.64  208.74 v U$$2881/Y (AO32x2_ASAP7_75t_R)
     3    7.80                           sel_0$10038 (net)
                 27.00    0.01  208.75 v rebuffer562/A (BUFx12f_ASAP7_75t_R)
                 10.56   19.23  227.98 v rebuffer562/Y (BUFx12f_ASAP7_75t_R)
...
```

STA after global routing (ie global route parasitics):
```
Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (propagated)
                        100.00  100.00 v input external delay
                  0.00    0.00  100.00 v a[43] (in)
     1    2.85                           a[43] (net)
                  0.36    0.11  100.11 v input38/A (BUFx24_ASAP7_75t_R)
                 31.38   23.16  123.28 v input38/Y (BUFx24_ASAP7_75t_R)
    72  158.75                           net38 (net)                                    <---- much worse
                286.00   90.07  213.35 v U$$2880/A (AND2x2_ASAP7_75t_R) 
                 15.29   63.44  276.79 v U$$2880/Y (AND2x2_ASAP7_75t_R)
     1    1.19                           t$10037 (net)
                 15.30    0.14  276.93 v U$$2881/B1 (AO32x2_ASAP7_75t_R)
                 31.32   39.53  316.46 v U$$2881/Y (AO32x2_ASAP7_75t_R)
     3    9.04                           sel_0$10038 (net)
                 31.55    1.53  317.99 v rebuffer562/A (BUFx12f_ASAP7_75t_R)
                 11.23   19.42  337.42 v rebuffer562/Y (BUFx12f_ASAP7_75t_R)
...
```

The Steiner tree for `net38` looks like this. The driver (`input38`) is the yellow circle on the right and the load (`U$$2880`) is the yellow circle on the left.

[image]

And the 2D tree produced by fastroute:

[image]

The path from `input38` to `U$$2880` got much worse, which seems to explain much of the difference. Interestingly, if I reduce the global routing resource reduction specified, then the 2D tree remains similar to the Steiner tree:

```
-  set_global_routing_layer_adjustment $env(MIN_ROUTING_LAYER)-$env(MAX_ROUTING_LAYER) 0.5
+  set_global_routing_layer_adjustment $env(MIN_ROUTING_LAYER)-$env(MAX_ROUTING_LAYER) 0.4
```

[image]

And setup slack also improves. So we tried to maintain the Steiner tree layout, but ran out of routing resources and modified it significantly. Is this the multi pin maze routing in the global router?

I was trying to think of ways we can improve this (either through better parameters for this design, or changes to OR and ORFS). Some thoughts:

OR changes:
- Global routing should resist modifying the Steiner tree layout for any net on the n% worst timing paths (both setup or hold I would think). Either route them first, or tag them so maze routing doesn't alter the Steiner tree layout.
- Add an ECO stage to fix setup (and hold) violations after detailed routing (we've discussed the value of an ECO flow for hold elsewhere)

ORFS changes:
- Reduce `set_global_routing_layer_adjustment`. The default value used by ASAP7 of 50% seems high, and I confirmed reducing this does result in the tree more or less matching the original Steiner tree, and the the setup slack for this path improves. 
- Specify a reasonable max fan out (`set_max_fanout`) for ASAP7 (and perhaps other platforms). The problem net has a very large fan out (72), and this issue would seem to affect larger fan out nets more. We can't rely on other parameters driving the resizer, because in it's original layout we didn't need to do fan out management. As a test, I specified a max fan out of 20 and it improved the setup slack significantly.
- Run another pass of resizer timing fixup after global routing. This is an option on Openlane. Since we have to go through another pass of detailed placement and global routing, this could back fire (there's nothing to prevent a significant amount of the design getting rerouted). Would an incremental global route step in OR help here? We need that for an ECO flow anyway.

Design changes (ie me)
- Increase the slack margin to overfix the design. This is a bit of a blunt hammer
- Reduce density of my design so there are more routing resources available

@maliberty, @jjcherry56, @tspyrou any thoughts?

maliberty, 2022-07-06:
"taping out on ASAP7" makes no sense as it is an academic process.  Do you mean testing?

antonblanchard, 2022-07-06:
@maliberty yeah, just producing a GDS. With sky130hd being so slow, I wanted a second PDK to test my multiplier design on.

tspyrou, 2022-07-06:
@antonblanchard you can try tweaking the global routing script to run another repair design after global routing. We haven't turned this on by default yet but OpenLane has.

maliberty, 2022-07-06:
The key idea is to make global router more timing aware in its decision about which nets to reroute to fix congestion.  I think that is a reasonable goal as the current method (FastRouteCore::StNetOrder) is based only on congestion.

Repair after detailed routing is a good goal but to make it really work we need an incremental detailed router which we don't have currently.  It is a non-trivial project that we will do eventually.

Library max fanout is often larger than the optimal fanout.  We have some manual controls but could better automate selection.

maliberty, 2022-07-06:
@luis201420 please look at making rip-up-reroute timing aware (@eder-matheus can help).

antonblanchard, 2022-07-19:
> The first place I would look is the RC values. I do not know if they have been correlated for this technology and that is one way to get the sort of discrepancy you are seeing. It looks like the resistance used in the placement based parasitics is too low. Notice that the capacitances are very close. But the slews are very different. That is a resistance issue.

I updated the RC values at https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/524

antonblanchard, 2022-07-21:
I've revisited this issue after fixing the RC layer estimates. Working with OpenROAD git SHA1 d4c41f0b5d1627e0c6bad8362006a51134011fc9.

This path just makes timing (with ~10ps slack) when looking at `estimate_parasitics -placement`:

```
Startpoint: a[7] (input port clocked by clk)
Endpoint: _574_ (falling edge-triggered flip-flop clocked by clk')
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (propagated)
                         50.00   50.00 v input external delay
                  0.00    0.00   50.00 v a[7] (in)
     1    2.90                           a[7] (net)
                  0.29    0.09   50.09 v input62/A (BUFx24_ASAP7_75t_R)
                  9.95   16.45   66.54 v input62/Y (BUFx24_ASAP7_75t_R)
    20   47.62                           net62 (net)                        <--- look here
                100.03   30.50   97.04 v U$$414/A (AND2x4_ASAP7_75t_R)
                  9.17   38.02  135.06 v U$$414/Y (AND2x4_ASAP7_75t_R)
     1    1.27                           t$8795 (net)
                  9.17    0.11  135.18 v U$$415/B1 (AO32x2_ASAP7_75t_R)
                 15.55   29.42  164.60 v U$$415/Y (AO32x2_ASAP7_75t_R)
     1    2.87                           sel_0$8796 (net)
                 15.55    0.12  164.72 v rebuffer415/A (BUFx12f_ASAP7_75t_R)
                 13.28   17.25  181.97 v rebuffer415/Y (BUFx12f_ASAP7_75t_R)
     8   17.95                           net967 (net)
                 24.82    6.79  188.77 v U$$517/A2 (AO22x2_ASAP7_75t_R)
                 10.98   23.35  212.12 v U$$517/Y (AO22x2_ASAP7_75t_R)
     1    1.68                           t$8848 (net)
                 10.98    0.01  212.13 v U$$518/A (XOR2x1_ASAP7_75t_R)
                 38.37   23.31  235.45 v U$$518/Y (XOR2x1_ASAP7_75t_R)
     1    2.07                           booth_b6_m50 (net)
                 38.37    0.02  235.47 v dadda_fa_0_56_1/A (FAx1_ASAP7_75t_R)
                 28.62   43.28  278.75 v dadda_fa_0_56_1/SN (FAx1_ASAP7_75t_R)
     1    0.95                           sn$14 (net)
                 28.62    0.04  278.79 v U$$4536/A (INVx1_ASAP7_75t_R)
                 17.86   14.39  293.18 ^ U$$4536/Y (INVx1_ASAP7_75t_R)
     1    1.78                           s$487 (net)
                 17.86    0.05  293.23 ^ dadda_fa_1_56_8/CI (FAx1_ASAP7_75t_R)
                 22.55   17.21  310.44 v dadda_fa_1_56_8/CON (FAx1_ASAP7_75t_R)
     8    0.92                           con$483 (net)
                 22.55    0.03  310.47 v U$$4993/A (INVx1_ASAP7_75t_R)
                 18.30   14.29  324.76 ^ U$$4993/Y (INVx1_ASAP7_75t_R)
     1    2.13                           c$1859 (net)
                 18.30    0.05  324.80 ^ dadda_fa_2_57_3/A (FAx1_ASAP7_75t_R)
                 22.56   35.48  360.28 v dadda_fa_2_57_3/SN (FAx1_ASAP7_75t_R)
     1    0.74                           sn$1858 (net)
                 22.56    0.01  360.29 v U$$5938/A (INVx1_ASAP7_75t_R)
                 18.76   14.61  374.90 ^ U$$5938/Y (INVx1_ASAP7_75t_R)
     1    2.23                           s$3659 (net)
                 18.76    0.02  374.92 ^ dadda_fa_3_57_3/B (FAx1_ASAP7_75t_R)
                 30.56   40.32  415.23 v dadda_fa_3_57_3/SN (FAx1_ASAP7_75t_R)
     1    1.11                           sn$3657 (net)
                 30.56    0.06  415.30 v U$$6822/A (INVx1_ASAP7_75t_R)
                 18.71   15.00  430.30 ^ U$$6822/Y (INVx1_ASAP7_75t_R)
     1    1.83                           s$5306 (net)
                 18.71    0.06  430.37 ^ dadda_fa_4_57_2/CI (FAx1_ASAP7_75t_R)
                 30.84   33.85  464.22 ^ dadda_fa_4_57_2/SN (FAx1_ASAP7_75t_R)
     1    0.80                           sn$5303 (net)
                 30.84    0.01  464.23 ^ U$$7558/A (INVx1_ASAP7_75t_R)
                 16.41   13.00  477.23 v U$$7558/Y (INVx1_ASAP7_75t_R)
     1    1.82                           s$6688 (net)
                 16.41    0.06  477.29 v dadda_fa_5_57_1/CI (FAx1_ASAP7_75t_R)
                 29.59   19.27  496.56 ^ dadda_fa_5_57_1/CON (FAx1_ASAP7_75t_R)
     8    0.76                           con$6684 (net)
                 29.59    0.01  496.57 ^ U$$8147/A (INVx1_ASAP7_75t_R)
                 18.00   14.19  510.75 v U$$8147/Y (INVx1_ASAP7_75t_R)
[... 285 lines cut ...]
                  3.76    1.19  501.19 ^ clkbuf_0_clk/A (BUFx4_ASAP7_75t_R)
                 11.00   18.21  519.39 ^ clkbuf_0_clk/Y (BUFx4_ASAP7_75t_R)
     2    3.67                           clknet_0_clk (net)
                 11.03    0.28  519.67 ^ clkbuf_1_1__f_clk/A (BUFx4_ASAP7_75t_R)
                 23.05   25.50  545.17 ^ clkbuf_1_1__f_clk/Y (BUFx4_ASAP7_75t_R)
     5   10.70                           clknet_1_1__leaf_clk (net)
                 23.35    1.47  546.64 ^ clkbuf_leaf_4_clk/A (BUFx4_ASAP7_75t_R)
                114.01   62.55  609.19 ^ clkbuf_leaf_4_clk/Y (BUFx4_ASAP7_75t_R)
    30   65.92                           clknet_leaf_4_clk (net)
                114.58    4.75  613.94 ^ net726_198/A (INVx3_ASAP7_75t_R)
                 15.18    7.76  621.70 v net726_198/Y (INVx3_ASAP7_75t_R)
     1    0.75                           net750 (net)
                 15.18    0.05  621.75 v _574_/CLK (DFFLQNx2_ASAP7_75t_R)
                          0.00  621.75   clock reconvergence pessimism
[... cut ...]
                                614.87   data required time
-----------------------------------------------------------------------------
                                614.87   data required time
                               -614.53   data arrival time
-----------------------------------------------------------------------------
                                  0.34   slack (MET)
```

Test case: 
[grt-ripup-large-net.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9156110/grt-ripup-large-net.tar.gz) (see `doit.tcl`)

maliberty, 2022-07-21:
@eder-matheus @luis201420 please use this as a test case for your timing driven ripup in grt.  Can we achieve a similar result with the net specific hacking?

antonblanchard, 2022-07-21:
I ran the design all the way through detailed routing, before and after the net specific hack. I noticed one of the modifications to the net was tapping off at the mid point of a wire, and wondered if `estimate_parasitics -global_routing` could give pessimistic results in that case.

Anyway, the issue is still there after DRT, if a little less prononounced:

```
Startpoint: a[7] (input port clocked by clk)
Endpoint: _574_ (falling edge-triggered flip-flop clocked by clk')
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (propagated)
                         50.00   50.00 v input external delay
                  0.00    0.00   50.00 v a[7] (in)
     1    2.87                           a[7] (net)
                  0.44    0.14   50.14 v input62/A (BUFx24_ASAP7_75t_R)
                  8.54   15.92   66.06 v input62/Y (BUFx24_ASAP7_75t_R)
    20   50.09                           net62 (net)                          <------ here
                149.80   46.60  112.66 v U$$414/A (AND2x4_ASAP7_75t_R)
                 11.06   45.89  158.55 v U$$414/Y (AND2x4_ASAP7_75t_R)
     1    1.75                           t$8795 (net)
                 11.07    0.24  158.79 v U$$415/B1 (AO32x2_ASAP7_75t_R)
                 14.33   29.01  187.80 v U$$415/Y (AO32x2_ASAP7_75t_R)
     1    2.99                           sel_0$8796 (net)
                 14.34    0.24  188.04 v rebuffer415/A (BUFx12f_ASAP7_75t_R)
                 12.72   16.22  204.26 v rebuffer415/Y (BUFx12f_ASAP7_75t_R)
     8   20.37                           net967 (net)
                 32.24    9.55  213.80 v U$$517/A2 (AO22x2_ASAP7_75t_R)
                 12.91   24.46  238.26 v U$$517/Y (AO22x2_ASAP7_75t_R)
     1    1.69                           t$8848 (net)
                 12.91    0.07  238.34 v U$$518/A (XOR2x1_ASAP7_75t_R)
                 41.48   23.29  261.63 v U$$518/Y (XOR2x1_ASAP7_75t_R)
     1    2.04                           booth_b6_m50 (net)
                 41.48    0.07  261.69 v dadda_fa_0_56_1/A (FAx1_ASAP7_75t_R)
                 27.68   42.19  303.88 v dadda_fa_0_56_1/SN (FAx1_ASAP7_75t_R)
     1    1.00                           sn$14 (net)
                 27.68    0.07  303.96 v U$$4536/A (INVx1_ASAP7_75t_R)
                 16.50   13.46  317.42 ^ U$$4536/Y (INVx1_ASAP7_75t_R)
     1    1.90                           s$487 (net)
                 16.51    0.12  317.54 ^ dadda_fa_1_56_8/CI (FAx1_ASAP7_75t_R)
                 22.34   16.76  334.30 v dadda_fa_1_56_8/CON (FAx1_ASAP7_75t_R)
     8    1.09                           con$483 (net)
                 22.34    0.07  334.36 v U$$4993/A (INVx1_ASAP7_75t_R)
                 16.91   13.33  347.69 ^ U$$4993/Y (INVx1_ASAP7_75t_R)
     1    2.09                           c$1859 (net)
                 16.92    0.14  347.83 ^ dadda_fa_2_57_3/A (FAx1_ASAP7_75t_R)
                 20.20   32.57  380.40 v dadda_fa_2_57_3/SN (FAx1_ASAP7_75t_R)
     1    0.75                           sn$1858 (net)
                 20.20    0.02  380.42 v U$$5938/A (INVx1_ASAP7_75t_R)
                 16.30   12.89  393.31 ^ U$$5938/Y (INVx1_ASAP7_75t_R)
     1    2.23                           s$3659 (net)
                 16.30    0.07  393.38 ^ dadda_fa_3_57_3/B (FAx1_ASAP7_75t_R)
                 32.26   40.58  433.97 v dadda_fa_3_57_3/SN (FAx1_ASAP7_75t_R)
     1    1.36                           sn$3657 (net)
                 32.26    0.13  434.09 v U$$6822/A (INVx1_ASAP7_75t_R)
                 18.38   14.78  448.87 ^ U$$6822/Y (INVx1_ASAP7_75t_R)
     1    2.02                           s$5306 (net)
                 18.39    0.20  449.07 ^ dadda_fa_4_57_2/CI (FAx1_ASAP7_75t_R)
                 27.97   31.63  480.71 ^ dadda_fa_4_57_2/SN (FAx1_ASAP7_75t_R)
     1    0.82                           sn$5303 (net)
                 27.97    0.04  480.75 ^ U$$7558/A (INVx1_ASAP7_75t_R)
                 14.54   11.70  492.44 v U$$7558/Y (INVx1_ASAP7_75t_R)
     1    1.91                           s$6688 (net)
                 14.54    0.12  492.56 v dadda_fa_5_57_1/CI (FAx1_ASAP7_75t_R)
                 27.27   17.42  509.99 ^ dadda_fa_5_57_1/CON (FAx1_ASAP7_75t_R)
     8    0.73                           con$6684 (net)
                 27.27    0.01  510.00 ^ U$$8147/A (INVx1_ASAP7_75t_R)
                 15.72   12.46  522.46 v U$$8147/Y (INVx1_ASAP7_75t_R)
[... 127 lines cut ...]
                  0.00    0.00  500.00 ^ clk (in)
     1    5.51                           clk (net)
                  4.62    1.46  501.46 ^ clkbuf_0_clk/A (BUFx4_ASAP7_75t_R)
                 11.28   18.58  520.03 ^ clkbuf_0_clk/Y (BUFx4_ASAP7_75t_R)
     2    3.81                           clknet_0_clk (net)
                 11.32    0.35  520.38 ^ clkbuf_1_1__f_clk/A (BUFx4_ASAP7_75t_R)
                 21.77   24.67  545.05 ^ clkbuf_1_1__f_clk/Y (BUFx4_ASAP7_75t_R)
     5    9.70                           clknet_1_1__leaf_clk (net)
                 22.13    1.60  546.65 ^ clkbuf_leaf_4_clk/A (BUFx4_ASAP7_75t_R)
                 94.18   52.41  599.06 ^ clkbuf_leaf_4_clk/Y (BUFx4_ASAP7_75t_R)
    30   53.18                           clknet_leaf_4_clk (net)
                 95.23    5.84  604.90 ^ net726_198/A (INVx3_ASAP7_75t_R)
[... cut ...]
                          0.00  612.20   clock reconvergence pessimism
                         -7.17  605.03   library setup time
                                605.03   data required time
-----------------------------------------------------------------------------
                                605.03   data required time
                               -586.01   data arrival time
-----------------------------------------------------------------------------
                                 19.01   slack (MET)

```

antonblanchard, 2022-07-21:
> BTW @antonblanchard - Have you looked at the work Teo did to explore adders? The eventual goal was to expand that to other things like multipliers too.

@mithro I have checked it out, good stuff. I've been more focused on improving the frequency of various algorithms via improvements to OpenROAD, which should hopefully benefit Teo too.

jjcherry56, 2022-07-21:
There is another knob you can use in this situation.
The set_routing_alpha command controls the tradeoff between minimizing wire length and distance from the driver.
Unfortunately the author never bothered to document the command.
The default value is .3. I changed it to .6 and the placement based slack went to 25ps, global to 10ps.
Optimizing the distance from the driver to the loads on critical nets in small geometry technologies where wire resistance becomes important is probably more beneficial than global routing optimizations.
It would make sense to bump alpha on critical nets and save them through the flow.

It is always going to be important to have some -slack_margiin to prevent paths from surfacing between timing repair and detailed routing. There is always going to be imperfections in the approximations used along the way until every part of the flow is incremental (in a galaxy far, far away). I can't tell how you got the db in this testcase, but another place this could be addressed is by repair_timing to split the net. But it can't do that unless it is within its radar, which means within the slack margin.

Note that the database in the testcase was not readable in an older version of OR that I am working on to address your fanout issue. The db rev's pretty frequently and there is no backward compatibility so it is better to use DEF for the testcase to survive the transitiions.

antonblanchard, 2022-07-24:
> The set_routing_alpha command controls the tradeoff between minimizing wire length and distance from the driver.

Thanks @jjcherry56 this does look to have an impact on some of these large nets. I'll experiment with it.

> Note that the database in the testcase was not readable in an older version of OR that I am working on to address your fanout issue

Good to know, I'll make sure to attach LEF/DEFs.

antonblanchard, 2022-08-01:
Here is another path where timing gets significantly worse from placement to global routing. A big issue in this case is a 2 pin net, `sn$1367`:

placement parasitics:

```
Startpoint: a[25] (input port clocked by clk)
[WARNING GUI-0066] Heat map "Routing Congestion" has not been populated with data.
Endpoint: _554_ (falling edge-triggered flip-flop clocked by clk')
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (propagated)
                         50.00   50.00 ^ input external delay
                  0.00    0.00   50.00 ^ a[25] (in)
     1    0.80                           a[25] (net)
                  0.03    0.01   50.01 ^ input18/A (BUFx3_ASAP7_75t_R)
                  9.85   12.01   62.02 ^ input18/Y (BUFx3_ASAP7_75t_R)
     1    3.23                           net18 (net)
                  9.87    0.29   62.31 ^ fanout986/A (BUFx16f_ASAP7_75t_R)
                 11.66   15.38   77.69 ^ fanout986/Y (BUFx16f_ASAP7_75t_R)
     7   15.17                           net986 (net)
                 24.66    6.89   84.58 ^ fanout981/A (BUFx16f_ASAP7_75t_R)
                  9.79   19.71  104.29 ^ fanout981/Y (BUFx16f_ASAP7_75t_R)
     6   10.38                           net981 (net)
                 10.19    1.01  105.30 ^ U$$1768/A (INVx1_ASAP7_75t_R)
                 15.81   10.72  116.02 v U$$1768/Y (INVx1_ASAP7_75t_R)
     1    2.66                           notblock$9471[0] (net)
                 15.86    0.53  116.55 v U$$1771/B3 (AO33x2_ASAP7_75t_R)
                 14.39   35.51  152.06 v U$$1771/Y (AO33x2_ASAP7_75t_R)
     1    2.71                           sel_0$9472 (net)
                 14.39    0.05  152.11 v fanout501/A (BUFx16f_ASAP7_75t_R)
                 10.68   17.83  169.94 v fanout501/Y (BUFx16f_ASAP7_75t_R)
     6   20.75                           net501 (net)
                 11.46    1.11  171.06 v fanout498/A (BUFx12f_ASAP7_75t_R)
                  8.99   14.96  186.02 v fanout498/Y (BUFx12f_ASAP7_75t_R)
     7    9.50                           net498 (net)
                  8.99    0.08  186.10 v split524/A (BUFx2_ASAP7_75t_R)
                 15.97   19.43  205.53 v split524/Y (BUFx2_ASAP7_75t_R)
     5    4.89                           net1843 (net)
                 15.97    0.10  205.63 v U$$1795/A2 (AO22x1_ASAP7_75t_R)
                 13.22   23.25  228.88 v U$$1795/Y (AO22x1_ASAP7_75t_R)
     1    1.74                           t$9485 (net)
                 13.22    0.03  228.91 v U$$1796/A (XOR2x1_ASAP7_75t_R)
                 16.28   23.42  252.33 v U$$1796/Y (XOR2x1_ASAP7_75t_R)
     1    1.82                           booth_b26_m11 (net)
                 16.29    0.06  252.39 v dadda_fa_2_37_2/CI (FAx1_ASAP7_75t_R)
                 60.32   58.78  311.16 v dadda_fa_2_37_2/SN (FAx1_ASAP7_75t_R)
     1    2.42                           sn$1367 (net)                          <---- here
                 60.33    0.43  311.59 v U$$5663/A (INVx1_ASAP7_75t_R)
                 36.21   27.66  339.26 ^ U$$5663/Y (INVx1_ASAP7_75t_R)
     1    3.61                           s$3258 (net)
                 36.25    0.71  339.96 ^ dadda_fa_3_37_3/A (FAx1_ASAP7_75t_R)
                 36.49   26.36  366.32 v dadda_fa_3_37_3/CON (FAx1_ASAP7_75t_R)
     8    2.12                           con$3256 (net)
                 60.25   26.95  393.27 ^ dadda_fa_3_37_3/SN (FAx1_ASAP7_75t_R)
     1    2.21                           sn$3257 (net)
                 60.26    0.35  393.61 ^ U$$6629/A (INVx1_ASAP7_75t_R)
                 27.27   20.88  414.49 v U$$6629/Y (INVx1_ASAP7_75t_R)
     1    2.81                           s$5006 (net)
                 27.29    0.42  414.91 v dadda_fa_4_37_2/CI (FAx1_ASAP7_75t_R)
                 41.78   28.74  443.64 ^ dadda_fa_4_37_2/CON (FAx1_ASAP7_75t_R)
     8    1.71                           con$5002 (net)
                 41.78    0.19  443.84 ^ U$$7404/A (INVx1_ASAP7_75t_R)
                 22.72   17.65  461.49 v U$$7404/Y (INVx1_ASAP7_75t_R)
     1    2.60                           c$6493 (net)
                 22.74    0.33  461.82 v dadda_fa_5_38_0/CI (FAx1_ASAP7_75t_R)
                 39.82   47.44  509.26 v dadda_fa_5_38_0/SN (FAx1_ASAP7_75t_R)
[... 149 lines cut ...]
                          0.00  598.61   clock reconvergence pessimism
                         -8.52  590.09   library setup time
                                590.09   data required time
-----------------------------------------------------------------------------
                                590.09   data required time
                               -906.73   data arrival time
-----------------------------------------------------------------------------
                               -316.64   slack (VIOLATED)
```

Steiner tree:

[image]

Global route 2D tree:

[image]

We've struggled to route this net, and unfortunately it was on a critical path. A few thoughts:

- While we can print wire length after global and detailed routing (with `report_wire_length`), I see no option to print it after placement.To be accurate it should match what the resizer is using. We can use that for both placement and global route metrics (eg average increase in wire length, worst case increase in wire length etc). I had a go at augmenting `report_wire_len [... cut ...]
- An estimation of how well global routing did, or perhaps a warning when very scenic routes are created would be useful. Before I looked at these specific nets, I wasn't sure how to further optimise my design. Now I'm thinking: perhaps the design is too dense, perhaps global placement could be tweaked etc.
- Might be another test case for global routing (it's possible the issue goes back to global placement however). Should have routed this earlier, considering it was on a critical path.

Test case:  [bad-global-route.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9233800/bad-global-route.tar.gz)

maliberty, 2025-08-25:
we have post-grt repair now