The-OpenROAD-Project/OpenROAD #2039: Net not routable

13vaidehi, 2022-07-07:
While using OpenROAD for PnR of leaf chip using sky130hd pdk, an error saying net not routable occures, how to solve this issue.
This is the error displayed.
[ERROR DRT-0305] Net one_ of signal type POWER is not routable by TritonRoute. Move to special nets.
[ERROR GUI-0070] Error: flow.tcl, 193 DRT-0305

How to solve this

13vaidehi, 2022-07-07:
I synthesised I checked my netlist, it doesn't contain a 1'b1 or 1'b0.
I used yosys to synthesis netlist 

[image]

I used thsi flow.

vijayank88, 2022-07-08:
@13vaidehi 
Are you using OpenROAD standalone for PnR or along with OpenLane flow?

13vaidehi, 2022-07-08:
I am just using OpenROAD for PnR

vijayank88, 2022-07-08:
@13vaidehi Have you checked DEF file?
Share the netlist and DEF to debug the issue.

Use the @maliberty suggestion during synthesis and start the PnR again

13vaidehi, 2022-07-08:
> We usually use this in synthesis to handle tying off constants:
> 
> ```
> hilomap -singleton \
>         -hicell {*}$::env(TIEHI_CELL_AND_PORT) \
>         -locell {*}$::env(TIELO_CELL_AND_PORT)
> ```

I tried this, I still have the same error

13vaidehi, 2022-07-08:
> @13vaidehi Have you checked DEF file? Share the netlist and DEF to debug the issue.
> 
> Use the @maliberty suggestion during synthesis and start the PnR again

/home/vaidehi/Downloads/leaf-mpw6-main/def/leaf_chip.def
This is my def file.

/home/vaidehi/Documents/leaf_chip.v
This is the netlist after using the hilo command during synthesis.

[image]
This how the error looks in gui

vijayank88, 2022-07-08:
@13vaidehi Upload the zip file here by drag and drop

13vaidehi, 2022-07-08:
[leaf_chip_netlist.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9069268/leaf_chip_netlist.zip)
[leaf_chip_def.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9069271/leaf_chip_def.zip)

vijayank88, 2022-07-08:
@13vaidehi 
Can you share complete flow script what you used with all required inputs?
Which version of OpenROAD currently used?

13vaidehi, 2022-07-08:
[tcl file that calls all necessary files.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9069434/tcl.file.that.calls.all.necessary.files.zip)
[file that has flow.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9069440/file.that.has.flow.zip)
[file with all the necessary files and data for flow.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9069443/file.with.all.the.necessary.files.and.data.for.flow.zip)

0cc525594fe7106e680cb4ac0840a3e41f4dc187 
This is the version I am using

vijayank88, 2022-07-08:
@13vaidehi 
Following block causing the issue with `'h` 
```
down_counter_3 tx_counter (
    .clk(clk),
    .clr(curr_state[0]),
    .en(tx_counter_en),
    .load(3'h7),
    .mode(1'h0),
    .val(\tx_counter:798 )
  );
  piso_8 tx_piso (
    .clk(clk),
    .clr(curr_state[0]),
    .en(tx_piso_en),
    .load(rd_data),
    .mode(curr_state[2]),
    .ser(\tx_piso:818 )
  );
  assign _094_[0] = reset;
  assign _104_[1] = 1'h0;
  assign _119_[0] = reset;
  assign _120_[1] = reset;
  assign _121_[0] = reset;
  assign { _122_[2], _122_[0] } = { next_state[2], next_state[0] };
  assign { _127_[14:11], _127_[8:7], _127_[4:3], _127_[1] } = { 2'h0, curr_state[0], 2'h0, curr_state[2], _105_[0], _105_[0], 1'h0 };
  assign _128_[2:1] = { _104_[0], _106_ };
  assign _02_[2:1] = { 1'h0, rd_en };
```

Use `insert_tiecells` feature from openroad to avoid routing issue.

13vaidehi, 2022-07-08:
Can you help me with the exact use of insert_tiecells?
I am confused where to use it?

vijayank88, 2022-07-08:
https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/ifp#inserting-tieoff-cells Documentation here. 
It should be done during floorplan itself

13vaidehi, 2022-07-08:
What will be the master and output port in this case?

vijayank88, 2022-07-08:
@13vaidehi 
Are you trying this design https://github.com/daniel-santos-7/leaf-mpw6?

13vaidehi, 2022-07-08:
Yes, this is the design I am trying

vijayank88, 2022-07-08:
```
insert_tiecells sky130_fd_sc_hd__conb_1/HI -prefix "TIE_ONE_"
```
Use above line after `initialize_floorplan`

vijayank88, 2022-07-08:
@13vaidehi 
Test your design with OpenLane flow and let me know.

13vaidehi, 2022-07-08:
> @13vaidehi Have you tried same design with OpenLane flow? Its already passed routing stage.

I haven't tried this with OpenLane flow.
If I am not wrong OpeLane will do the synthesis to routing stage right?

13vaidehi, 2022-07-08:
This is the
[image]
 error and the line that the error indicates
[image]

13vaidehi, 2022-07-08:
Is this the only way to go about this issue?
I wanted to use OpenROAD specifically for PnR.

vijayank88, 2022-07-08:
@13vaidehi 
I have modified few things wrt., your `flow.tcl` script and now detail routing working fine. Used the updated `flow.tcl` and let me know.
Is your `results` directory contains all the DB files?

vijayank88, 2022-07-08:
[update_flow.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9070954/update_flow.zip)

vijayank88, 2022-07-08:
> This is the [image] error and the line that the error indicates [image]

It is because of special nets. nets not tied properly. With the `insert_tiecells` its working fine now. Check my `f.ow.tcl` attached.

13vaidehi, 2022-07-08:
[image]
I used the new flow, this is the error I get now

13vaidehi, 2022-07-08:
[image]
I used the updated flow, this is the new error I encountered.

vijayank88, 2022-07-08:
Guide file could not be opened. Check guide file name is correct or not?

vijayank88, 2022-07-09:
@13vaidehi Hope you can able to complete the routing.

13vaidehi, 2022-07-11:
Even after numerous attempts, the error still remains.
The error isn't in the guide file name
[image]

vijayank88, 2022-07-11:
@13vaidehi 
Can you share your flow.tcl and detailed routing log?

vijayank88, 2022-07-11:
I suggest you switch to https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts as it is a more complete production flow versus the unit test flow you seem to be using. Your tarball doesn't include your netlist and isn't usable as a test case.

With ORFS you can package a test case easily with the _issue targets.

13vaidehi, 2022-07-11:
[flow.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9081567/flow.zip)

13vaidehi, 2022-07-11:
I will switch to the flow scripts, but can you once have a look and tell me if its not debuggable at all

vijayank88, 2022-07-11:
@13vaidehi 
At my end i'm not facing any issues with detail routing.

update following part and run again:
```
# Global routing

pin_access
set route_guide "./lc_route_guide.guide"
global_route -guide_file $route_guide \
  -congestion_iterations 100

set verilog_file "lc.v"
write_verilog -remove_cells $filler_cells $verilog_file

################################################################
# Antenna repair

check_antennas
utl::metric "GRT::ANT::errors" [ant::antenna_violation_count]
#repair_antennas

################################################################
# Detailed routing

set_thread_count [exec getconf _NPROCESSORS_ONLN]
detailed_route -guide $route_guide \
               -output_guide "./lc_output_guide.mod" \
		-output_drc  "./lc_route_drc.rpt" \
               -output_maze "./lc_maze.log" \
               -verbose 1
```

13vaidehi, 2022-07-11:
I will do it. Did you run it in OpenROAD?

13vaidehi, 2022-07-11:
I will do it. Did you run it on OpenROAD?

13vaidehi, 2022-07-11:
[image]
How do I change the permission of a file that is being created?

vijayank88, 2022-07-11:
check the `results` directory access/permissions.

that's why suggesting to use tested script wrapper from https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts

What ever variable you're passing, same thing you can use in that ORFS flow itself/ its already pre-defined. its easy to track and all required inputs and configuration path set properly.

What is your final goal?

vijayank88, 2022-07-12:
@13vaidehi 
are you able to fix the access issue?

vijayank88, 2022-08-01:
Closing this issue its passed routing stage.

closed 2022-08-01 by vijayank88