The-OpenROAD-Project/OpenROAD #2055: Stack trace when running partition_design

braamBeresford, 2022-07-12:
When trying to run partition_design as part of the following flow I receive a stack trace. Cannot find source of issue.

Command:
`set top_design ariane
set def_file ./design/ariane.def
set ALL_LEFS "
    ./lefs/NangateOpenCellLibrary.tech.lef
    ./lefs/NangateOpenCellLibrary.macro.mod.lef
    ./lefs/fakeram45_256x16.lef
"
foreach lef_file ${ALL_LEFS} {
    read_lef $lef_file
}
read_def ./design/ariane.def
partition_design -max_num_inst 2000000 -min_num_inst 40000 -max_num_macro 12 -min_num_macro 4 -net_threshold 20  -virtual_weight 50 -num_hop 0 -timing_weight 1000 -report_file ${top_design}.hgr 
exit
`

Error:
`OpenROAD GITDIR-NOTFOUND 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./lefs/NangateOpenCellLibrary.tech.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0226] Finished LEF file:  ./lefs/NangateOpenCellLibrary.tech.lef
[INFO ODB-0222] Reading LEF file: ./lefs/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0225]     Created 135 library cells
[INFO ODB-0226] Finished LEF file:  ./lefs/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0222] Reading LEF file: ./lefs/fakeram45_256x16.lef
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./lefs/fakeram45_256x16.lef
[INFO ODB-0127] Reading DEF file: ./design/ariane.def
[INFO ODB-0128] Design: ariane
[INFO ODB-0097] 		Created 100000 Nets
[INFO ODB-0130]     Created 495 pins.
[INFO ODB-0131]     Created 87585 components and 578948 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 175170 connections.
[INFO ODB-0133]     Created 113121 nets and 403778 connections.
[INFO ODB-0134] Finished DEF file: ./design/ariane.def
Running Partition Design...
[INFO PAR-0402] Traversed logical hierarchy
	Number of std cell instances: 0
	Total area: 0.0
	Number of hard macros: 0
Stack trace:
 0# 0x0000000000C15A16 in openroad
 1# 0x00007FEA8BA99400 in /lib64/libc.so.6
 2# sta::LibertyCell::isBuffer() const in openroad
 3# par::AutoClusterMgr::getBufferNetUtil(odb::dbBlock*, std::vector<std::pair<odb::dbNet*, odb::dbNet*>, std::allocator<std::pair<odb::dbNet*, odb::dbNet*> > >&) in openroad
 4# par::AutoClusterMgr::getBufferNet() in openroad
 5# par::AutoClusterMgr::partitionDesign(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, bool, char const*, char const*, float, float, float, float) in openroad
 6# par::PartitionMgr::partitionDesign(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, bool, char const*, char const*, float, float, float, float) in openroad
 7# 0x00000000018ADCD6 in openroad
 8# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
 9# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
10# TclObjInterpProcCore in /lib64/libtcl8.5.so
11# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
12# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
13# 0x00007FEA8E62E647 in /lib64/libtcl8.5.so
14# TclEvalObjEx in /lib64/libtcl8.5.so
15# 0x00007FEA8E66927F in /lib64/libtcl8.5.so
16# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
17# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
18# 0x00007FEA8E62E647 in /lib64/libtcl8.5.so
19# TclEvalObjEx in /lib64/libtcl8.5.so
20# 0x00007FEA8E5E91D0 in /lib64/libtcl8.5.so
21# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
22# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
23# 0x00007FEA8E62E647 in /lib64/libtcl8.5.so
24# TclEvalObjEx in /lib64/libtcl8.5.so
25# 0x00007FEA8E5ECF00 in /lib64/libtcl8.5.so
26# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
27# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
28# TclObjInterpProcCore in /lib64/libtcl8.5.so
29# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
30# 0x00007FEA8E62636C in /lib64/libtcl8.5.so
31# TclObjInterpProcCore in /lib64/libtcl8.5.so
32# 0x00007FEA8E5E1EB2 in /lib64/libtcl8.5.so
33# 0x00007FEA8E5E2F1E in /lib64/libtcl8.5.so
34# Tcl_EvalEx in /lib64/libtcl8.5.so
35# Tcl_Eval in /lib64/libtcl8.5.so
36# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_Main in /lib64/libtcl8.5.so
39# main in openroad
40# __libc_start_main in /lib64/libc.so.6
41# 0x0000000000C125F7 in openroad
`
LEF and DEF collaterals:
https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat

maliberty, 2022-07-12:
I'll look at making it not crash but the problem is you don't have any Liberty files loaded:

2# sta::LibertyCell::isBuffer() const in openroad

braamBeresford, 2022-07-12:
@maliberty Interesting. I'm running this within the build docker environment so shouldn't I have all dependencies already?

maliberty, 2022-07-12:
Its not a case of dependencies but of data loading into OR.  You are loading .lef & .def but not .lib.

braamBeresford, 2022-07-12:
Ok. Thank you for the swift response. I'll try and read in the libs as well.

vijayank88, 2023-06-19:
@braamBeresford 
Have you had chance to work on this and is that resolved?

maliberty, 2025-08-28:
no updates