The-OpenROAD-Project/OpenROAD #2066: `-guide` regression

donn, 2022-07-18:
Using OpenROAD 475ff5e827b8937f4810ecae75973ee74e457816 

`detailed_route -guide` is no longer loading the guide file properly. The reproducible outputs the following:

```
[...]
Number of guides:     0

[INFO DRT-0185] Post process initialize RPin region query.

Generating GCell with size = 15 tracks, using layer li1 pitch  = 0.46
li1 V Track-Pitch = 0.46000  line-2-Via Pitch = 0.34000
met1 H Track-Pitch = 0.34000  line-2-Via Pitch = 0.32500
met2 V Track-Pitch = 0.46000  line-2-Via Pitch = 0.34000
met3 H Track-Pitch = 0.68000  line-2-Via Pitch = 0.61000
met4 V Track-Pitch = 0.92000  line-2-Via Pitch = 0.61500
met5 H Track-Pitch = 3.40000  line-2-Via Pitch = 3.11000

initializing congestion map...

start routing resource analysis ...

             Routing   #Avail     #Track     #Total     %Gcell
Layer      Direction    Track    Blocked      Gcell    Blocked
--------------------------------------------------------------
li1             V        6348          0     215730     10.29%
met1            H       10353          0     215730     10.20%
met2            V        6348          0     215730     10.40%
met3            H        5176          0     215730      9.70%
met4            V        3174          0     215730      7.88%
met5            H        1035          0     215730     12.55%
--------------------------------------------------------------
Total                   32434      0.00%    1294380     10.17%

[INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 113.60 (MB), peak = 113.60 (MB)

generating net topology...
done net topology...
[INFO DRT-0032] FR_MASTERSLICE grObj region query size = 0.
[INFO DRT-0032] FR_VIA grObj region query size = 0.
[INFO DRT-0032] li1 grObj region query size = 1818.
[INFO DRT-0032] mcon grObj region query size = 0.
[INFO DRT-0032] met1 grObj region query size = 0.
[INFO DRT-0032] via grObj region query size = 0.
[INFO DRT-0032] met2 grObj region query size = 0.
[INFO DRT-0032] via2 grObj region query size = 0.
[INFO DRT-0032] met3 grObj region query size = 0.
[INFO DRT-0032] via3 grObj region query size = 0.
[INFO DRT-0032] met4 grObj region query size = 0.
[INFO DRT-0032] via4 grObj region query size = 0.
[INFO DRT-0032] met5 grObj region query size = 0.

start reporting 2D congestion ...

              #OverCon  %OverCon
Direction        GCell     GCell
--------------------------------
    H             478     0.22%
    V              85     0.04%
worstConH: 316.13%
worstConV: 112.50%

start 0th optimization iteration ...
[INFO DRT-0267] cpu time = 00:00:06, elapsed time = 00:00:06, memory = 132.32 (MB), peak = 139.60 (MB)

start 1st optimization iteration ...
[INFO DRT-0267] cpu time = 00:00:04, elapsed time = 00:00:04, memory = 132.95 (MB), peak = 139.60 (MB)

start 2nd optimization iteration ...
[INFO DRT-0267] cpu time = 00:00:05, elapsed time = 00:00:05, memory = 132.99 (MB), peak = 139.60 (MB)

start reporting 2D congestion ...

              #OverCon  %OverCon
Direction        GCell     GCell
--------------------------------
    H               4     0.00%
    V               0     0.00%
worstConH: 138.71%
worstConV:  81.08%

layer assignment...
done layer assignment...

start 0th optimization iteration ...
[INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 153.43 (MB), peak = 153.43 (MB)

start reporting 3D congestion ...

---------- li1 ----------
numOverConGCell: 0, %OverConGCell:   0.00%
---------- met1 ----------
numOverConGCell: 2, %OverConGCell:   0.00%
---------- met2 ----------
numOverConGCell: 0, %OverConGCell:   0.00%
---------- met3 ----------
numOverConGCell: 3, %OverConGCell:   0.00%
---------- met4 ----------
numOverConGCell: 77, %OverConGCell:   0.04%
---------- met5 ----------
numOverConGCell: 4, %OverConGCell:   0.00%

[WARNING DRT-0203] dbGcellGrid already exists in db. Clearing existing dbGCellGrid.
[ERROR DRT-0155] Guide in net io_in[25] uses layer met5 (12) that is outside the allowed routing range [met1 (4), (met4)].
Error: droute.tcl, 46 DRT-0155
```
Curiously, altering the invocation as follows makes it route fine:

```
read_guides $::env(CURRENT_GUIDE)

detailed_route\
    -bottom_routing_layer $min_layer\
    -top_routing_layer $max_layer\
    -output_guide $::env(TRITONROUTE_FILE_PREFIX).guide\
    -output_maze $::env(TRITONROUTE_FILE_PREFIX)_maze.log\
    -output_drc $::env(TRITONROUTE_RPT_PREFIX).drc\
    -droute_end_iter $::env(DRT_OPT_ITERS)\
    -or_seed 42\
    -verbose 1
```
Not really sure what's going on here. We're going to be altering the invocation for now for OpenLane.

[repro.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9133173/repro.tar.gz)

maliberty, 2022-07-18:
As I said in https://github.com/The-OpenROAD-Project/OpenLane/issues/1159 we are relying on the db more and side files less.  This is a case where the guides are now in the db in ORFS.  There is a read_guides command if you still want to operate in the DEF style (we kept it for ispd benchmarks).

donn, 2022-07-18:
Then remove the `-guide` option? What is it doing still being processed if it does nothing?

maliberty, 2022-07-18:
Fair enough - we'll remove it.

maliberty, 2022-07-18:
note that -output_guide is removed as well (it was never really useful to users anyways).

maliberty, 2022-07-18:
Removed in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2067

closed 2022-07-18 by maliberty