The-OpenROAD-Project/OpenROAD #2082: Error: grid_strategy-M1-M4-M7.tcl, 4 STA-0402.  Error of single Rocket core with Nangate45 PDK when running in OpenROAD flow scripts

JiaDYuan, 2022-07-22:
I set up a Single core Rocket, and run under the OpenROAD flow scripts with the platform of Nangate45.

But here is the error happened:

OpenROAD v2.0-2819-ge490a3f10
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR STA-0402] add_global_connection -defer_connection is not a known keyword or flag.
Error: grid_strategy-M1-M4-M7.tcl, 4 STA-0402
Command exited with non-zero status 1

why **add_global_connection -defer_connection** is unknown???????????????????????????? The error message is so limited, how can I deal with it?

vijayank88, 2022-07-22:
@JiaDYuan 
Please provide re-producible test case by running `make pdn_issue`

maliberty, 2022-07-22:
You are on a very old version of openroad.  Please update and try again.

JiaDYuan, 2022-07-23:
**Hi, yes, the updating the OpenROAD version can solve the problem. But here is another error, the process is stucking in the global_place_gp stage. And by the way, the process took 8 hours to run, is this normal???**

OpenROAD v2.0-4473-g0d17a7302
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO GPL-0002] DBU: 2000
[INFO GPL-0003] SiteSize: 380 2800
[INFO GPL-0004] CoreAreaLxLy: 40280 39200
[INFO GPL-0005] CoreAreaUxUy: 7278520 3158400
[INFO GPL-0006] NumInstances: 1111893
[INFO GPL-0007] NumPlaceInstances: 1092925
[INFO GPL-0008] NumFixedInstances: 18968
[INFO GPL-0009] NumDummyInstances: 0
[INFO GPL-0010] NumNets: 1435642
[INFO GPL-0011] NumPins: 4328596
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 7318800 3197600
[INFO GPL-0014] CoreAreaLxLy: 40280 39200
[INFO GPL-0015] CoreAreaUxUy: 7278520 3158400
[INFO GPL-0016] CoreArea: 22577518208000
[INFO GPL-0017] NonPlaceInstsArea: 20181952000
[INFO GPL-0018] PlaceInstsArea: 11477327568000
[INFO GPL-0019] Util(%): 50.88
[INFO GPL-0020] StdInstsArea: 11477327568000
[INFO GPL-0021] MacroInstsArea: 0
[InitialPlace]  Iter: 1 CG residual: 0.00355486 HPWL: 42509104118
[InitialPlace]  Iter: 2 CG residual: 0.00032160 HPWL: 13402372951
[InitialPlace]  Iter: 3 CG residual: 0.00022189 HPWL: 13400886018
[InitialPlace]  Iter: 4 CG residual: 0.00005148 HPWL: 13396827050
[InitialPlace]  Iter: 5 CG residual: 0.00005324 HPWL: 13381094381
[InitialPlace]  Iter: 6 CG residual: 0.00008841 HPWL: 13378887651
[InitialPlace]  Iter: 7 CG residual: 0.00004164 HPWL: 13378797984
[InitialPlace]  Iter: 8 CG residual: 0.00003799 HPWL: 13379558534
[InitialPlace]  Iter: 9 CG residual: 0.00003579 HPWL: 13379053762
[InitialPlace]  Iter: 10 CG residual: 0.00044856 HPWL: 13379823777
[InitialPlace]  Iter: 11 CG residual: 0.00009343 HPWL: 13394675825
[InitialPlace]  Iter: 12 CG residual: 0.00003725 HPWL: 13384452816
[InitialPlace]  Iter: 13 CG residual: 0.00003590 HPWL: 13381225665
[InitialPlace]  Iter: 14 CG residual: 0.00003757 HPWL: 13381738547
[InitialPlace]  Iter: 15 CG residual: 0.00004011 HPWL: 13381426155
[InitialPlace]  Iter: 16 CG residual: 0.00003597 HPWL: 13381785999
[InitialPlace]  Iter: 17 CG residual: 0.00003774 HPWL: 13381420292
[InitialPlace]  Iter: 18 CG residual: 0.00003420 HPWL: 13381623253
[InitialPlace]  Iter: 19 CG residual: 0.00005310 HPWL: 13381078982
[InitialPlace]  Iter: 20 CG residual: 0.00003638 HPWL: 13381806564
[INFO GPL-0031] FillerInit: NumGCells: 1095492
[INFO GPL-0032] FillerInit: NumGNets: 1435642
[INFO GPL-0033] FillerInit: NumGPins: 4328596
[INFO GPL-0023] TargetDensity: 0.51
[INFO GPL-0024] AveragePlaceInstArea: 10501477
[INFO GPL-0025] IdealBinArea: 20591132
[INFO GPL-0026] IdealBinCnt: 1096468
[INFO GPL-0027] TotalBinArea: 22577518208000
[INFO GPL-0028] BinCnt: 1024 1024
[INFO GPL-0029] BinSize: 7069 3047
[INFO GPL-0030] NumBins: 1048576
[NesterovSolve] Iter: 1 overflow: 0.999473 HPWL: 3202675283
[NesterovSolve] Iter: 10 overflow: 0.998919 HPWL: 3411390855
[NesterovSolve] Iter: 20 overflow: 0.998291 HPWL: 3613767813
[NesterovSolve] Iter: 30 overflow: 0.99765 HPWL: 3826664613
[NesterovSolve] Iter: 40 overflow: 0.997232 HPWL: 4017670899
[NesterovSolve] Iter: 50 overflow: 0.996924 HPWL: 4173727428
[NesterovSolve] Iter: 60 overflow: 0.996641 HPWL: 4286505525
[NesterovSolve] Iter: 70 overflow: 0.996444 HPWL: 4362062452
[NesterovSolve] Iter: 80 overflow: 0.996284 HPWL: 4410178858
[NesterovSolve] Iter: 90 overflow: 0.996171 HPWL: 4440513218
[NesterovSolve] Iter: 100 overflow: 0.996078 HPWL: 4459614153
[NesterovSolve] Iter: 110 overflow: 0.995993 HPWL: 4475941562
[NesterovSolve] Iter: 120 overflow: 0.995918 HPWL: 4494873625
[NesterovSolve] Iter: 130 overflow: 0.995833 HPWL: 4517053045
[... 52 lines cut ...]
[NesterovSolve] Iter: 610 overflow: 0.546942 HPWL: 33457960736
[NesterovSolve] Iter: 620 overflow: 0.518452 HPWL: 38168656059
[NesterovSolve] Iter: 630 overflow: 0.514085 HPWL: 37581479908
[INFO GPL-0100] worst slack -2.15e-09
[INFO GPL-0103] Weighted 143558 nets.
[NesterovSolve] Iter: 640 overflow: 0.470804 HPWL: 34397333619
[NesterovSolve] Iter: 650 overflow: 0.466778 HPWL: 35967999312
[NesterovSolve] Iter: 660 overflow: 0.415558 HPWL: 33672381009
[NesterovSolve] Iter: 670 overflow: 0.412021 HPWL: 33253342286
[NesterovSolve] Iter: 680 overflow: 0.368686 HPWL: 33280953716
[NesterovSolve] Iter: 690 overflow: 0.338376 HPWL: 31427691942
[NesterovSolve] Iter: 700 overflow: 0.326631 HPWL: 30068300281
[NesterovSolve] Iter: 710 overflow: 0.304877 HPWL: 29465430429
[INFO GPL-0100] worst slack -2.02e-09
[INFO GPL-0103] Weighted 143553 nets.
[NesterovSolve] Iter: 720 overflow: 0.280691 HPWL: 29228281359
[NesterovSolve] Iter: 730 overflow: 0.251856 HPWL: 29099461513
[NesterovSolve] Iter: 740 overflow: 0.226439 HPWL: 28926181559
[INFO GPL-0100] worst slack -2.22e-09
[INFO GPL-0103] Weighted 143560 nets.
[NesterovSolve] Iter: 750 overflow: 0.200645 HPWL: 28792313400
[INFO GPL-0075] Routability numCall: 1 inflationIterCnt: 1 bloatIterCnt: 0
Command terminated by signal 9
Elapsed time: 8:25:44[h:]min:sec. CPU time: user 29458.91 sys 719.07 (99%). Peak memory: 64012372KB.
make: *** [Makefile:486: results/nangate45/SingleRocket/base/3_3_place_gp.odb] Error 137

vijayank88, 2022-07-25:
@JiaDYuan 
How much RAM system has?
Can you increase swap memory and try it?

JiaDYuan, 2022-07-25:
@vijayank88 
[image]

Hi, I was running the flow on a server, here are the memory details. 

                total                  used          free                   shared     buff/cache    available
Mem:       115559420       2184372     58420228         428          54954820     111814128
Swap:       310235124       197936       310037188

vijayank88, 2022-07-25:
@JiaDYuan 
Run following command from `flow` directory and attach test case.
`make global_placement_issue`

JiaDYuan, 2022-07-25:
Here is the result. Strange...

yuanjiadong@Serveur-calcul:~/OpenROAD-flow-scripts/flow$ make global_placement_issue
[INFO][FLOW] Using platform directory ./platforms/nangate45
make: *** No rule to make target 'global_placement_issue'.  Stop.

vijayank88, 2022-07-25:
sorry. Its mistake. Try this: `make global_place_issue`

JiaDYuan, 2022-07-25:
@vijayank88 

yuanjiadong@Serveur-calcul:~/OpenROAD-flow-scripts/flow$ make global_place_issue
[INFO][FLOW] Using platform directory ./platforms/nangate45
# Creating run-me-gcd-nangate45-base.sh script
# Creating vars-gcd-nangate45-base.sh/tcl script
# remove variables starting with a dot
# remove non portable commands
# remove QT variable
# Archiving issue to global_place_gcd_nangate45_base_2022-07-25_10-59.tar.gz
tar: ./logs/nangate45/gcd/base: Warning: Cannot stat: No such file or directory
tar: ./objects/nangate45/gcd/base: Warning: Cannot stat: No such file or directory
tar: ./reports/nangate45/gcd/base: Warning: Cannot stat: No such file or directory
tar: ./results/nangate45/gcd/base: Warning: Cannot stat: No such file or directory

Sorry, I didn't run the gcd design. I was running a single core Rocket design with the nangate45 PDK.

vijayank88, 2022-07-25:
Try `DESIGN_CONFIG=./designs/nangate45/SingleRocket/config.mk make global_place_issue`

JiaDYuan, 2022-07-25:
@vijayank88 
yes, I succeeded getting the _global_place_SingleRocket_nangate45_base_2022-07-25_11-51.tar.gz_  file.

But it's 820MB heavy... file size is too big..

vijayank88, 2022-07-25:
@JiaDYuan 
Share `/designs/nangate45/SingleRocket/config.mk` `/designs/nangate45/SingleRocket/constraint.sdc` as well. Then only i can able to replicate your error

JiaDYuan, 2022-07-25:
@vijayank88

Here are config.mk, constraint.sdc, lib and lef files attached:
[config+constraint+lib+lef.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9180727/config%2Bconstraint%2Blib%2Blef.zip)

vijayank88, 2022-07-26:
@vvbandeira 
What's this error meant?
```
[INFO GPL-0075] Routability numCall: 1 inflationIterCnt: 1 bloatIterCnt: 0
Elapsed time: 5:35:57[h:]min:sec. CPU time: user 19597.61 sys 545.80 (99%). Peak memory: 57298472KB.
make: *** [results/nangate45/SingleRocket/base/3_3_place_gp.odb] Error 9
```

vvbandeira, 2022-07-26:
@vijayank88 this looks like an out-of-memory error.

vijayank88, 2022-07-27:
@vvbandeira 
Can you run this design at your end? 
Even at my end its got killed twice. Both absolute/relative die size.
Source files attached above.

vvbandeira, 2022-07-28:
Using the latest code looks like a density issue (see below).

```
OpenROAD v2.0-4526-gc525def45
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO GPL-0002] DBU: 2000
[INFO GPL-0003] SiteSize: 380 2800
[INFO GPL-0004] CoreAreaLxLy: 40280 39200
[INFO GPL-0005] CoreAreaUxUy: 7278520 3158400
[INFO GPL-0006] NumInstances: 1111893
[INFO GPL-0007] NumPlaceInstances: 1092925
[INFO GPL-0008] NumFixedInstances: 18968
[INFO GPL-0009] NumDummyInstances: 0
[INFO GPL-0010] NumNets: 1435642
[INFO GPL-0011] NumPins: 4328218
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 7318800 3197600
[INFO GPL-0014] CoreAreaLxLy: 40280 39200
[INFO GPL-0015] CoreAreaUxUy: 7278520 3158400
[INFO GPL-0016] CoreArea: 22577518208000
[INFO GPL-0017] NonPlaceInstsArea: 20181952000
[INFO GPL-0018] PlaceInstsArea: 11477327568000
[INFO GPL-0019] Util(%): 50.88
[INFO GPL-0020] StdInstsArea: 11477327568000
[INFO GPL-0021] MacroInstsArea: 0
[ERROR GPL-0302] Use a higher -density or re-floorplan with a larger core area.
Given target density: 0.30
Suggested target density: 0.51
Error: global_place_skip_io.tcl, 30 GPL-0302
Elapsed time: 0:06.84[h:]min:sec. CPU time: user 6.31 sys 0.52 (100%). Peak memory: 2721284KB.
make: *** [Makefile:471: results/nangate45/SingleRocket/base/3_1_place_gp_skip_io.odb] Error 1
```

Tough, after I changed to using the recommended density of `0.51`: I am running now with Valgrind to make sure is not a memory error. If it is not, we will need to investigate the GPL code, 129GB of RAM looks excessive.
```
[INFO GPL-0100] worst slack -2.02e-09
[INFO GPL-0103] Weighted 143553 nets.
[NesterovSolve] Iter: 720 overflow: 0.280691 HPWL: 29228281359
[NesterovSolve] Iter: 730 overflow: 0.251856 HPWL: 29099461513
[NesterovSolve] Iter: 740 overflow: 0.226439 HPWL: 28926181559
[INFO GPL-0100] worst slack -2.22e-09
[INFO GPL-0103] Weighted 143560 nets.
[NesterovSolve] Iter: 750 overflow: 0.200645 HPWL: 28792313400
[INFO GPL-0075] Routability numCall: 1 inflationIterCnt: 1 bloatIterCnt: 0
Elapsed time: 1:59:04[h:]min:sec. CPU time: user 7033.68 sys 106.78 (99%). Peak memory: 129120956KB.
make: *** [Makefile:485: results/nangate45/SingleRocket/base/3_3_place_gp.odb] Error 9
```

vijayank88, 2022-07-28:
@vvbandeira 
Sorry about that. I forgot to mention to update density value to `0.51`.
Yes maybe the reason for failure is RAM. I had only 128GB RAM in my GCP machine.

JiaDYuan, 2022-07-30:
@vijayank88 

Hi, may I ask why does it require so many RAM to run global placement? Even 128G is not enough? Is this case normal?

maliberty, 2022-07-30:
You have 1M placeable objects so this is quite a large block but that still seems excessive.  
@vijayank88 do you have this case setup to run?

vijayank88, 2022-07-31:
Hi Matt,

User shared RTL in the thread.

Vitor is trying i think.

vijayank88, 2022-08-01:
@vvbandeira 
Any update on this? Is able to pass global placement stage?

vvbandeira, 2022-08-01:
@vijayank88 I ran with Valgrind and found no memory error. It might be a problem with the internal logic. @maliberty suggestions? I have the design setup on my GCP machine in case you want to take a look let me know.

vijayank88, 2022-08-06:
I think its similar to https://github.com/The-OpenROAD-Project/OpenLane/issues/1126 Rocket mini chip code. 
as per @maliberty  suggestion `Dropping in a ram (dff or openram) would likely solve many of your issues`

vijayank88, 2023-06-19:
@JiaDYuan 
Have you tried Matt's suggestion to update the RTL?

closed 2025-08-28 by maliberty