The-OpenROAD-Project/OpenROAD #2089: GRT: SEGV when running STA after read_guides

antonblanchard, 2022-07-26:
I'm trying to make changes to the global routing guides and then run STA to see what effect it has. First up - I'm not sure if this is expected to work, so feel free to close if it is invalid.

My test case looks like:

```
read_liberty asap7sc7p5t_AO_RVT_FF_nldm_201020.lib      
read_liberty asap7sc7p5t_OA_RVT_FF_nldm_201020.lib   
read_liberty asap7sc7p5t_SIMPLE_RVT_FF_nldm_201020.lib
read_liberty asap7sc7p5t_INVBUF_RVT_FF_nldm_201020.lib  
read_liberty asap7sc7p5t_SEQ_RVT_FF_nldm_201020.lib  

read_lef asap7_tech_1x_201209.lef
read_lef asap7sc7p5t_27_R_1x_201211.lef  

read_def before.def

read_sdc 4_cts.sdc

source setRC.tcl

set_global_routing_layer_adjustment M2-M7 0.3
set_routing_layers -signal M2-M7

#global_route
read_guides route.guide

set_propagated_clock [all_clocks]
estimate_parasitics -global_routing

report_checks -path_delay max -fields {slew cap input nets fanout} -format full_clock_expanded
```

If I uncomment `global_route` this works (presumably because we have set up the required state), but otherwise it fails with:

```
Stack trace:
 0# 0x0000000000BB1B4E in openroad
 1# 0x00007F3F56DBD350 in /lib64/libc.so.6
 2# grt::MakeWireParasitics::estimateParasitcs(odb::dbNet*, std::vector<grt::Pin, std::allocator<grt::Pin> >&, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&) in openroad
 3# grt::GlobalRouter::estimateRC() in openroad
 4# rsz::Resizer::estimateParasitics(rsz::ParasiticsSrc) in openroad
```

I added this to `readGuides`, which fixed the SEGV but I get a bunch of errors about missing routes to pins:

```
    nets = initNetlist();
    initNets(nets);
```

At that point I wondered if we need to call `addRemainingGuides`, `connectPadPins` etc.

Test case:  [read_guides.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9185821/read_guides.tar.gz)

eder-matheus, 2022-08-16:
@antonblanchard Could you check the latest master branch? We've merged an update that should fix your issue.

antonblanchard, 2022-09-16:
Thanks @eder-matheus this has fixed my issue, and made it much easier to go back and look at STA differences between grt parasitics and SPEF parasitics.

closed 2022-09-16 by antonblanchard