The-OpenROAD-Project/OpenROAD #2091: Multiple port issue in OpenDB

Yupjun, 2022-07-27:
Hi, it seems like OpenDB can't parse multiple ports.
I'm currently using real Foundary PDK and this design contains a lot of multiple ports.
I got an error (DEFPARS-6010) while processing my DEF file.
Are there any options that I can do?

maliberty, 2022-07-27:
Do you get any other messages?  Can you reduce this to a minimal test case that you can make public (you don't need to make the LEF public, just a reduced DEF since it is a port issue).

Yupjun, 2022-07-27:
Before I share my content, sorry I'm a super newbie in this physical design area.

multiple ports of DEF file looks like this.

~~~
- VSS + NET VSS + SPECIAL +DIRECTION INOUT + USE GROUND
 + PORT 
     +LAYER D6 ( -188 0 ) (188 376 )
     +FIXED (5600 0) N
 + PORT 
     +LAYER D6 ( -188 0 ) (188 376 )
     +FIXED (5600 1490400 ) S
 + PORT 
     +LAYER D6 ( -188 0 ) (188 376 )
     +FIXED (9930 0 ) N
 + PORT 
     +LAYER D6 ( -188 0 ) (188 376 )
     +FIXED (9930 1490400 ) S
 + PORT 
     +LAYER D6 ( -188 0 ) (188 376 )
     +FIXED (14250 0 ) N
 + PORT 
     +LAYER D6 ( -188 0 ) (188 376 )
     +FIXED (14250 1490400 ) S
~~~
... and so on. I'm not sure why the Innovus tool exports the DEF file like this. Again, sorry for my lack of knowledge about VLSI physical design.

closed 2022-07-28 by Yupjun