The-OpenROAD-Project/OpenROAD #2106: terminate called after throwing an instance of 'std::length_error'

JiaDYuan, 2022-08-01:
Hi， I am failing the placement process of Single Rocket because of the possible memory problem:  #2082 

So I tried to run a simple one, here is the RTL:
[serial.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9231296/serial.zip)

And  the PDK is also nangate45:
[config&PDK.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9231307/config.PDK.zip)

But the synthesis failed beacause of a error, _"terminate called after throwing an instance of 'std::length_error'"_

Here is the log: 
[1_1_yosys.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/9231317/1_1_yosys.log)
BTW, the synthesis passed successfully in FPGA flow (Vivado) without any error.

What's more, I have an additional question. I tried to generate a FFT through Chipyard. But it seems like that only FFT **with a processor** (Rocket) can be generated. So can Chipyard generate RTLs of accelerators only without any processor or core?

vijayank88, 2022-08-01:
@vvbandeira FYI

vvbandeira, 2022-08-01:
This looks like an issue with Yosys and not OpenROAD.

vvbandeira, 2022-08-01:
@JiaDYuan are you using ORFS / OpenLane? I would suggest you try the latest version of Yosys to see if the problem got fixed --- ORFS and OpenLane usually are behind. If the problem persists, file an issue in the upstream repo: https://github.com/YosysHQ/yosys

JiaDYuan, 2022-08-07:
@vvbandeira 
Hi, I installed the newest Yosys:   **Yosys 0.20+1 (git sha1 60a787fa5, clang 11.0.1-2 -fPIC -Os)**

But the synthesis still failed with the same error:

7.23.2. Executing OPT_MERGE pass (detect identical cells).
terminate called after throwing an instance of 'std::length_error'
  what():  hash table exceeded maximum size. use a ILP64 abi for larger tables.
Command terminated by signal 6
Elapsed time: 30:35.81[h:]min:sec. CPU time: user 1801.48 sys 33.81 (99%). Peak memory: 19007064KB.
make: *** [Makefile:372: results/nangate45/serial/base/1_1_yosys.v] Error 134

maliberty, 2022-08-07:
The message "use a ILP64 abi" suggests you have compiled for 32bits.  

Check if your executable is 32 bit or 64 bit.  I did:
```
% file /workspaces/mliberty/w1/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
/workspaces/mliberty/w1/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys: ELF 64-bit LSB executable, x86-64, version 1 (GNU/Linux), dynamically linked (uses shared libs), for GNU/Linux 2.6.32, BuildID[sha1]=225a0e0281bd0a519b433d718ae65d5955961a94, not stripped
```
the "64-bit LSB executable" is what we are looking for.

JiaDYuan, 2022-08-07:
@maliberty 

Hi, I think it is 64-bit LSB executable:

`$ file ~/YOSYS/usr/local/bin/yosys
/home/yuanjiadong/YOSYS/usr/local/bin/yosys: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=2c59d312ae6c641d9ca693495f26f96d63670331, for GNU/Linux 3.2.0, not stripped
`

maliberty, 2022-08-08:
I have filed a yosys issue for you.  Note that the yosys team is separate so I have no control over whether if/when they choose to address this.

closed 2022-10-10 by vvbandeira