The-OpenROAD-Project/OpenROAD #2107: CMake Error: The following variables are used in this project, but they are set to NOTFOUND.

wjrforcyber, 2022-08-01:
I found this CMake error after I built (without #1922 (although I don't think they are relevant...))
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
TCL_LIBRARY 
    linked by target "openroad" in directory /home/Github/OpenROAD/OpenROAD/src
    linked by target "odb" in directory /home/Github/OpenROAD/OpenROAD/src/odb/src/db
    linked by target "TestCallBacks" in directory /home/Github/OpenROAD/OpenROAD/src/odb/test/cpp
    linked by target "TestGeom" in directory /home/Github/OpenROAD/OpenROAD/src/odb/test/cpp
    linked by target "TestModule" in directory /home/Github/OpenROAD/OpenROAD/src/odb/test/cpp
    linked by target "TestLef58Properties" in directory /home/Github/OpenROAD/OpenROAD/src/odb/test/cpp
    linked by target "TestGuide" in directory /home/Github/OpenROAD/OpenROAD/src/odb/test/cpp
    linked by target "TestJournal" in directory /home/Github/OpenROAD/OpenROAD/src/odb/test/cpp
    linked by target "TestGroup" in directory /home/Github/OpenROAD/OpenROAD/src/odb/test/cpp
    linked by target "TestGCellGrid" in directory /home/Github/OpenROAD/OpenROAD/src/odb/test/cpp
    linked by target "TestAccessPoint" in directory /home/Github/OpenROAD/OpenROAD/src/odb/test/cpp
    linked by target "odbpy" in directory /home/Github/OpenROAD/OpenROAD/src/odb/src/swig/python
    linked by target "odbtcl" in directory /home/Github/OpenROAD/OpenROAD/src/odb/src/swig/tcl
    linked by target "sta" in directory /home/Github/OpenROAD/OpenROAD/src/sta
    linked by target "TestDistributed" in directory /home/Github/OpenROAD/OpenROAD/src/dst/test/cpp
    linked by target "TestBalancer" in directory /home/Github/OpenROAD/OpenROAD/src/dst/test/cpp
    linked by target "TestWorker" in directory /home/Github/OpenROAD/OpenROAD/src/dst/test/cpp

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

vijayank88, 2022-08-01:
@wjrforcyber 
I don't understand your issue?
Please provide following:

1. What command you're tried and facing this issue or instruction followed?
2. OpenROAD version currently using?
3. OS related

wjrforcyber, 2022-08-01:
> @wjrforcyber I don't understand your issue? Please provide following:
> 
> 1. What command you're tried and facing this issue or instruction followed?
> 2. OpenROAD version currently using?
> 3. OS related

Sorry, I haven't wrote issue for a long time, here is the information:

1. I was following the readme and cloned master, in **Build by hand** section I tried `cmake`, and the error showed up.
2. May I ask how could I get the version of OpenROAD? 
4. MacOs

vijayank88, 2022-08-01:
Is https://github.com/The-OpenROAD-Project/OpenROAD#install-dependencies installed?
you've tried `cmake ..` right?

wjrforcyber, 2022-08-01:
> https://github.com/The-OpenROAD-Project/OpenROAD#install-dependencies

1. Yes, dependencies are installed except the qt issue  #1922 as I mentioned above.
2. Yes, tried `cmake ..`.

vijayank88, 2022-08-01:
Its a fresh installation or updating existing repo?

wjrforcyber, 2022-08-01:
> Its a fresh installation or updating existing repo?

Cloned last Friday.

vijayank88, 2022-08-01:
Try the following:
```
git checkout master
git pull
git submodule init
git submodule update
git submodule sync
git clean -xdi
mkdir build
cd build
cmake ..
make
```

wjrforcyber, 2022-08-01:
> > @wjrforcyber I don't understand your issue? Please provide following:
> > 
> > 1. What command you're tried and facing this issue or instruction followed?
> > 2. OpenROAD version currently using?
> > 3. OS related
> 
> Sorry, I haven't written issue for a long time, here is the information:
> 
> 1. I was following the readme and cloned master, in **Build by hand** section I tried `cmake`, and the error showed up.
> 2. May I ask how could I get the version of OpenROAD?
> 3. MacOs

Sorry, about 3, I used it on ubuntu in docker. @vijayank88

wjrforcyber, 2022-08-01:
> Try the following:
> 
> ```
> git checkout master
> git pull
> git submodule init
> git submodule update
> git submodule sync
> git clean -xdi
> mkdir build
> cd build
> cmake ..
> make
> ```

Tried, still the same error. I found that there a [similar one](https://github.com/The-OpenROAD-Project/OpenROAD/issues/136#issuecomment-606298703) in the issue history, don't know how they dealt with it.

vijayank88, 2022-08-01:
May i know the reason for OpenROAD developement branch installation?
If that ok, you can install https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts as stable flow.

vijayank88, 2022-08-01:
> > https://github.com/The-OpenROAD-Project/OpenROAD#install-dependencies
> 
>     1. Yes, dependencies are installed except the qt issue  [qt5-default package from DependencyInstaller.sh is deprecated #1922](https://github.com/The-OpenROAD-Project/OpenROAD/issues/1922) as I mentioned above.
> 
>     2. Yes, tried `cmake ..`.

have you ran both?
```
    ./etc/DependencyInstaller.sh -run
       ./etc/DependencyInstaller.sh -dev
```

wjrforcyber, 2022-08-01:
> have you ran both?
>
> ```
>     ./etc/DependencyInstaller.sh -run
>        ./etc/DependencyInstaller.sh -dev
> ```

Yes, the first one and the second one both showed the qt error, others are fine, I can even `apt search tcllib` and `apt search tcl-dev` and find them.
Just for study, I may try OpenROAD-flow-scripts later.

vvbandeira, 2022-08-01:
@wjrforcyber if you are using docker you can use `openroad/ubuntu20-dev` image which is validated to have all the dependencies to run the app.

If you are on macOS, which version of Cmake do you have? Also, setting the Tcl path can be tricky, did you install Tcl with brew? You can take a look at how we do for our CI [here](https://github.com/The-OpenROAD-Project/OpenROAD/blob/5ca26d885644fc725f9884cb5f73aa3aeca7c1ee/.github/workflows/github-actions-macos.yml#L36-L45)

vijayank88, 2022-08-09:
@wjrforcyber 
Are you able to resolve it now?

vvbandeira, 2022-10-03:
Closing as stale. Please re-open or file a new issue in case this is still relevant.

closed 2022-10-03 by vvbandeira