The-OpenROAD-Project/OpenROAD #2122: Build problem with abc

niv280, 2022-08-07:
Hi,
I am trying to compile and I gets errors:
 [54%] Linking CXX executable abc
cd /volPhysicalPath/albatross_backend_workArea/nive/OpenROAD/build/third-party/abc && /home/nive/bin/cmake -E cmake_link_script CMakeFiles/abc.dir/link.txt --verbose=1
/home/nive/bin/gcc -O3 -DNDEBUG -rdynamic CMakeFiles/abc.dir/src/base/main/main.c.o -o abc  libabc.a -lpthread -lm -ldl -lrt 
libabc.a(bmcMaj3.c.o): In function `Zyx_ManPrintSolutionFile':
bmcMaj3.c:(.text+0x193): undefined reference to `bmcg_sat_solver_read_cex_varvalue'
bmcMaj3.c:(.text+0x203): undefined reference to `bmcg_sat_solver_read_cex_varvalue'

From there there is a lot of 'undefined reference to ...' error and then:
giaQbf.c:(.text+0x8165): undefined reference to `bmcg_sat_solver_solve'
giaQbf.c:(.text+0x8203): undefined reference to `bmcg_sat_solver_read_cex_varvalue'
collect2: error: ld returned 1 exit status
make[2]: *** [third-party/abc/abc] Error 1
make[2]: Leaving directory `/volPhysicalPath/albatross_backend_workArea/nive/OpenROAD/build'
make[1]: *** [third-party/abc/CMakeFiles/abc.dir/all] Error 2
make[1]: Leaving directory `/volPhysicalPath/albatross_backend_workArea/nive/OpenROAD/build'
make: *** [all] Error 2

Can you give me direction to what this issue related to?

maliberty, 2022-08-08:
On what platform with what compiler?  This isn't enough information except to say that it builds fine with our CI.

vijayank88, 2022-08-09:
@maliberty 
Its better to keep issue template mandatory for all, based on installation/design issue.
So its easy to understand issue. 
If its ok i'll make a enhancement proposal.

maliberty, 2022-08-09:
@vijayank88 ok

niv280, 2022-08-10:
I think that as part of the problem is that I install most of the dependencies locally since i don't got root permission.
I think that what causing the make errors, I will try to install it in different environment. 

Thanks, and sorry for the unclear issue.

closed 2022-08-10 by niv280