The-OpenROAD-Project/OpenROAD #2126: pdngen segfault when using an `.odb` file as an input

donn, 2022-08-08:
* Using OpenROAD 273a9ac3a

```
OpenROAD v2.0-4585-g273a9ac3a 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO PDN-0001] Inserting grid: stdcell_grid
[WARNING PSM-0016] Voltage pad location (VSRC) file not specified, defaulting pad location to checkerboard pattern on core area.
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0018] Y direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0019] Voltage on net VPWR is not explicitly set.
Stack trace:
 0# 0x00005621B7FCAAF5 in openroad
 1# 0x00007F7C1440E8E0 in /usr/lib/libc.so.6
 2# sta::LibertyLibrary::defaultOperatingConditions() const in openroad
 3# psm::SupplyVoltage::getSupplyVoltage(sta::dbSta*) in openroad
 4# psm::IRSolver::readC4Data() in openroad
 5# psm::IRSolver::buildConnection() in openroad
 6# psm::PDNSim::check_connectivity() in openroad
 7# 0x00005621B8E28DF5 in openroad
 8# TclNRRunCallbacks in /usr/lib/libtcl.so
 9# 0x00007F7C1603BBDB in /usr/lib/libtcl.so
10# Tcl_EvalEx in /usr/lib/libtcl.so
11# Tcl_Eval in /usr/lib/libtcl.so
12# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
13# ord::tclAppInit(Tcl_Interp*) in openroad
14# Tcl_MainEx in /usr/lib/libtcl.so
15# main in openroad
16# 0x00007F7C143F9290 in /usr/lib/libc.so.6
17# __libc_start_main in /usr/lib/libc.so.6
18# _start in openroad
```

[repro.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9281447/repro.tar.gz)

maliberty, 2022-08-08:
The crash is from psm not pdn.  To calculate IR drop we need to know the supply voltage.  You can specify it with set_pdnsim_net_voltage.  If not specified it will try to take it from the Liberty.  In your case none is loaded.  It should crash and I'll fix that but you'll just get an error message instead.  You need to use one of the options to give the voltage.

donn, 2022-08-08:
Gotcha. Thank you.

PR #2128 (2022-08-08): psm: error if no Liberty loaded and voltage is unspecified by the user
Fixes #2126
commit cc4c925a (2022-08-08): psm: error if no Liberty loaded and voltage is unspecified by the user

Fixes #2126
closed 2022-08-08 by maliberty