The-OpenROAD-Project/OpenROAD #2135: Crash during `repair_antennas`

donn, 2022-08-10:
* OpenROAD 273a9ac3a

```
OpenROAD v2.0-4585-g273a9ac3a 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Setting signal min routing layer to: met1 and clock min routing layer to met1. 
[INFO]: Setting signal max routing layer to: met5 and clock max routing layer to met5. 
-congestion_iterations 50 -verbose
[INFO GRT-0020] Min routing layer: met1
[INFO GRT-0021] Max routing layer: met5
[INFO GRT-0022] Global adjustment: 30%
[INFO GRT-0023] Grid origin: (0, 0)
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer li1     Track-Pitch = 0.4600  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met1    Track-Pitch = 0.3400  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met2    Track-Pitch = 0.4600  line-2-Via Pitch: 0.3500
[INFO GRT-0088] Layer met3    Track-Pitch = 0.6800  line-2-Via Pitch: 0.6150
[INFO GRT-0088] Layer met4    Track-Pitch = 0.9200  line-2-Via Pitch: 1.0400
[INFO GRT-0088] Layer met5    Track-Pitch = 3.4000  line-2-Via Pitch: 3.1100
[INFO GRT-0019] Found 0 clock nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 65
[INFO GRT-0003] Macros: 0
[INFO GRT-0004] Blockages: 896

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical            0             0          0.00%
met1       Horizontal       4480          2626          41.38%
met2       Vertical         3360          2308          31.31%
met3       Horizontal       2240          1521          32.10%
met4       Vertical         1344           949          29.39%
met5       Horizontal        448           208          53.57%
---------------------------------------------------------------

[INFO GRT-0197] Via related to pin nodes: 1363
[INFO GRT-0198] Via related Steiner nodes: 18
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 1494
[INFO GRT-0112] Final usage 3D: 5428

[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
li1                  0             0            0.00%             0 /  0 /  0
met1              2626           372           14.17%             0 /  0 /  0
met2              2308           574           24.87%             0 /  0 /  0
met3              1521             0            0.00%             0 /  0 /  0
met4               949             0            0.00%             0 /  0 /  0
met5               208             0            0.00%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total             7612           946           12.43%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 11799 um
[INFO GRT-0014] Routed nets: 372
Error: groute.tcl, 41 TypeError in method 'sta_to_db_mterm', argument 1 of type 'LibertyPort *'
```

[repro.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9299551/repro.tar.gz)

eder-matheus, 2022-08-10:
This is crashing because when the code is looking for the diode specified, it returns a list of three diode ports (from the `sta::get_lib_pins` function). I see that the diode cell `sky130_fd_sc_hd__diode_2` is defined in three different liberty files. 

Are you reading all of them in this run?

eder-matheus, 2022-08-10:
@donn @maliberty Also, is it valid to have multiple definitions of the same cell in different liberty files and read them in the same run? If so, I think I have a fix for this issue, but I'm not sure if it's ok to have this scenario.

maliberty, 2022-08-10:
You could have multiple liberty files for different corners however there should only be one LEF.

```
    foreach lib $::env(LIB_SLOWEST) {
        read_liberty -corner ss $lib
    }
    foreach lib $typ {
        read_liberty -corner tt $lib
    }
    foreach lib $::env(LIB_FASTEST) {
        read_liberty -corner ff $lib
    }
```
is legal

closed 2022-08-11 by eder-matheus