The-OpenROAD-Project/OpenROAD #2149: OpenROAD GUI corrupts it's own memory

armleo, 2022-08-13:
More information: https://github.com/The-OpenROAD-Project/OpenLane/issues/1265

maliberty, 2022-08-13:
I've never seen this problem and have no clue.  I don't think there is anything I can do without a test case.  If they want support I suggest they contact @tspyrou for under NDA help through Precision Innovations.

arlpetergadfort, 2022-08-14:
Could this be related to docker? I haven't seen anything like that either. Have you tried outside of docker?

maliberty, 2022-08-14:
Or a bad X server

armleo, 2022-08-14:
> I've never seen this problem and have no clue. I don't think there is anything I can do without a test case. 

If you open the gui and just open and close random menues then you get this issue. There is no test case per se

> If they want support I suggest they contact @tspyrou for under NDA help through Precision Innovations.

I am able to reproduce it outside NDA enviornment. Posted the logs. See original thread.

> Could this be related to docker? I haven't seen anything like that either. Have you tried outside of docker?

No, I do not have OpenROAD installation outside of Docker.

> Or a bad X server

It might be three things: Bad GPU driver, Bad X server, Bad drivers inside Docker.

I still do not know how to approach it. But it seems that all of the machines are VirtualBox instances with installed 3D acceleration. Can anybody else install Ubuntu inside VirtualBox and attempt to reproduce it?

arlpetergadfort, 2022-08-16:
@maliberty since the related issue on OpenLANE was closed, should this one be closed as well?

closed 2022-08-16 by maliberty