The-OpenROAD-Project/OpenROAD #2154: About install OpenRoad problem

Elon-J, 2022-08-15:
When I try to install the OpenRoad into a new Machine, at the step: Build using support script './etc/Build.sh', I meet the problem 'collect2: error: ld returned 1 exit status' as below, How can I solve it? Thanks!
[image]

Elon-J, 2022-08-15:
Yeah, I have used it (including: ./etc/DependencyInstaller.sh -run and ./etc/DependencyInstaller.sh -dev); tclsh version is 8.6; OS is CentOS7.

vvbandeira, 2022-08-15:
@Elon-J can you run the following? If there is still an error, please upload the `build.log` file here.
```
./etc/Build.sh -clean 2>&1 | tee build.log
```

Elon-J, 2022-08-15:
Yeah, there is still an error; this is the build.log file.

[build.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/9338897/build.log)

vijayank88, 2022-08-16:
@Elon-J 
Try these steps: 
I am not facing any issues with Centos07 installation
```
source /opt/rh/devtoolset-8/enable
source /opt/rh/llvm-toolset-7.0/enable
git checkout master
git pull
git submodule init
git submodule update
git submodule sync
git clean -xdi
./etc/Build.sh
```

vvbandeira, 2022-08-16:
Looking at the log and comparing with one from a successful run (also on a Centos7) on the same commit the only difference between our setups are:

```
# yours
-- Boost  found.
-- Found Python3: /usr/local/lib/libpython3.8.a (found version "3.8.12") found components:  Development
-- Found FLEX: /usr/bin/flex (found version "2.5.37")
# my
-- Found Boost: /usr/local/lib/cmake/Boost-1.72.0/BoostConfig.cmake (found version "1.72.0")
-- Found Python3: /usr/include/python3.6m (found version "3.6.8") found components: Development Development.Module Development.Embed
-- Found FLEX: /usr/local/bin/flex (found version "2.6.4")
```

Tough I suspect these difference would not make any difference in compile. However, I also noticed that in your build there are no `.d` files or the `-MD -MF` compile flags -- these flags create the `.d` files which specify dependencies for the compiler. I have no clue why yours is behaving differently. You could try compiling with a single thread, but that might not solve the issue.

```
./etc/Build.sh -clean -threads=1 2>&1 | tee build-single.log
```

I do not see any issues with OpenROAD itself here, happy to try and help you but this looks like an issue with our setup/environment.

saicharan0112, 2022-08-16:
@vvbandeira @Elon-J I think it is something to do with the python version

saicharan0112, 2022-08-16:
I tried this on a docker container (centos:centos7). I have same Boost version `/usr/local/lib/cmake/Boost-1.76.0/` and same FLEX version `flex 2.5.37` as @Elon-J . But I have the python version same as @vvbandeira `/usr/include/python3.6m/`. I am able to build the binary `OpenROAD v2.0-4634-g3342b36`

saicharan0112, 2022-08-16:
And something that is notable is the conda package of the openroad.. it uses python3.7 unlike klayout conda package which uses python3.8. Try installing klayout from `conda install -c flaport klayout` after installing openroad package. You will see a message that openroad will be removed because of python version differences.

msaligane, 2022-08-16:
Thanks @saicharan0112 this should work. @Elon-J Please try it.

saicharan0112, 2022-08-16:
Here is the installation script for OpenFASoC flow which uses the conda installation approach. 
https://github.com/idea-fasoc/OpenFASOC/blob/main/Dependencies.sh
Klayout is installed from the .deb or .rpm packages based on the OS

Elon-J, 2022-08-16:
@saicharan0112, Thank you, Sai! Let me try it.

vvbandeira, 2022-10-07:
@Elon-J any updates? Were you able to fix the issue with the suggestion?

vijayank88, 2023-06-19:
@Elon-J 
Are you able to complete the installation?