The-OpenROAD-Project/OpenROAD #2157: `write_verilog -include_pwr_gnd` not working properly for buffers created by TritonCTS

donn, 2022-08-16:
Confession: I'm not sure if this is been a long-standing problem we've been working around before or if there's an oversight on my part, but:

* OpenROAD 273a9ac3a 
```
OpenROAD v2.0-4585-g273a9ac3a 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Setting RC values...
[INFO]: Configuring cts characterization...
[INFO]: Performing clock tree synthesis...
[INFO]: Looking for the following net(s): clk
[INFO]: Running Clock Tree Synthesis...
[INFO CTS-0049] Characterization buffer is: sky130_fd_sc_hd__clkbuf_8.
[INFO CTS-0038] Number of created patterns = 50000.
[INFO CTS-0038] Number of created patterns = 100000.
[INFO CTS-0039] Number of created patterns = 137808.
[INFO CTS-0084] Compiling LUT.
Min. len    Max. len    Min. cap    Max. cap    Min. slew   Max. slew
2           8           1           35          1           48          
[WARNING CTS-0043] 4752 wires are pure wire and no slew degradation.
TritonCTS forced slew degradation on these wires.
[INFO CTS-0046]     Number of wire segments: 137808.
[INFO CTS-0047]     Number of keys in characterization LUT: 1760.
[INFO CTS-0048]     Actual min input cap: 1.
[WARNING CTS-0083] No clock nets have been found.
[INFO CTS-0008] TritonCTS found 0 clock nets.
[WARNING CTS-0082] No valid clock nets in the design.
[INFO]: Repairing long wires on clock nets...
[INFO RSZ-0058] Using max wire length 3048um.
Writing OpenROAD database to ./out.odb...
Writing powered netlist to ./out.nl.v...
Writing timing constraints to ./results/cts/spm.sdc...
```

"Writing powered netlist is done via" `write_verilog -include_pwr_gnd $::env(SAVE_POWERED_NETLIST)`, but the problem is the netlist generated is as follows:

```verilog
 sky130_fd_sc_hd__buf_12 input1 (.A(clk),
    .X(net1));
 sky130_fd_sc_hd__clkbuf_1 input2 (.A(rst),
    .X(net2));
 sky130_fd_sc_hd__buf_1 input3 (.A(x[0]),
    .X(net3));
 sky130_fd_sc_hd__buf_1 input4 (.A(x[10]),
    .X(net4));
 sky130_fd_sc_hd__buf_1 input5 (.A(x[11]),
    .X(net5));
 sky130_fd_sc_hd__buf_1 input6 (.A(x[12]),
    .X(net6));
```

As you can see, those particular cells are lacking their power and ground ports. Again- I'm not sure if I'm simply missing something or if there's a bug here, but help would be appreciated regardless.
[repro.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9351753/repro.tar.gz)

donn, 2022-08-16:
Apologies-- I forgot the `make install`. Problem persists on OpenROAD 3342b3666.

maliberty, 2022-08-16:
input1 is already there in the input 11-spm.odb and is not generated by CTS.  They would come from "buffer_ports -inputs" in resizer.tcl

maliberty, 2022-08-16:
Please provide a test case from the correct step

arlpetergadfort, 2022-08-16:
This looks like the new instances are not getting connected to the power rails, a call to global_connect (if set up correctly) would solve that. But we probably need to either ensure the new instances are connected correctly, we provide a mechanism to extract the power/ground connectivity, or the information in global connect is saved to the database so it can get called multiple times after the initial connections.

maliberty, 2022-08-16:
@arlpetergadfort for the near term we could use the kinda lame isWildConnected/setWildConnected that odb has already.  Once we have UPF this will not be sufficient

maliberty, 2022-08-17:
@ahmadelrouby please consider this in your UPF work

donn, 2022-08-24:
Updated the issue in question- sorry it took me so long

maliberty, 2022-08-28:
> Updated the issue in question- sorry it took me so long

What do you mean by this?  Does the link now point at a different reproducer?

donn, 2022-08-28:
There's a dropbox link in the submission for the new reproducer.

donn, 2022-08-31:
Just wanna mention- I'm blocked on this for the ODB migration

maliberty, 2022-08-31:
I'll give it some attention

maliberty, 2022-08-31:
Never mind about the global_connect, it only applies if you use -defer

vijayank88, 2023-06-19:
@donn 
Is that resolved?

donn, 2024-09-30:
This should be fine now, yeah

closed 2024-09-30 by donn