The-OpenROAD-Project/OpenROAD #2163: GPL stacktrace with empty design

proppy, 2022-08-17:
It seems that sending an empty design thru OpenROAD flow script:
```
module noop(clk);
  input clk;
  wire clk;
endmodule
```
makes GPL error out with a stacktrace:
```
Stack trace:
 0# 0x0000000000A86976 in /opt/conda/envs/silicon/bin/openroad
 1# 0x00007F09253A1090 in /lib/x86_64-linux-gnu/libc.so.6
 2# gpl::NesterovBase::initFillerGCells() in /opt/conda/envs/silicon/bin/openroad
 3# gpl::NesterovBase::init() in /opt/conda/envs/silicon/bin/openroad
 4# gpl::NesterovBase::NesterovBase(gpl::NesterovBaseVars, std::shared_ptr<gpl::PlacerBase>, utl::Logger*) in /opt/conda/envs/silicon/bin/openroad
 5# gpl::Replace::initNesterovPlace() in /opt/conda/envs/silicon/bin/openroad
 6# gpl::Replace::doNesterovPlace(int) in /opt/conda/envs/silicon/bin/openroad
 7# 0x0000000000ADA84B in /opt/conda/envs/silicon/bin/openroad
 8# TclNRRunCallbacks in /opt/conda/envs/silicon/bin/../lib/libtcl8.6.so
 9# 0x00007F0925D061BF in /opt/conda/envs/silicon/bin/../lib/libtcl8.6.so
10# Tcl_EvalEx in /opt/conda/envs/silicon/bin/../lib/libtcl8.6.so
11# Tcl_Eval in /opt/conda/envs/silicon/bin/../lib/libtcl8.6.so
12# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /opt/conda/envs/silicon/bin/openroad
13# ord::tclAppInit(Tcl_Interp*) in /opt/conda/envs/silicon/bin/openroad
14# Tcl_MainEx in /opt/conda/envs/silicon/bin/../lib/libtcl8.6.so
15# main in /opt/conda/envs/silicon/bin/openroad
16# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
17# 0x0000000000A80C02 in /opt/conda/envs/silicon/bin/openroad
```
I realize this is a limit case, but I'm curious if that's the expected behavior.
[global_place_noop_asap7_base_2022-08-17_12-07.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9359367/global_place_noop_asap7_base_2022-08-17_12-07.tar.gz)

maliberty, 2022-08-17:
crashes are never expected

maliberty, 2022-08-17:
I don't know how you generated the packaging of this test case but it is not useable.  

Error: read_liberty.tcl, 17 cannot read file ./platforms/asap7/lib/asap7sc7p5t_AO_RVT_FF_nldm_201020.lib.

There is no platforms directory.  I tried linking one and then I get

[ERROR ORD-0007] ./results/asap7/noop/base/3_2_place_iop.odb does not exist.

Please package a useable test case using make issue.

proppy, 2022-08-18:
I had to pass `EXCLUDE_PLATFORM=1 otherwise the `make global_place_issue` fails with:
```
[INFO][FLOW] Using platform directory ./platforms/asap7
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
# Creating run-me-noop-asap7-base.sh script
# Creating vars-noop-asap7-base.sh/tcl script
# remove variables starting with a dot
# remove non portable commands
# remove QT variable
# Archiving issue to global_place_noop_asap7_base_2022-08-18_02-29.tar.gz
tar: ./platforms/asap7/*.cfg: Warning: Cannot stat: No such file or directory
```
should I file a different bug about that?

vvbandeira, 2022-08-18:
> I had to pass `EXCLUDE_PLATFORM=1` otherwise the `make global_place_issue` command fails with:
> 
> ```
> [INFO][FLOW] Using platform directory ./platforms/asap7
> [INFO-FLOW] ASU ASAP7 - version 2
> Default PVT selection: BC
> # Creating run-me-noop-asap7-base.sh script
> # Creating vars-noop-asap7-base.sh/tcl script
> # remove variables starting with a dot
> # remove non portable commands
> # remove QT variable
> # Archiving issue to global_place_noop_asap7_base_2022-08-18_02-29.tar.gz
> tar: ./platforms/asap7/*.cfg: Warning: Cannot stat: No such file or directory
> ```
> 
> should I file a different bug about that?

Please do it in the ORFS repo so I can keep track, I will take a look as soon as I am able. Thanks.

proppy, 2022-08-18:
> Please do it in the ORFS repo so I can keep track, I will take a look as soon as I am able. Thanks.

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/561

vijayank88, 2023-06-19:
@proppy 
Unable to check as asap7 library got updated to `211120.lib.gz`. 
If still there is a problem can you attach latest reproducible.?

closed 2025-08-30 by maliberty