The-OpenROAD-Project/OpenROAD #2173: Add MacOS to dependency installer

rovinski, 2022-08-19:
It looks like we have the steps to install dependencies on MacOS in a GitHub action:

https://github.com/The-OpenROAD-Project/OpenROAD/blob/f0685da2f7304ba8d0871377086486522990485a/.github/workflows/github-actions-macos.yml#L8-L45

Is it possible to move this to [`etc/DependencyInstaller.sh`](https://github.com/The-OpenROAD-Project/OpenROAD/blob/f0685da2f7304ba8d0871377086486522990485a/etc/DependencyInstaller.sh)?

I don't have immediate access to MacOS, but I can probably do so soon for a PR.

vvbandeira, 2022-08-19:
Sound good to me, it would be just the Install lemon (which can probably be reused from the current code in the installer) and the brew dependencies. The exports would be nice to print at the end as an action item for the user to set or add to their bashrc -- I would avoid trying to make the modification as part of the installer.

donn, 2022-08-22:
*whispers* https://github.com/donn/homebrew-lemon-graph

rovinski, 2022-08-22:
@donn Any plans to upstream? :)

donn, 2022-08-24:
What, and deal with Homebrew's arcane requirements for package notability? Nah, I'm good.

rovinski, 2022-08-24:
It looks like most requirements for upstream are manageable except for ["be maintained (i.e. the last release wasn’t ages ago"](https://docs.brew.sh/Acceptable-Formulae) which we have no control over. Maybe someday someone can make an effort (there are still things being committed on the dev branch?)

@maliberty can we fork @donn's repo and use it during the build process?

rovinski, 2022-08-30:
@vvbandeira can you elaborate on how your MacOS build is set up? I tried manually installing but I run into issues related to #2097 

The latest `spdlog` available from homebrew is `1.10.0_1` and `fmt` is `9.1.0`. I have the following combinations of spdlog and fmt, and none of them worked:

| `spdlog`    |  `fmt`  |
|------------|--------|
| `1.10.0_1` | `9.1.0` |
| `1.10.0_1` | `8.1.1` |
| `1.9.2`      | `9.1.0` |
| `1.9.2`      | `8.1.1` |

Most of the compile errors occur inside the `fmt` headers such as
```
/usr/include/fmt/format.h:2493:38: error: expected unqualified-id
template <typename T> constexpr bool isnan(T value) {
```

donn, 2022-08-30:
+1 - I thought this was just me. I also had trouble compiling OpenROAD for Mac recently but I assumed something was on my side because the CI worked and so I gave up.

vvbandeira, 2022-08-30:
Below is our working setup. On my machine, I installed the dependencies when they worked and never updated them by using `brew pin`. Eder recently broke his setup with a `brew upgrade` and reported that he was able to get the correct versions following this: https://remarkablemark.org/blog/2017/02/03/install-brew-package-version/

```
% brew info fmt
fmt: stable 8.1.1 (bottled), HEAD
Open-source formatting library for C++
https://fmt.dev/
/opt/homebrew/Cellar/fmt/8.1.1_1 (27 files, 1MB) *
  Poured from bottle on 2022-03-14 at 17:47:37
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/fmt.rb
License: MIT
==> Dependencies
Build: cmake ✘
==> Options
--HEAD
        Install HEAD version
==> Analytics
install: 68,721 (30 days), 140,452 (90 days), 608,689 (365 days)
install-on-request: 1,967 (30 days), 2,941 (90 days), 9,449 (365 days)
build-error: 2 (30 days)

% brew info spdlog
spdlog: stable 1.10.0 (bottled), HEAD
Super fast C++ logging library
https://github.com/gabime/spdlog
/opt/homebrew/Cellar/spdlog/1.9.2 (95 files, 1.4MB) *
  Poured from bottle on 2022-03-14 at 17:47:38
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/spdlog.rb
License: MIT
==> Dependencies
Build: cmake ✘
Required: fmt ✔
==> Options
--HEAD
        Install HEAD version
==> Analytics
install: 4,343 (30 days), 10,018 (90 days), 24,547 (365 days)
install-on-request: 963 (30 days), 2,186 (90 days), 7,365 (365 days)
build-error: 1 (30 days)
```

vvbandeira, 2022-08-30:
@eder-matheus can you confirm that the versions above are the same you installed? If not, please provide the version combination that worked for you. Thanks

eder-matheus, 2022-08-30:
@vvbandeira I have the same fmt version, but a slightly different spdlog version. Check below:
```
%brew info spdlog
==> spdlog: stable 1.10.0 (bottled), HEAD
Super fast C++ logging library
https://github.com/gabime/spdlog
/usr/local/Cellar/spdlog/1.10.0 (99 files, 1.5MB) *
  Poured from bottle on 2022-08-22 at 19:15:34
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/spdlog.rb
License: MIT
==> Dependencies
Build: cmake ✔
Required: fmt ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 4,343 (30 days), 10,018 (90 days), 24,547 (365 days)
install-on-request: 963 (30 days), 2,186 (90 days), 7,365 (365 days)
build-error: 1 (30 days)
```

rovinski, 2022-08-30:
I have already tried both of those combinations and they didn't work.

What compiler versions are you using? I have `AppleClang 13.1.6.13160021`

rovinski, 2022-08-30:
Also @vvbandeira why is your `spdlog` version `1.10.0` but it is installed at `/opt/homebrew/Cellar/spdlog/1.9.2`?

> ```
> spdlog: stable 1.10.0 (bottled), HEAD
> Super fast C++ logging library
> https://github.com/gabime/spdlog
> /opt/homebrew/Cellar/spdlog/1.9.2 (95 files, 1.4MB) *
> ```

eder-matheus, 2022-08-30:
> I have already tried both of those combinations and they didn't work.
> 
> What compiler versions are you using? I have `AppleClang 13.1.6.13160021`

@rovinski  Here's my clang version:
```
Apple clang version 13.1.6 (clang-1316.0.21.2.5)
Target: x86_64-apple-darwin21.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

rovinski, 2022-08-30:
Mine is the same as yours (except as part of Xcode). I have tried simply installing spdlog from source and that doesn't work either. 
I've compared build logs from my failing build and the passing CI and I can't find any differences in environment...

rovinski, 2022-09-02:
For posterity, I finally got OpenROAD to compile on MacOS using `fmt 8.1.1` and `spdlog 1.10.0`. The issue was that my Xcode installation was messed up somehow. I was getting errors about `std::` functions not being found, which indicates there was some glibc problem. I uninstalled and reinstalled Xcode (`Apple clang version 13.1.6 (clang-1316.0.21.2.5)`) and it compiled all the way.

I can now continue working on the dependency script 👍

PR #2240 (2022-09-09): Update DependencyInstaller.sh to support MacOS
Draft PR to support MacOS. Closes #2173.

This isn't a perfect solution to the problem, but I think it's a good start. There are several issues I worked around while making this.

* Problem: Homebrew's `spdlog` does not work out of the box as of right now.
* Cause: This has been mentioned a few times in issues such as #2173, #2220, and #2097, but the main issue seems to be that `spdlog` is dependent on `fmt`. While `fmt 8` works with OpenROAD's code, `fmt 9` has more strict standards for conversions of objects into strings for logging, thus result [... cut]
closed 2022-09-09 by rovinski