The-OpenROAD-Project/OpenROAD #2175: timing analysis and heirchical blocks

kareefardi, 2022-08-20:
this isn't a question it's merely an inquiry about openroad capabilities vs standalone opensta:

suppose that we have block `A`. block `A` has an instance of block `B` and block `B` has an instance of block `C`. so in short `A->B->C`.

in opensta i would read std cell lib/liberty files followed by verilog views of all the blocks. i can also read a spef file of block `C` using `read_spef -path ..`. and i am capable of viewing nets and instance inside block `B` and `C`.

my question is, if possible, how can i achieve the same thing using openroad given that i have def, lef, verilog views for all said blocks?

QuantamHD, 2022-08-20:
OpenSTA works on a kind of interface model. It allows downstream users of its library to replace its internal "database" with its host's "database". In OpenROAD's case this is OpenDB, and its a great system except for the fact that OpenDB doesn't support hierarchy.

Since it's data model is only a thin wrapper around DEF it chooses to flatten everything into a single level, thus your `A -> B -> C` turns into just `A`. All the things inside B and C become part of A. Which is what I suspect you're running into.

kareefardi, 2022-08-20:
thanks that's what i was suspecting as well. so for full static timing analysis for designs with multiple levels of hierarchy one should avoid OpenROAD and use OpenSTA?

maliberty, 2022-08-21:
We do have write_liberty which lets you generate a basic timing model for hierarchical rollup.  Otherwise you would use opensta as you describe.

In the future we will support hierarchy in a similar fashion.

closed 2022-08-22 by kareefardi