The-OpenROAD-Project/OpenROAD #2182: ImportError: libtcl8.5.so: cannot open shared object file: No such file or directory

gbrown40, 2022-08-22:
When trying to run files that use opendbpy, I get this error:
`Traceback (most recent call last):
  File "src/odb/test/python/08-write_lef_and_def_test.py", line 1, in <module>
    import opendbpy as odb
  File "/usr/local/lib/python3.8/dist-packages/opendbpy/__init__.py", line 1, in <module>
    from ._opendbpy import *
ImportError: libtcl8.5.so: cannot open shared object file: No such file or directory`
libtcl8.5 appears to be obsolute, so I cannot install it, and installing libtcl8.6 does not fix the issue.

maliberty, 2022-08-22:
Did you build OR youself?  If so, you must have it installed or you couldn't have linked OR in the first place.

gbrown40, 2022-08-22:
I have built OR successfully both by hand and with the support script.

maliberty, 2022-08-22:
Then your LD_LIBRARY_PATH needs to point to the location of your tcl library.

gbrown40, 2022-08-22:
I don't think the tcl library is installed. At least I cannot find it in `/usr/local/lib` or anywhere else.

maliberty, 2022-08-22:
You couldn't have built OR if it wasn't installed.  When you run cmake do you see a line like:

-- TCL library: /usr/lib64/libtcl.so

gbrown40, 2022-08-22:
Found it. I added it to the path, but am still getting the same error. I think because the version of libtcl installed when building is 3.6.

maliberty, 2022-08-22:
The cmake line above should tell you which one is being picked up.  You can also use ldd on the executable to see what it is expecting.

gbrown40, 2022-08-22:
When running cmake, it lists TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so, but looking in /usr/lib/x86_64-linux-gn it lists libtcl.so and libtcl8.6.so. Running ldd and the executable shows "libtcl8.5.so => not found".

maliberty, 2022-08-22:
When it links openroad you can see the -L and -l it uses for tcl.  What do you see?

gbrown40, 2022-08-22:
I don't see -L or -l anywhere in the link process.

maliberty, 2022-08-22:
Do you see a line with "-o openroad"?  That is the link line, look for a mention of *libtcl*

gbrown40, 2022-08-22:
It lists /usr/lib/x86_64-linux-gnu/libtcl.so

maliberty, 2022-08-22:
Are you running ldd on the same binary?  That's what it is linked with...

gbrown40, 2022-08-22:
Running ldd for the openroad executable shows `libtcl8.6.so => /usr/lib/x86_64-linux-gnu/libtcl8.6.so`. The issue is with opendbpy which appears to want libtcl8.5.so

maliberty, 2022-08-22:
What commit of OR are you using?

vvbandeira, 2022-08-22:
```
/usr/local/lib/python3.8/dist-packages/opendbpy/__init__.py
```
@gbrown40 did you install `opendbpy` with `pip install` or similar? This looks to be getting a system version of `opendbpy` not the version you built from the OR source code.

gbrown40, 2022-08-25:
@vvbandeira I used pip to install opendbpy

gbrown40, 2022-08-25:
@maliberty I currently have commit 49fdec96d3cefaa9e5c3b39b6dd1e33bcc7f0d24 from August 3.

vvbandeira, 2022-10-17:
Hi @gbrown40, could you try with the latest commit. We recently merged #2375 which might help your issue. Also, you probably should remove the pip version since is very old -- it has not been updated in over a year.

vvbandeira, 2022-11-28:
Closing due to lack of activity. Please open a new issue in case the problem is still relevant and occurring with the latest commit.

closed 2022-11-28 by vvbandeira