The-OpenROAD-Project/OpenROAD #2183: pdngen: is there a way to short the VSS of two voltage domains?

luccareinehr, 2022-08-22:
I have a design from OpenFASOC with two voltage domains (CORE and TEMP_ANALOG), and I'm trying to generate its power delivery network with the new tcl commands. When generating their voltage rings, both ground tracks are completely isolated (i.e. they're not being connected):

[image]

Preferably, what I would like to achieve is for a metal strap to connect them together, like so (pardon my dreadful drawing):

[image]

Is there a way to do this in pdngen?

### To reproduce the case:

Download [reproducible_pdngen.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9396456/reproducible_pdngen.tar.gz) and source `script.tcl` in openroad (or openroad gui)

I'm on OpenROAD commit no. 5ca26d8 from Aug 4 2022.

arlpetergadfort, 2022-08-23:
@maliberty @luccareinehr looking at it, it appears there is no way to connect the two (that would be an oversight).
There are two things here:
1. need be able to connect two two grids (I think I know how to handle that)
2. At present, PDNGEN will not put straps into other grids, so for how we would need to move the met5 part of the ring down to met3 until that is fixed. This I think would require changing the order of the builds in the grids, so that will probably take longer to fix since it can come with lots of unintended changes.

luccareinehr, 2022-08-23:
@arlpetergadfort Can you explain how to connect the two grids?

By the way, thanks for the help

arlpetergadfort, 2022-08-23:
@luccareinehr I initially thought I would need to modify the code.

If I use the script below I can connect VSS between the two grids. The key here (and this could be a future improvement for PDNGEN) is to move the top layer of the ANALOG_TEMP grid to met4, so the CORE grid can place straps on met5 over that area. The second item needed is to tell the ANALOG_TEMP grip to connect from met4 to met5. Please try this out and let me know if there is anything else.

```
# Read odb file
read_db design_before_pdn.odb

# Add global connections
add_global_connection -net VDD -inst_pattern {temp_analog_1.*} -pin_pattern VPWR -power
add_global_connection -net VDD -inst_pattern {temp_analog_1.*} -pin_pattern VPB
add_global_connection -net VIN -inst_pattern {temp_analog_0.*} -pin_pattern VPWR -power
add_global_connection -net VIN -inst_pattern {temp_analog_0.*} -pin_pattern VPB
add_global_connection -net VSS -inst_pattern {.*} -pin_pattern VGND -ground
add_global_connection -net VSS -inst_pattern {.*} -pin_pattern VNB

# Set voltage domains
# TEMP_ANALOG region created with the create_voltage_domain command
set_voltage_domain -name CORE -power VDD -ground VSS
set_voltage_domain -region TEMP_ANALOG -power VIN -ground VSS

# Standard cell grids
define_pdn_grid -name stdcell -pins met5 -starts_with POWER -voltage_domains CORE

add_pdn_stripe -grid stdcell -layer met1 -width 0.49 -pitch 6.66 -offset 0 -extend_to_core_ring -followpins
add_pdn_ring -grid stdcell -layer {met4 met5} -widths {5.0 5.0} -spacings {2.0 2.0} -core_offsets {2.0 2.0}
add_pdn_stripe -grid stdcell -layer met4 -width 1.2 -pitch 56.0 -offset 2 -extend_to_core_ring
add_pdn_stripe -grid stdcell -layer met5 -width 1.6 -pitch 56.0 -offset 2 -extend_to_core_ring

add_pdn_connect -grid stdcell -layers {met4 met5}
add_pdn_connect -grid stdcell -layers {met1 met4}

define_pdn_grid -name stdcell_analog -pins met4 -starts_with POWER -voltage_domains TEMP_ANALOG

add_pdn_stripe -grid stdcell_analog -layer met1 -width 0.49 -pitch 6.66 -offset 0 -extend_to_core_ring -followpins
add_pdn_ring -grid stdcell_analog -layer {met4 met3} -widths {5.0 5.0} -spacings {2.0 2.0} -core_offsets {2.0 2.0}
add_pdn_stripe -grid stdcell_analog -layer met4 -width 1.2 -pitch 56.0 -offset 2 -extend_to_core_ring

add_pdn_connect -grid stdcell_analog -layers {met4 met3}
add_pdn_connect -grid stdcell_analog -layers {met1 met4}
add_pdn_connect -grid stdcell_analog -layers {met4 met5}

# Run pdngen
pdngen

# Save resulting design
write_db design_after_pdn.odb
```

luccareinehr, 2022-08-26:
@arlpetergadfort That gives a nice result, thank you! 

May I ask why pdngen works this way? Does it check for intercepting geometries of the same layer when connecting straps to a ring? So, for example, if a ring and a strap are both made of met4, they wouldn't get connected (like was happening in the original case)

arlpetergadfort, 2022-08-26:
the macro grids generate a keepout area around them which is what prevents the met5 strap if you have the ring in met5. Otherwise, yes, it connects on the overlapping shapes based on the connect layers defined.

closed 2022-08-26 by luccareinehr