The-OpenROAD-Project/OpenROAD #2187: ant: OpenROAD and magic disagree on antenna violations

antonblanchard, 2022-08-23:
Using the following test case OpenROAD reports no antenna violations, but magic reports some quite large ones. An example:

```
Cell: fanout1050
Antenna violation detected at plane metal3
Effective antenna ratio 1584.67 > limit 400
Gate rect (15531 76759) to (15561 76887)
Antenna rect (36813 76739) to (36879 76742)
```

I'm not sure if OpenROAD or magic is wrong.

Test case: [antenna.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9398600/antenna.tar.gz)

maliberty, 2022-08-23:
Anton noticed a drop in antenna violations with your recent changes. Please investigate

antonblanchard, 2022-08-23:
I don't see any `THICKNESS` properties in the TLEF file. Could be a bug in open_pdks. Investigating.

maliberty, 2022-08-23:
They are there in sky130A.

antonblanchard, 2022-08-23:
Recent shuttles are using sky130B, and that's where the issue is:

https://github.com/RTimothyEdwards/open_pdks/pull/272

ramezmoussa, 2022-08-23:
@maliberty is there a defualt `THICKNESS` that should be used in cases like this? The current implementation uses 0 for the thickness and I can confirm that this is why no violations were reported. I experimented with various values for the `THICKNESS` and multiple violations arose.

antonblanchard, 2022-08-23:
Since this was a problem in the LEF file, one option might be to just return an error. A default value could mask the issue.

antonblanchard, 2022-08-23:
After fixing the tech LEF `THICKNESS` issues, I still see some rather large differences. In this example magic calculates the antenna violation to be about twice as much as OpenROAD:

```
magic:
Cell: final_adder.U$$49
Antenna violation detected at plane metal2
Effective antenna ratio 1479.6 > limit 400
```

```
OpenROAD:
  final_adder.U$$49/A (sky130_fd_sc_hd__ha_1)
    met2
    PAR:  759.75* Ratio:  400.00 (S.Area)
```

@rtimothyedwards I think there may be an error in how magic calculates antenna violations, at least if we want to match
the tech LEF. The antenna rule is:

```
  ANTENNADIFFSIDEAREARATIO PWL ( ( 0 400 ) ( 0.0125 400 ) ( 0.0225 2609 ) ( 22.5 11600 ) ) ;
```

Which I think means we just sum up the side wall area of the layer in question, ignoring layers below it. Magic appears to sum up all connected layers, scaled by the antenna ratio limit for that layer:

```
                    for (i = 0; i < DBNumTypes; i++)
                    {
                        if (ExtCurStyle->exts_antennaRatio[i].ratioGate > 0)
                        {
                            anttotal += (double)antennaarea[i] /
                                    (double)ExtCurStyle->exts_antennaRatio[i].ratioGate;
                        }
                        if (ExtCurStyle->exts_antennaRatio[i].ratioGate > saveRatio)
                            saveRatio = ExtCurStyle->exts_antennaRatio[i].ratioGate;
                    }
```

The LEF spec was somewhat confusing to read through, so I might have misunderstood the `ANTENNADIFFSIDEAREARATIO` statement.

RTimothyEdwards, 2022-08-23:
The magic technology file specifies the antenna calculation method as "partial", which is supposed to check only the layer under investigation and not all layers below it (which is method "cumulative").  Your result suggests that the method type in the tech file is not being handled correctly, perhaps.

maliberty, 2022-08-23:
@ramezmoussa let's make it an error to not have THICKNESS specified as otherwise the analysis is meaningless.

antonblanchard, 2022-08-26:
I think we can close this now:
- The PDK issue (missing `THICKNESS`) has now been fixed
- OpenROAD now errors out when `THICKNESS` is missing or zero
- Magic has been fixed to closer match the OpenROAD antenna checker results

closed 2022-08-26 by antonblanchard