The-OpenROAD-Project/OpenROAD #2189: OpenROAD GPU Compile

monibahmed, 2022-08-23:
Hi, 

Is there a recommended way to compile for GPU? CMake doesn't seem to find the correct libraries/executables

**Commands Used**
sudo ./etc/DependencyInstaller.sh -runtime; \
sudo ./etc/DependencyInstaller.sh -development; \
./etc/Build.sh `-cmake="-DGPU=true

**Output**
```
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CUDA_DEVICE_LINK_EXECUTABLE
```

I've tried following this:
https://cliutils.gitlab.io/modern-cmake/chapters/packages/CUDA.html 

But more issues pop up.

Thanks!

maliberty, 2022-08-23:
The GPU code is quite new and probably not worth the bother at this point.  If you want to persist then you need to install CUDA on your machine which is not handled through the dependency installer.

QuantamHD, 2022-08-23:
You can install cuda from Nvidia's website. https://developer.nvidia.com/cuda-downloads

QuantamHD, 2022-08-24:
@monibahmed Any luck getting it to work? Would love to have another user of this feature.

monibahmed, 2022-09-02:
Yes, some luck install but its a manual hacking process. Trying to figure out some cmake flags

QuantamHD, 2022-09-02:
If there is anything not working it would be great to try and get those fixes in the main project.

vijayank88, 2023-06-19:
@monibahmed 
Have you implemented this?

monibahmed, 2024-03-08:
Closing issue as OpenLane2 and Nix seem to solve this problem

closed 2024-03-08 by monibahmed