The-OpenROAD-Project/OpenROAD #2209: routing / pdngen: how to connect a stdcell to two different voltage domains?

luccareinehr, 2022-08-27:
Again in a design from OpenFASoC with two voltage domains (VDD and VIN), I have some cells placed in a VDD / GND grid that must also connect to the VIN net. However, when running `global_route` and `detailed_route`, there's no track from these cells to the VIN voltage domain ring (generated via pdngen).

The two voltage domains:
[image]

The cells that must connect to VIN (they're already in the VDD grid):
[image]

The resulting route after `global_route` and `detailed_route` are run (there are no tracks from the cells to the VIN ring):
[image]

**How to create this connection?**

### To reproduce the case:
- Download [reproduce_case_routing.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9438785/reproduce_case_routing.tar.gz)
- Run `openroad` or `openroad -gui`
- Source the `script.tcl` file

luccareinehr, 2022-08-27:
I've noticed the routing to VIN works when using an older `pdn.cfg` for this design (deprecated in pdngen).

However, when using the newer `pdn.tcl` we wrote (like in the images above), it doesn't.

Here are these files, if need be: 
[pdngen_files_case_routing.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9438787/pdngen_files_case_routing.tar.gz)

maliberty, 2022-08-28:
Power nets are not routed by the signal routers (grt/drt).  pdn doesn't create tracks.  The description of this problem is very confusing.

maliberty, 2022-08-28:
This looks like the enhancement request for sroute that I keep asking for a spec for.  Nothing will happen until this project has been properly defined.  @msaligane

msaligane, 2022-08-28:
I have no bandwidth and I am not great at writing specs. 

@luccareinehr can you work on that with @maliberty .

luccareinehr, 2022-08-29:
@maliberty Sorry for the confusing description. What I meant was: there's a cell that should connect to a voltage domain it is not in (VIN), but the routing doesn't get generated.

There's a workaround for this where a second VIN net is created exactly in the same place as the original, but with a SIGNAL type (instead of POWER), and this new net gets connected to the cells in the other voltage domain instead. Then routing does get generated automatically, but I doubt this is the intended solution:

[image]

@msaligane mentioned requesting the addition of a point-to-point router to OpenROAD, is this the enhancement request Matt mentions in his comment?

luccareinehr, 2022-08-29:
@msaligane mentioned requesting the addition of a point-to-point router to OpenROAD, is this the enhancement request Matt mentions in his comment?

luccareinehr, 2022-08-29:
@maliberty Indeed, I'll reserve time to talk to mehdi and give you a proper spec.

As for this issue, yes it could be solved with a p2p router, but I don't think that's the best solution. Afaik, OpenROAD is intended to be a no-human-in-the-loop flow, but in its current state, if there are multiple voltage domains in a design and a cell which has connections to more than one of them (and is not a level converter), it won't get routed automatically.

Please correct me if I misunderstood something, or if my request is not very clear. Thanks for the support by the way

rovinski, 2022-08-29:
> Afaik, OpenROAD is intended to be a no-human-in-the-loop flow

From my understanding, OpenROAD is intended to be no-human-in-the-loop for digital designs. It's a lofty goal that still has a lot of work to be done.

> if there are multiple voltage domains in a design and a cell which has connections to more than one of them (and is not a level converter), it won't get routed automatically.

For clarity, what you are describing is a special net router and is not typically used in digital flows. The use case for this would be for non-digital signals, as in your case. So as @maliberty said, it's not currently supported and would have to be implemented.

luccareinehr, 2022-08-29:
@rovinski Thank you for the clarification!

@maliberty should I close this issue?

QuantamHD, 2022-08-29:
You can rename this issue to ask for the for feature.

maliberty, 2022-08-31:
The cell is connecting to a power ring not another cell.  That is not the normal multi-power domain P&R use model.  Even in proprietary tools this is a completely separate router because it is a totally different requirement from signal net routing.

QuantamHD, 2022-08-31:
@luccareinehr Could you potentially locate these cells in a third voltage domain and then connect the vin from temperature sensor block?

Beyond the routing issue I'm a little bit confused about how the pdn in the logic is isolated from the Vin of the tempature block if you connect them together. Is there something I'm missing?

rovinski, 2022-08-31:
Some comments on this feature:

I've actually done this before in a taped-out chip when connecting an on-chip voltage regulator to the power rails of the regulated domain. So basically the same use case, but our regulator was a block, not a standard cell.

It becomes a very tricky problem because the main concern is not about DRVs or timing, but the IR drop across/peak current through the routing. We had to estimate the peak current draw by the power domain and size the routing large enough (using multiple connections) to ensure that electromigration and IR drop were minimized.

Such a tool in OpenROAD would be a large undertaking. A simple repurposing of the detailed router to perform NDR connections is possible, but risky from a design perspective because it wouldn't be able to optimize for power integrity.

QuantamHD, 2022-08-31:
@rovinski Could we make a small change to the router to have it replicate connections for power purposes as quick hack? So in @luccareinehr's case we would create N duplicate nets to route to the power ring/grid. That's obviously not optimal, but would at least give designers a lever to be more or less conservative. 

We could eventually run IR drop analysis and increase N until we were happy in future improvement.

arlpetergadfort, 2022-08-31:
In this case, what is VIN supposed to be used for? Is this for a ring oscillator to modulate it's speed? How is VIN generated?
I'm just wondering.

rovinski, 2022-09-01:
@QuantamHD The best first strategy would be to use a single connection with non default width (non-default routing / NDR). This strategy is still limited by the pin size on the block / cell which is generating the signal (i.e. if the pin is small, there is a practical limitation on how wide the metal route can be).

Creating multiple connections is only necessary if the net resistance requirements can't be met with a single connection (e.g. the desired metal width is greater than the max allowable width, or the width is capped by the size of the source pin). In this case, the power source would need to have multiple pins or a pin which is larger than max width. This only occurs for larger power sinks and is likely not applicable to @luccareinehr's design (my design was a multicore processor).

My concern isn't that there's no quick hack to fix @luccareinehr's problem; the concern is that the tools would not have a way of recognizing and warning the user of potentially chip-breaking mistakes.

It would be possible to use the detailed router to provide a partial solution this problem (such as @luccareinehr's case) but a general solution to the problem (making N connections; verifying power integrity) would likely require significantly different algorithms. For what it's worth, I don't think even commercial tools handle this case automatically. It usually takes significant hand-holding to get the special router to generate clean, quality routing.

luccareinehr, 2022-09-01:
@QuantamHD Despite how the pdn looks in the images, only the VSS of both voltage domains are shorted!

@arlpetergadfort Yes that's it, VIN modulates a ring oscillator. It is generated from a voltage drop in VDD after a couple of transistors in subthreshold regime, upon temperature variation (the design in question is a temperature sensor)

gadfort, 2024-10-18:
Sroute was added

closed 2024-10-18 by gadfort