The-OpenROAD-Project/OpenROAD #2210: tapcell: tap placement doesn't account for already placed stdcells, resulting in overlaps

luccareinehr, 2022-08-27:
When running `tapcell` in a design that already has manually placed instances of stdcells, the tap cells simply ignore these instances, sometimes being placed on top of a std cell, which results in overlaps (and fails the `check_placement` command).

Take this design, for example:
[image]

Running `tapcell` results in overlapping instances:
[image]
[image]

Thus failing the overlap check after running `check_placement`:
```
[WARNING DPL-0005] Overlap check failed (4).
 TAP_131 overlaps temp_analog_1.a_header_0
 TAP_136 overlaps temp_analog_1.a_header_2
 TAP_141 overlaps temp_analog_1.a_header_4
 TAP_147 overlaps temp_analog_1.a_header_6
```

Increasing the distance between tapcells quickly solves the problem, but shouldn't `tapcell` go around stdcells if they're already present in the design?

### To reproduce this case:
- Download [reproduce_case_tapcell.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9438832/reproduce_case_tapcell.tar.gz)
- Open OpenROAD or OpenROAD GUI
- Source the `script.tcl` file

QuantamHD, 2022-08-28:
This is just not implemented in tapcell. I don't think it has any concept of blockages.

maliberty, 2022-08-28:
Does the cell you are placing contain a tapcell?  If not how will you avoid a well tap violation even if he tap is placed on either side?

eder-matheus, 2022-08-29:
The tapcell tool assumes that the global placement will run after it. The tapcells have a specific spacing that needs to be respected, so they're placed as fixed instances. During the global and detailed placement, the std cells can be moved to avoid overlaps with the tapcells. That's why the tapcell tool is not looking for std cell blockages.

luccareinehr, 2022-08-30:
Yes, this stdcell has tap cells in it, but I understand it. Thank you all for the feedback

closed 2022-08-30 by luccareinehr