The-OpenROAD-Project/OpenROAD #2214: extract_parasitics crash

kareefardi, 2022-08-28:
openroad is crashing when running `extract_parasitics` it starts with a bunch of these messages:
```
Error 0: Internal Error: tmg_conn::detachTilePins: tilepin inside iterm. should not happen.
```
followed by this stack trace:
```
Stack trace:
 0# 0x0000000000C1AD66 in openroad
 1# 0x00007FA4AEE80400 in /lib64/libc.so.6
 2# odb::tmg_conn::copyWireIdToVisitedShorts(int) in openroad
 3# odb::tmg_conn::addToWire(int, int, int, bool, bool) in openroad
 4# odb::tmg_conn::treeReorder(bool, bool) in openroad
 5# odb::tmg_conn::analyzeNet(odb::dbNet*, bool, bool, bool, int, int, bool) in openroad
 6# odb::orderWires(odb::dbBlock*, bool, bool, int, int) in openroad
 7# rcx::Ext::extract(rcx::Ext::ExtractOptions) in openroad
 8# extract(char const*, int, double, float, int, int, int, char const*, bool, bool) in openroad
 9# 0x000000000139C4BE in openroad
10# 0x00007FA4B1BCDEB2 in /lib64/libtcl8.5.so
11# 0x00007FA4B1C1236C in /lib64/libtcl8.5.so
12# TclObjInterpProcCore in /lib64/libtcl8.5.so
13# 0x00007FA4B1BCDEB2 in /lib64/libtcl8.5.so
14# 0x00007FA4B1BCEF1E in /lib64/libtcl8.5.so
15# Tcl_EvalEx in /lib64/libtcl8.5.so
16# TclEvalObjEx in /lib64/libtcl8.5.so
17# 0x00007FA4B1BD5A65 in /lib64/libtcl8.5.so
18# 0x00007FA4B1BCDEB2 in /lib64/libtcl8.5.so
19# 0x00007FA4B1C1236C in /lib64/libtcl8.5.so
20# 0x00007FA4B1C1A647 in /lib64/libtcl8.5.so
21# TclEvalObjEx in /lib64/libtcl8.5.so
22# 0x00007FA4B1BD51D0 in /lib64/libtcl8.5.so
23# 0x00007FA4B1BCDEB2 in /lib64/libtcl8.5.so
24# 0x00007FA4B1C1236C in /lib64/libtcl8.5.so
25# 0x00007FA4B1C1A647 in /lib64/libtcl8.5.so
26# TclEvalObjEx in /lib64/libtcl8.5.so
27# 0x00007FA4B1C5527F in /lib64/libtcl8.5.so
28# 0x00007FA4B1BCDEB2 in /lib64/libtcl8.5.so
29# 0x00007FA4B1C1236C in /lib64/libtcl8.5.so
30# TclObjInterpProcCore in /lib64/libtcl8.5.so
31# 0x00007FA4B1BCDEB2 in /lib64/libtcl8.5.so
32# 0x00007FA4B1BCEF1E in /lib64/libtcl8.5.so
33# Tcl_EvalEx in /lib64/libtcl8.5.so
34# Tcl_Eval in /lib64/libtcl8.5.so
35# ord::tclAppInit(Tcl_Interp*) in openroad
36# Tcl_Main in /lib64/libtcl8.5.so
37# main in openroad
38# __libc_start_main in /lib64/libc.so.6
39# 0x0000000000C17807 in openroad
```
I understand that the def file might be malformed. If that is the case, a better error message without a crash would be much more helpful. Attached is a def, a lef and the extraction rules file used. openroad version used is `4a99e88667b0840531ca0096c4fa0da8f80d6cb1`
[test-case.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9440047/test-case.tar.gz)

maliberty, 2022-08-29:
I don't think it is your problem but I noticed this ugly connection in housekeeping/mgmt_gpio_in[19]
[image]

maliberty, 2022-08-29:
How was the routing generated?   I see some odd shapes on met2 as well:
[image]

kareefardi, 2022-08-29:
This routing was done using very old version of openROAD. However, the def isn't directly generated by openROAD. It is output of magic from a gds.

maliberty, 2022-08-30:
In housekeeping/mgmt_gpio_in[19] I see multiple overlapping segs:
```
      NEW met3 ( 1503955 4979300 ) ( 1525820 * )
      NEW met3 ( 1503955 4979300 ) ( 1525820 * )
      NEW met3 ( 1504260 4979300 ) ( 1525820 * )
```
this triggers a bug in some of the most horrible code that is very hard to fix.  While the tool shouldn't crash it will take a lot of effort to resolve this.  I suggest you generate an updated routing that is more well formed in the short term.

closed 2022-08-30 by maliberty

kareefardi, 2022-09-07:
@RTimothyEdwards this def was generated using magic. is there a workaround to merge shapes like the one above into a single shape?

kareefardi, 2022-09-07:
@maliberty i see that openroad successfully right a spef for the above testcase. is the quality of the spef affected by the original cause of the crash ?

maliberty, 2022-09-07:
I don't know how you got a spef after it crashed.  I would run it again.

kareefardi, 2022-09-08:
I meant that I ran it again and the update prevented the crash and the extraction finished. I am wondering if after the crash fix the spef is valid.

maliberty, 2022-09-08:
As far as I know it is valid

PR #2224 (2022-08-30): odb: fix crash in de-looping during order wires
This code is completely mysterious so this just prevents the crash.  Whether
there is a deeper problem or not I can't say.  The input data has multiple
overlapping segments which is peculiar.

Fixes #2214
commit 4e646ea4 (2022-08-30): odb: fix crash in de-looping during order wires

This code is completely mysterious so this just prevents the crash.  Whether
there is a deeper problem or not I can't say.  The input data has multiple
overlapping segments which is peculiar.

Fixes #2214