The-OpenROAD-Project/OpenROAD #2220: compile error in spdlog 1.10.0 on M1 MacBook Air

stefanottili, 2022-08-30:
Hi Folks,

I've been successfully compiling/running openroad on a M1 MacBook Air using brew for the dependencies for a while now.
A recent "brew update" seems to have broken the compile with spdlog: 1.10.0

`[INFO FLW-0018] Compiling OpenROAD.
-- OpenROAD version: v2.0-4818-g4174c3ad8
-- System name: Darwin
-- Compiler: AppleClang 13.1.6.13160021
-- Build type: RELEASE
-- Install prefix: /Users/user/OpenROAD-flow-scripts/tools/install/OpenROAD
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: /opt/homebrew/opt/tcl-tk/include/tcl.h
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.79.0/BoostConfig.cmake (found version "1.79.0")  
-- spdlog: 1.10.0

[ 70%] Built target abc
In file included from /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/LoadBalancer.cc:34:
In file included from /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/src/utl/include/utl/Logger.h:50:
In file included from /opt/homebrew/include/spdlog/spdlog.h:12:
In file included from /opt/homebrew/include/spdlog/common.h:45:
In file included from /opt/homebrew/include/spdlog/fmt/fmt.h:27:
/opt/homebrew/include/fmt/core.h:1756:3: error: static_assert failed due to requirement 'formattable' "Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt"
  static_assert(
  ^
/opt/homebrew/include/fmt/core.h:1777:10: note: in instantiation of function template specialization 'fmt::detail::make_value<fmt::basic_format_context<fmt::appender, char>, const boost::asio::ip::address &>' requested here
  return make_value<Context>(val);
         ^
/opt/homebrew/include/fmt/core.h:1899:23: note: in instantiation of function template specialization 'fmt::detail::make_arg<true, fmt::basic_format_context<fmt::appender, char>, fmt::detail::type::custom_type, const boost::asio::ip::address &, 0>' requested here
        data_{detail::make_arg<
                      ^
/opt/homebrew/include/fmt/core.h:1918:10: note: in instantiation of function template specialization 'fmt::format_arg_store<fmt::basic_format_context<fmt::appender, char>, boost::asio::ip::address, unsigned short, unsigned short>::format_arg_store<const boost::asio::ip::address &, const unsigned short &, const unsigned short &>' requested here
  return {FMT_FORWARD(args)...};
         ^
/opt/homebrew/include/spdlog/logger.h:370:52: note: in instantiation of function template specialization 'fmt::make_format_args<fmt::basic_format_context<fmt::appender, char>, const boost::asio::ip::address &, const unsigned short &, const unsigned short &>' requested here
            fmt::detail::vformat_to(buf, fmt, fmt::make_format_args(std::forward<Args>(args)...));
                                                   ^
/opt/homebrew/include/spdlog/logger.h:90:9: note: in instantiation of function template specialization 'spdlog::logger::log_<const boost::asio::ip::address &, const unsigned short &, const unsigned short &>' requested here
        log_(loc, lvl, fmt, std::forward<Args>(args)...);
        ^
/opt/homebrew/include/spdlog/logger.h:96:9: note: in instantiation of function template specialization 'spdlog::logger::log<const boost::asio::ip::address &, const unsigned short &, const unsigned short &>' requested here
        log(source_loc{}, lvl, fmt, std::forward<Args>(args)...);
        ^
/Users/user/OpenROAD-flow-scripts/tools/OpenROAD/src/utl/include/utl/Logger.h:111:16: note: in instantiation of function template specialization 'spdlog::logger::log<const boost::asio::ip::address &, const unsigned short &, const unsigned short &>' requested here
      logger_->log(spdlog::level::level_enum::off, message, args...);
               ^
/Users/user/OpenROAD-flow-scripts/tools/OpenROAD/src/dst/src/LoadBalancer.cc:47:16: note: in instantiation of function template specialization 'utl::Logger::report<boost::asio::ip::address, unsigned short, unsigned short>' requested here
      logger_->report("Worker {}/{} handled {} jobs",
               ^
1 error generated.
`

rovinski, 2022-08-30:
See #2097. I'm not exactly sure what the fix is, but the discussion there will probably help.

maliberty, 2022-08-30:
@vvbandeira how is this setup on our mac CI?

maliberty, 2022-08-30:
Very similar to #2173

vvbandeira, 2022-08-30:
Just posted here: https://github.com/The-OpenROAD-Project/OpenROAD/issues/2173#issuecomment-1231700671
TL;DR; we installed a version that worked a while back and did a `brew pin` to avoid upgrading and breaking things.

QuantamHD, 2022-08-31:
New versions of fmtlib are very picky about user defined types. In this case it's unhappy that DST is trying to print a boost IP address

src/dst/src/LoadBalancer.cc
```c++
  if (jobs_ != 0 && jobs_ % 100 == 0) {
    logger_->info(utl::DST, 7, "Processed {} jobs", jobs_);
    auto copy = workers_;
    while (!copy.empty()) {
      auto worker = copy.top();
      logger_->report("Worker {}/{} handled {} jobs",
                      worker.ip,
                      worker.port,
                      worker.priority);
      copy.pop();
    }
  }
```

worker.ip is a boost::ip::address which is why it's complaining. This needs to be converted to a string before adding it to the logger.

rovinski, 2022-09-03:
It's not just this one instance, it's actually several of them. See the patch in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2097#issuecomment-1208952962.

stefanottili, 2022-09-08:
Following the commands outlined in #2240 to install and fix an older, working combination of fat/spdlog gets the compile working again.

closed 2022-09-08 by stefanottili