The-OpenROAD-Project/OpenROAD #2221: Rocket Mini to GDSII failed at step 15

Ryabkomv, 2022-08-30:


maliberty, 2022-08-30:
@donn  the OL relevant piece is "OL should also drop checking max fanout at the end of the flow as I've also suggested."

JiaDYuan, 2022-10-26:
> I did not read the whole thread. But I noticed some abnormalities worth investigating:
> 
> 1. CLOCK_PORT is supposed to be "clock". Assuming RockerChip uses Chisel default clock naming. I burnt there a couple of times.
> 2. the density is set to the corresponding value as the FP CORE UTIL. This may cause congestion issues down the line. I would recommend increasing the density to be at least 10% higher than core util. In the design according to the logs, the density is much higher than it is set to.
> 3. Always create a new design using -init_design. Copy pasting some designs folder may cause the wrong files to be picked, which you did not intend to. Burnt there many times.
> 4. the RocketChip is supposed to have the RAM cells replaced with tech-specific. Otherwise, the area usage is going to be insane => long runtimes and a lot of memory usage. It generates the config file, which can be used to configure the DFFRAM.

Hi,  I want to ask how to use the DFFRAM or openRAM in the ORFS flow to place and route a Rocket core?

maliberty, 2022-11-01:
You would run openram to generate the ram and then add the LEF to the P&R flow with the gds used for the final gds generation.  There is no direct link between the two.

vijayank88, 2023-06-14:
@JiaDYuan 
Have you updated your source file with OpenRAM?