The-OpenROAD-Project/OpenROAD #2222: pdngen (I think): vias between power rings don't respect via spacing rule for arrays (ADJACENTCUTS) in tech LEF

Syndace, 2022-08-30:
Hi,

our tech LEF has two rules for spacing of vias: a general one and one for spacing in arrays. This is (roughly) what the relevant entries look like:

```
LAYER via4
  ...
  SPACING 0.25 ;
  SPACING 0.35 ADJACENTCUTS 3 WITHIN 0.4 ;
  ...
END via4
```

We are generating rather large power rings that are connected with via arrays of more than 10x10 vias, and the vias are spaced with 0.25, when they should be spaced with 0.35 according to the second rule.

Syndace, 2022-08-31:
I just realized that the documentation states that VIARULES are used for the array generation. Those are present in the technology LEF too:

```
ViaRULE via4Array GENERATE
    LAYER metal4 ;
        ENCLOSURE 0.050 0.005 ;

    LAYER metal5 ;
        ENCLOSURE 0.050 0.005 ;

    LAYER via4 ;
        RECT -0.095 -0.095 0.095 0.095 ;
        SPACING 0.480 BY 0.480 ;
        RESISTANCE 20.0 ;
END via4Array
```

arlpetergadfort, 2022-08-31:
I suspect you are correct. Would you be able to put together a testcase for us?

arlpetergadfort, 2022-08-31:
@maliberty does ODB store the second spacing? It does not appear to do so (atleast from my looking around)

maliberty, 2022-08-31:
@arlpetergadfort they are there in the horribly named getV54SpacingRules().  I'm going fix  up that API so it is reachable from TCL

Syndace, 2022-09-01:
wow, that was blazingly fast, thank you! What's the best way for me to test this within OpenLane? Do I compile OpenROAD outside and then move the binary into the container?

arlpetergadfort, 2022-09-01:
@Syndace I don't use OpenLane so I don't know. Maybe @maliberty or @donn can help with that.

maliberty, 2022-09-01:
You'll need to update the commit id in dependencies/tool_metadata.yml and then update the tools by using the steps in https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docker/README.md

Syndace, 2022-09-09:
Sorry for the delay, the vias are spaced correctly with the newest commit, thanks!

closed 2022-09-09 by Syndace