The-OpenROAD-Project/OpenROAD #2231: Placement results orientation is always the same

SkanderMarsit, 2022-09-01:
Hi, 

I'm trying to perform placement optimization with the global_place and detailed_place function. But I'm seeing that the result has always all the components placed north. 
I'm trying to figure out what to change in order to allow for all possible orientations. 

I thought it could be the SITE and ROW configuration.  I could be totally wrong.

The site config is : 
SITE FreePDK45_38x28_10R_NP_162NW_34O
  CLASS core ;
  SIZE 6 BY 6 ;
END FreePDK45_38x28_10R_NP_162NW_34O

and the Rows : 

ROW CORE_ROW_0 FreePDK45_38x28_10R_NP_162NW_34O 0 0  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_1 FreePDK45_38x28_10R_NP_162NW_34O 0 12000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_2 FreePDK45_38x28_10R_NP_162NW_34O 0 24000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_3 FreePDK45_38x28_10R_NP_162NW_34O 0 36000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_4 FreePDK45_38x28_10R_NP_162NW_34O 0 48000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_5 FreePDK45_38x28_10R_NP_162NW_34O 0 60000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_6 FreePDK45_38x28_10R_NP_162NW_34O 0 72000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_7 FreePDK45_38x28_10R_NP_162NW_34O 0 84000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_8 FreePDK45_38x28_10R_NP_162NW_34O 0 96000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_9 FreePDK45_38x28_10R_NP_162NW_34O 0 108000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_10 FreePDK45_38x28_10R_NP_162NW_34O 0 120000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_11 FreePDK45_38x28_10R_NP_162NW_34O 0 132000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_12 FreePDK45_38x28_10R_NP_162NW_34O 0 144000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_13 FreePDK45_38x28_10R_NP_162NW_34O 0 156000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_14 FreePDK45_38x28_10R_NP_162NW_34O 0 168000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_15 FreePDK45_38x28_10R_NP_162NW_34O 0 180000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_16 FreePDK45_38x28_10R_NP_162NW_34O 0 192000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_17 FreePDK45_38x28_10R_NP_162NW_34O 0 204000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_18 FreePDK45_38x28_10R_NP_162NW_34O 0 216000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_19 FreePDK45_38x28_10R_NP_162NW_34O 0 228000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_20 FreePDK45_38x28_10R_NP_162NW_34O 0 240000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_21 FreePDK45_38x28_10R_NP_162NW_34O 0 252000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_22 FreePDK45_38x28_10R_NP_162NW_34O 0 264000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_23 FreePDK45_38x28_10R_NP_162NW_34O 0 276000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_24 FreePDK45_38x28_10R_NP_162NW_34O 0 288000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_25 FreePDK45_38x28_10R_NP_162NW_34O 0 300000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_26 FreePDK45_38x28_10R_NP_162NW_34O 0 312000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_27 FreePDK45_38x28_10R_NP_162NW_34O 0 324000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_28 FreePDK45_38x28_10R_NP_162NW_34O 0 336000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_29 FreePDK45_38x28_10R_NP_162NW_34O 0 348000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_30 FreePDK45_38x28_10R_NP_162NW_34O 0 360000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_31 FreePDK45_38x28_10R_NP_162NW_34O 0 372000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_32 FreePDK45_38x28_10R_NP_162NW_34O 0 384000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_33 FreePDK45_38x28_10R_NP_162NW_34O 0 396000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_34 FreePDK45_38x28_10R_NP_162NW_34O 0 408000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_35 FreePDK45_38x28_10R_NP_162NW_34O 0 420000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_36 FreePDK45_38x28_10R_NP_162NW_34O 0 432000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_37 FreePDK45_38x28_10R_NP_162NW_34O 0 444000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_38 FreePDK45_38x28_10R_NP_162NW_34O 0 456000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_39 FreePDK45_38x28_10R_NP_162NW_34O 0 468000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_40 FreePDK45_38x28_10R_NP_162NW_34O 0 480000  N DO 42 BY 1 STEP 12000 0 ;
ROW CORE_ROW_41 FreePDK45_38x28_10R_NP_162NW_34O 0 492000  N DO 42 BY 1 STEP 12000 0 ;

vijayank88, 2022-09-01:
@SkanderMarsit 
Can you please attach the reproducible test case?

vijayank88, 2022-09-01:
Have you tried design with this stable flow: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/global_place.tcl ?

vijayank88, 2022-09-01:
With your inputs getting following log:
```
[InitialPlace]  Iter: 6 CG residual: 0.01679070 HPWL: 7681458
[InitialPlace]  Iter: 7 CG residual: 651.81402588 HPWL: 8076945
[InitialPlace]  Iter: 8 CG residual: 0.00000231 HPWL: 1682469
[ERROR GPL-0305] RePlAce diverged at newStepLength.
[ERROR GUI-0070] GPL-0305
```
Is similar to #1069 ?

maliberty, 2022-09-01:
The instance orientation will match the row orientation and all of yours are N.  How did you generate the rows?

vijayank88, 2022-09-02:
after detailed placement:
[image]

SkanderMarsit, 2022-09-02:
Thank you @maliberty very much. I generate the rows manually. I'm actually performing a conversion from one file forrmat to LEF/DEF in order to perform the RePlace Automatic placement. 
Then, If I understand correctly, I should review define rows for every orientation, (4 of them) or should I only define the N and FS.

Thank you @vijayank88 as well, that's how it turns out to look like, mine is a bit different, maybe because I didn't use -routability-driven mode in the global_placement. 

I also have another question. Like you see in the placement result photo, the components are very close to one another, which make routabilty difficult or impossible. Do you know if there is any settings or options to set like the clearance value or the spacing to increase the spacing between component and disallow very close components.

vijayank88, 2022-09-02:
@SkanderMarsit 
Refer this document: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/mpl/README.md#commands
Use `-halo` and `-channel` switch for spacing

maliberty, 2022-09-02:
Only two orientations after normally used

SkanderMarsit, 2022-09-05:
@maliberty Thanks for the response. I took inspiration ftom other DEF files and I defined my ROWS by alterning N and FS orientations. But the result seem to be the same as only North oriented rows. 

@vijayank88 Thanks for the precious suggestion. But when running the macro_placement command I get and error :
[ERROR MPL-0095] Snap layer 4 is not a routing layer.
[ERROR GUI-0070] MPL-0095

The LEF, ad TCL file are left unchanged. 
Here is the new DEF file.
[test.def.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/9488113/test.def.txt)

maliberty, 2022-09-06:
The components in your def are not in your lef so there is nothing to place:
```
COMPONENTS 10 ;
- D1 LED:generic-LED1608 ;
- P1 pin-head:pinhead-1X08 ;
- P2 pin-head:pinhead-1X08::1 ;
- SW1 SMD_PUSHSWITCH:SKRPACE010 ;
- P3 GroveCon:GROVE ;
- U1 AMW037:AMW037 ;
- C1 Capacitor_SMD:C_0805_2012Metric ;
- C3 Capacitor_SMD:C_0603_1608Metric ;
- R2 Capacitor_SMD:C_0603_1608Metric ;
- R8 Resistor_SMD:R_0603_1608Metric ;
END COMPONENTS
```

maliberty, 2022-09-06:
[WARNING ODB-0092] error: unknown library cell referenced (LED:generic-LED1608) 
for instance (D1)
[WARNING ODB-0092] error: unknown library cell referenced (pin-head:pinhead-1X08
) for instance (P1)
[WARNING ODB-0092] error: unknown library cell referenced (pin-head:pinhead-1X08::1) for instance (P2)
[WARNING ODB-0092] error: unknown library cell referenced (SMD_PUSHSWITCH:SKRPACE010) for instance (SW1)
[WARNING ODB-0092] error: unknown library cell referenced (GroveCon:GROVE) for instance (P3)
[WARNING ODB-0092] error: unknown library cell referenced (AMW037:AMW037) for instance (U1)
[WARNING ODB-0092] error: unknown library cell referenced (Capacitor_SMD:C_0805_2012Metric) for instance (C1)

etc

SkanderMarsit, 2022-09-07:
Sorry I messed up I didn't double check the file before posting.
Here's the right one.
[Uploading test.def.txt…]()

SkanderMarsit, 2022-09-07:
[test.def.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/9504236/test.def.txt)

maliberty, 2022-09-07:
I see an MX instance now:
[image]

What is the issue?

SkanderMarsit, 2022-09-09:
Hey @maliberty Thanks for your precious help. 
I first thought that MX was the same thing as R0 but after verification It turns out they are not the same. MX, as far as I understood is the result of flipping the component.
I also would like to allow R90 R180 etc... 
Does it have to do with the SITE not having any symmetry.
My SIte Definition being:
SITE FreePDK45_38x28_10R_NP_162NW_34O
  CLASS core ;
  SIZE 6 BY 6 ;
END FreePDK45_38x28_10R_NP_162NW_34O
I can add a symmetry axis.
Does the shape of the SITE affect the orientation?

maliberty, 2022-09-12:
R90 would make no sense as your power rails wouldn't connect.  The site can define symmetry (eg https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/d43e3289253003b32fbe327c160ebc879777fcbd/flow/platforms/nangate45/lef/NangateOpenCellLibrary.tech.lef#L771) as can the cell (eg https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/d43e3289253003b32fbe327c160ebc879777fcbd/flow/platforms/nangate45/lef/NangateOpenCellLibrary.macro.lef#L40)

maliberty, 2022-09-12:
Both are needed to enable flipping.

vijayank88, 2023-06-19:
Staled out.

closed 2023-06-19 by vijayank88