The-OpenROAD-Project/OpenROAD #2232: routing: creates islands on metal2 that are smaller than required by tech LEF

Syndace, 2022-09-01:
Hi,

there seems to be a specific scenario in which the AREA requirement in the tech LEF for Metal2 is not met. Our technology LEF has

```
LAYER Metal2
  ...
  AREA      0.144 ;
  ...
END Metal2
```

This minimum area requirement seems to be respected, except in the following scenario: when a wire has to be via'd from Metal1 to Metal3, a Via1 is generated, a small island on Metal2, and a Via2. In this case, we sometimes see Metal2 islands that are a little smaller than the required 0.144um²:

[image]

(yellow is Via1, purple is Via2, blue is Metal2)

With Metal1 and Metal3 enabled:

[image]

The sources/PDK are under NDA. I'll try my best to provide a reproducible setup using an open-source PDK ASAP, but wanted to get the description out now, in case someone already has an idea what the cause could be from this information alone.

maliberty, 2022-09-01:
This is a common scenario and the router creates patch metal to meet min area.  You can see the same in sky130hd.  

@tspyrou can arrange support under NDA if required.

Syndace, 2022-09-01:
> This is a common scenario and the router creates patch metal to meet min area. You can see the same in sky130hd.

I'm sorry, I don't quite get what you're trying to say here. The router does not seem to create patch metal to meet the minimum area (correctly), since there are islands below minimum area in the final output GDS.

maliberty, 2022-09-01:
I saying this is already a feature of the router not an enhancement.  If it doesn't work in your case it is something unusual that I can't address without seeing the data.

Syndace, 2022-09-05:
Okay got it, thanks. I'll get back to this later.

Syndace, 2022-09-09:
This one was on us, sorry.

closed 2022-09-09 by Syndace