The-OpenROAD-Project/OpenROAD #2236: unexpect path found by sta

maliberty, 2022-09-01:
This test case produces a path begining:
```
Startpoint: driver_sel[1] (input port)
Endpoint: scan_select (output port clocked by clk)
Path Group: clk
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00 ^ input external delay
   0.00    0.00 ^ driver_sel[1] (in)
```
however there is no set_input_delay on driver_sel[1] so no path is expected.  @tspyrou confirms that there is no expected default inpupt delay.

[bug.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9471872/bug.tar.gz)

[From a slack discussion with user tnt]

maliberty, 2022-09-01:
The user has worked around this with a set_false_path but was confused by its presence.  It is an async input.

jjcherry56, 2022-09-01:
There is  no requirement that input have a set_input_delay to report a path from the input.
The reported external delay is 0.0 because there is no set_input_delay.
This is not a bug, it is the intended behavior.

closed 2022-09-01 by jjcherry56

jjcherry56, 2022-10-02:
I discovered that this is inconsistent with paths from inputs without set_input_delay to registers.
As of OpenSTA commit 
af35dfb 2022-09-23 paths output ports respect sta_input_port_default_clock
the path will not be reported unless sta_input_port_default_clock is 1.