The-OpenROAD-Project/OpenROAD #2244: Crash during global placement

donn, 2022-09-05:
* Using OpenROAD 4174c3ad802d2ac1d04d387d2c4b883903f6647e

```
OpenROAD 4174c3ad802d2ac1d04d387d2c4b883903f6647e 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Setting RC values...
[INFO]: Setting signal min routing layer to: met1 and clock min routing layer to met1. 
[INFO]: Setting signal max routing layer to: met5 and clock max routing layer to met5. 
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 460 2720
[INFO GPL-0004] CoreAreaLxLy: 5520 10880
[INFO GPL-0005] CoreAreaUxUy: 306360 310080
[INFO GPL-0006] NumInstances: 1098
[INFO GPL-0007] NumPlaceInstances: 0
[INFO GPL-0008] NumFixedInstances: 935
[INFO GPL-0009] NumDummyInstances: 163
[INFO GPL-0010] NumNets: 72
[INFO GPL-0011] NumPins: 144
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 312060 322780
[INFO GPL-0014] CoreAreaLxLy: 5520 10880
[INFO GPL-0015] CoreAreaUxUy: 306360 310080
[INFO GPL-0016] CoreArea: 90011328000
[INFO GPL-0017] NonPlaceInstsArea: 42691481600
[INFO GPL-0018] PlaceInstsArea: 0
[INFO GPL-0019] Util(%): 0.00
[INFO GPL-0020] StdInstsArea: 0
[INFO GPL-0021] MacroInstsArea: 0
[InitialPlace]  Iter: 1 CG residual: 0.00000012 HPWL: 28227480
[InitialPlace]  Iter: 2 CG residual: 0.00000012 HPWL: 28227480
[InitialPlace]  Iter: 3 CG residual: 0.00000012 HPWL: 28227480
[InitialPlace]  Iter: 4 CG residual: 0.00000012 HPWL: 28227480
[InitialPlace]  Iter: 5 CG residual: 0.00000012 HPWL: 28227480
Stack trace:
 0# 0x0000000000C78EE6 in openroad
 1# 0x00007FA00AD18400 in /lib64/libc.so.6
 2# gpl::NesterovBase::initFillerGCells() in openroad
 3# gpl::NesterovBase::init() in openroad
 4# gpl::NesterovBase::NesterovBase(gpl::NesterovBaseVars, std::shared_ptr<gpl::PlacerBase>, utl::Logger*) in openroad
 5# gpl::Replace::initNesterovPlace() in openroad
 6# gpl::Replace::doNesterovPlace(int) in openroad
 7# 0x0000000000CCA2E1 in openroad
 8# 0x00007FA00DA65EB2 in /lib64/libtcl8.5.so
 9# 0x00007FA00DAAA36C in /lib64/libtcl8.5.so
10# TclObjInterpProcCore in /lib64/libtcl8.5.so
11# 0x00007FA00DA65EB2 in /lib64/libtcl8.5.so
12# 0x00007FA00DAAA36C in /lib64/libtcl8.5.so
13# 0x00007FA00DAB2647 in /lib64/libtcl8.5.so
14# TclEvalObjEx in /lib64/libtcl8.5.so
15# 0x00007FA00DAED27F in /lib64/libtcl8.5.so
16# 0x00007FA00DA65EB2 in /lib64/libtcl8.5.so
17# 0x00007FA00DAAA36C in /lib64/libtcl8.5.so
18# 0x00007FA00DAB2647 in /lib64/libtcl8.5.so
19# TclEvalObjEx in /lib64/libtcl8.5.so
20# 0x00007FA00DA6D1D0 in /lib64/libtcl8.5.so
21# 0x00007FA00DA65EB2 in /lib64/libtcl8.5.so
22# 0x00007FA00DAAA36C in /lib64/libtcl8.5.so
23# 0x00007FA00DAB2647 in /lib64/libtcl8.5.so
24# TclEvalObjEx in /lib64/libtcl8.5.so
25# 0x00007FA00DA70F00 in /lib64/libtcl8.5.so
26# 0x00007FA00DA65EB2 in /lib64/libtcl8.5.so
27# 0x00007FA00DAAA36C in /lib64/libtcl8.5.so
28# TclObjInterpProcCore in /lib64/libtcl8.5.so
29# 0x00007FA00DA65EB2 in /lib64/libtcl8.5.so
30# 0x00007FA00DAAA36C in /lib64/libtcl8.5.so
31# TclObjInterpProcCore in /lib64/libtcl8.5.so
32# 0x00007FA00DA65EB2 in /lib64/libtcl8.5.so
33# 0x00007FA00DA66F1E in /lib64/libtcl8.5.so
34# Tcl_EvalEx in /lib64/libtcl8.5.so
35# Tcl_Eval in /lib64/libtcl8.5.so
36# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_Main in /lib64/libtcl8.5.so
39# main in openroad
40# __libc_start_main in /lib64/libc.so.6
41# 0x0000000000C75967 in openroad
```

Reproducible: [repro.tar.gz](https://www.dropbox.com/s/xj7jbmoprmsizzw/repro.tar.gz?dl=0)

maliberty, 2022-09-06:
I'll fix it but is it intentional that this design has no placeable objects (just 2 macros)?

donn, 2022-09-06:
Yes- it's a basic macro placement test in OpenLane. We've had that for a while now.

maliberty, 2022-09-06:
I don't see why this would be a new problem as the code in question is old.

donn, 2022-09-06:
That's... interesting. That means I likely dropped a fill step or something then. I'll take a look.

donn, 2022-09-06:
Okay, so, here's the deal: when the design is loaded from a DEF, this happens:

```
[WARN] All instances are FIXED
[WARN] No need to use replace
[WARN] Skipping...
```

When it's loaded from ODB, this crash occurs. That means it's not able to tell that all instances are FIXED when the database is loaded for some reason.

maliberty, 2022-09-06:
I don't have enough data to say if that is a problem in your scripts or in OR.

closed 2022-09-06 by maliberty

donn, 2022-09-06:
Would providing two reproducibles suffice?

maliberty, 2022-09-06:
I think so

donn, 2022-09-06:
Here's the working one from master: [repro_working.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9493473/repro_working.tar.gz)

You can compare with the broken one from the ODB branch.

maliberty, 2022-09-06:
In def out we map:
```
    case dbPlacementStatus::LOCKED:
    case dbPlacementStatus::FIRM: {
      fprintf(_out, " + FIXED ( %d %d ) %s", x, y, orient);
      break;
    }
```
where LOCKED is mapped to FIXED.  You are checking:
```
	if { $placement_status != "FIRM" } {
```
however when you use odb the original LOCKED type is preserved and this check fails.  It would be better to use
```
        if { ![$inst isFixed] } {
```

donn, 2022-09-07:
Thanks, that was the last problem I was having. ODB is just waiting on a review for the merge now.

PR #2246 (2022-09-06): gpl: skip placement if there are no placeable objects
Fixes #2244
commit cf78131d (2022-09-06): gpl: skip placement if there are no placeable objects

Fixes #2244