The-OpenROAD-Project/OpenROAD #2250: Routing time too high with latest OpenROAD

vijayank88, 2022-09-07:
Description:
With OpenLane tag: [2022.08.25](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.08.25) and OpenROAD commit: 4a99e88667b0840531ca0096c4fa0da8f80d6cb1 MPW design syntacore 2Hrs took to complete routing.

With OpenLane tag: [2022.09.05](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.05) and OpenROAD commit: 4174c3ad802d2ac1d04d387d2c4b883903f6647e its took 27Hrs to complete routing

MPW CI link with 2Hrs routing: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Fproduction-small-3/detail/main/16/pipeline/650

MPW CI link with 27Hrs routing: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Fproduction-small-3/detail/PR-80/1/pipeline/674

@maliberty please assign to right person to debug this.

Let me know if need more details on this.

Thanks

maliberty, 2022-09-07:
Can you test a single OL tag with the two versions?  Its hard to have two moving variables.

vijayank88, 2022-09-10:
@maliberty 
I just checked between all OpenLane tags till latest.
This issue starts from OpenLane [2022.09.03](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.03) tag. 
Till OpenLane [2022.09.02](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.02) tag its routing time was 5Hrs.

For both the tag OpenROAD commit is same 4174c3ad802d2ac1d04d387d2c4b883903f6647e
So its not a router issue. Something else causing the issue.

@donn Any suggestion how to trace and fix this runtime issue?

donn, 2022-09-10:
no

maliberty, 2022-09-12:
@vijayank88 would you package the drt step in both cases and I'll see if I can spot why it might be slower.  Likely the real problem is upstream but maybe we can find a hint

vijayank88, 2022-09-13:
Here is 5Hr routing test case: 
[issue_reproducible_0209.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9558285/issue_reproducible_0209.zip)

27Hrs routing test case: 
[issue_reproducible_0809.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9558301/issue_reproducible_0809.zip)

antonblanchard, 2022-09-13:
Almost certainly due to an update in Yosys. The standard cell count has gone from 92546 to 94380.

maliberty, 2022-09-13:
Even so it is not enough to explain such a large jump in routing time.  The congestion is a bit higher but it isn't crazy.

maliberty, 2022-09-14:
I looked at the runs and the slow one get stuck on a small number of hard DRVs.  Some I think should be fixable but it is in general hard.

maliberty, 2022-09-14:
Osama will take this on as part of his router improvement work but I don't think it represents any simple bug.  

@vijayank88  It would helpful to locate the upstream source of the additional instances that are increasing congestion.  It could be yosys or a change in settings or pdk.

vijayank88, 2022-09-15:
@maliberty 
I confirm only change is wrt., to Yosys update only.
I'll share the netlist with changes.

closed 2025-08-26 by maliberty