The-OpenROAD-Project/OpenROAD #2252: pdngen and padframes: conditions for `-connect_to_pads` to work?

Syndace, 2022-09-08:
I have created a padframe using [the `initialize_padframe` family of commands](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/pad/doc/README.md) and a core power ring with `add_pdn_ring` and the `-connect_to_pads` flag set. Sadly, while the frame and the grid individually look beautiful, the padframe doesn't end up connected to the core power net:

[image]

Is there a (working) example of a padframe connected to the core power net somewhere? Do you know of any caveats one needs to think of to make the connections appear?

This is how I've defined the power net (adjusted version of OpenLane's [pdn_cfg.tcl](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/openroad/pdn_cfg.tcl)):

```TCL
# stdcell power pins
add_global_connection \
    -net VDD \
    -pin_pattern VDD \
    -power

add_global_connection \
    -net VSS \
    -pin_pattern VSS \
    -ground

# padframe core power pins
add_global_connection \
    -net VDD \
    -pin_pattern VDDCORE \
    -power

add_global_connection \
    -net VSS \
    -pin_pattern VSSCORE \
    -ground

# SRAM power pins
add_global_connection \
    -net VDD \
    -pin_pattern VDD! \
    -power

add_global_connection \
    -net VDD \
    -pin_pattern VDDARRAY! \
    -power

add_global_connection \
    -net VSS \
    -pin_pattern VSS! \
    -ground

# core voltage domain
set_voltage_domain -name CORE -power VDD -ground VSS

# stdcell grid
define_pdn_grid \
    -name stdcell_grid \
    -starts_with POWER \
    -pins "Metal4 Metal5"

# stripes on Metal4 and Metal5
add_pdn_stripe \
    -grid stdcell_grid \
    -layer Metal4 \
    -width $::env(FP_PDN_VWIDTH) \
    -pitch $::env(FP_PDN_VPITCH) \
    -offset $::env(FP_PDN_VOFFSET) \
    -starts_with POWER \
    -extend_to_core_ring

add_pdn_stripe \
    -grid stdcell_grid \
    -layer Metal5 \
    -width $::env(FP_PDN_HWIDTH) \
    -pitch $::env(FP_PDN_HPITCH) \
    -offset $::env(FP_PDN_HOFFSET) \
    -starts_with POWER \
    -extend_to_core_ring

add_pdn_connect \
    -grid stdcell_grid \
    -layers "Metal4 Metal5"

# stdcell power rails on Metal1
add_pdn_stripe \
    -grid stdcell_grid \
    -layer Metal1 \
    -width $::env(FP_PDN_RAIL_WIDTH) \
    -followpins \
    -starts_with POWER \
    -extend_to_core_ring

add_pdn_connect \
    -grid stdcell_grid \
    -layers "Metal1 Metal4"

# core power rings
add_pdn_ring \
    -grid stdcell_grid \
    -layers "Metal4 Metal5" \
    -widths "30 30" \
    -spacings "5 5" \
    -core_offset "30 30" \
    -connect_to_pads

add_pdn_ring \
    -grid stdcell_grid \
    -layers "Metal4 Metal5" \
    -widths "30 30" \
    -spacings "5 5" \
    -core_offset "100 100" \
    -connect_to_pads

# SRAM grid and power ring
define_pdn_grid \
    -macro \
    -name macro \
    -cells SRAM \
    -starts_with POWER \
    -halo "70 70"

add_pdn_connect \
    -grid macro \
    -layers "Metal4 Metal5"

add_pdn_ring \
    -grid macro \
    -layers "Metal3 Metal4" \
    -widths "20 20" \
    -spacings "5 5" \
    -core_offset "5 5"

add_pdn_connect \
    -grid macro \
    -layers "Metal3 Metal4"
```
And this is how the padframe is defined:

```TCL
# I/O cells
define_pad_cell \
    -name padcell_io \
    -type io \
    -cell_name padcell_io \
    -orient { bottom R0 right R90 top R180 left R270 } \
    -pad_pin_name PAD

define_pad_cell \
    -name padcell_i \
    -type i \
    -cell_name padcell_i \
    -orient { bottom R0 right R90 top R180 left R270 } \
    -pad_pin_name PAD

# Power cells
define_pad_cell \
    -name vddpad \
    -type vddpad \
    -cell_name vddpad \
    -orient { bottom R0 right R90 top R180 left R270 } \
    -pad_pin_name VDDPAD

define_pad_cell \
    -name vddcore \
    -type vddcore \
    -cell_name vddcore \
    -orient { bottom R0 right R90 top R180 left R270 } \
    -pad_pin_name VDDCORE

define_pad_cell \
    -name gndpad \
    -type vsspad \
    -cell_name gndpad \
    -orient { bottom R0 right R90 top R180 left R270 } \
    -pad_pin_name VSSPAD

define_pad_cell \
    -name gndcore \
    -type vsscore \
    -cell_name gndcore \
    -orient { bottom R0 right R90 top R180 left R270 } \
    -pad_pin_name VSSCORE

# Corner cells
define_pad_cell \
    -name corner \
    -corner \
    -cell_name corner \
    -orient { ll R0 lr R90 ur R180 ul R270 } \
    -physical_only

# Filler cells
define_pad_cell \
    -name filler10u \
    -fill \
    -cell_name filler10u \
    -orient { bottom R0 right R90 top R180 left R270 } \
    -physical_only

define_pad_cell \
    -name filler4u \
    -fill \
    -cell_name filler4u \
    -orient { bottom R0 right R90 top R180 left R270 } \
    -physical_only

define_pad_cell \
    -name filler2u \
    -fill \
    -cell_name filler2u \
    -orient { bottom R0 right R90 top R180 left R270 } \
    -physical_only

define_pad_cell \
    -name filler1u \
    -fill \
    -cell_name filler1u \
    -orient { bottom R0 right R90 top R180 left R270 } \
    -physical_only

# Padring options
set_padring_options \
    -type wirebond \
    -power { VDD VDDPAD } \
    -ground { VSS VSSPAD } \
    -allow_filler_overlap

# Pads
add_pad -edge right -signal VDDPAD -inst_name u_vddpad_e0  -type vddpad
add_pad -edge right -signal VSSPAD -inst_name u_gndpad_e0  -type vsspad
add_pad -edge right -signal VDD    -inst_name u_vddcore_e0 -type vddcore
add_pad -edge right -signal VSS    -inst_name u_gndcore_e0 -type vsscore
add_pad -edge right -signal p_jtag_tms -inst_name u_jtag_tms -type i
add_pad -edge right -signal p_jtag_tdi -inst_name u_jtag_tdi -type i
add_pad -edge right -signal p_jtag_tdo -inst_name u_jtag_tdo -type io
add_pad -edge right -signal p_jtag_tck -inst_name u_jtag_tck -type i
add_pad -edge right -signal VDDPAD -inst_name u_vddpad_e1  -type vddpad
add_pad -edge right -signal VSSPAD -inst_name u_gndpad_e1  -type vsspad
add_pad -edge right -signal VDD    -inst_name u_vddcore_e1 -type vddcore
add_pad -edge right -signal VSS    -inst_name u_gndcore_e1 -type vsscore

add_pad -edge left -signal VDDPAD -inst_name u_vddpad_w0  -type vddpad
add_pad -edge left -signal VSSPAD -inst_name u_gndpad_w0  -type vsspad
add_pad -edge left -signal VDD    -inst_name u_vddcore_w0 -type vddcore
add_pad -edge left -signal VSS    -inst_name u_gndcore_w0 -type vsscore
add_pad -edge left -signal p_uart_txd -inst_name u_uart_txd -type io
add_pad -edge left -signal p_uart_rxd -inst_name u_uart_rxd -type i
add_pad -edge left -signal VDDPAD -inst_name u_vddpad_w1  -type vddpad
add_pad -edge left -signal VSSPAD -inst_name u_gndpad_w1  -type vsspad
add_pad -edge left -signal VDD    -inst_name u_vddcore_w1 -type vddcore
add_pad -edge left -signal VSS    -inst_name u_gndcore_w1 -type vsscore

add_pad -edge top -signal VDDPAD -inst_name u_vddpad_n0  -type vddpad
add_pad -edge top -signal VSSPAD -inst_name u_gndpad_n0  -type vsspad
add_pad -edge top -signal VDD    -inst_name u_vddcore_n0 -type vddcore
add_pad -edge top -signal VSS    -inst_name u_gndcore_n0 -type vsscore
add_pad -edge top -signal VDDPAD -inst_name u_vddpad_n1  -type vddpad
add_pad -edge top -signal VSSPAD -inst_name u_gndpad_n1  -type vsspad
add_pad -edge top -signal VDD    -inst_name u_vddcore_n1 -type vddcore
add_pad -edge top -signal VSS    -inst_name u_gndcore_n1 -type vsscore

add_pad -edge bottom -signal VDDPAD -inst_name u_vddpad_s0  -type vddpad
add_pad -edge bottom -signal VSSPAD -inst_name u_gndpad_s0  -type vsspad
add_pad -edge bottom -signal VDD    -inst_name u_vddcore_s0 -type vddcore
add_pad -edge bottom -signal VSS    -inst_name u_gndcore_s0 -type vsscore
add_pad -edge bottom -signal p_clk -inst_name u_clk -type i
add_pad -edge bottom -signal VDDPAD -inst_name u_vddpad_s1  -type vddpad
add_pad -edge bottom -signal VSSPAD -inst_name u_gndpad_s1  -type vsspad
add_pad -edge bottom -signal VDD    -inst_name u_vddcore_s1 -type vddcore
add_pad -edge bottom -signal VSS    -inst_name u_gndcore_s1 -type vsscore

# Initialize the padring
initialize_padring
```

This is about as much as I can provide you, since the PDK is under NDA sadly.

maliberty, 2022-09-08:
@arlpetergadfort have you done this in your work?

arlpetergadfort, 2022-09-08:
@maliberty yes. You can look at the test in: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/pdn/test/pads_black_parrot.tcl
The setup looks correct, but connecting to the pads is only possible if there are connections specified, so they might need to add a connection from metal2 to metal4/5 if that is where the pins are located on the pads.

Syndace, 2022-09-08:
Thanks, that's what I was missing. I added connections from Metal4/Metal5 to Metal7, which is where the power pads have their pins, and at least the horizontal power pads are now connected to the core power rings.

The vertical power pads are not connected though, which I guess has to do with Metal7 being a horizontal routing layer. How does that work for black parrot? It looks like the power pads have their pins on metal4. Does pdngen route "above" the pins on metal5 and then via down to them?

Syndace, 2022-09-09:
I've talked to a professional and it seems commercial tools like Cadence ignore the routing direction of layers during power network generation. This might be something to consider for `-connect_to_pads`.

arlpetergadfort, 2022-09-13:
@Syndace I think it would be relatively easy to implement something like that if it would be helpful.

Syndace, 2022-09-14:
It would be very helpful, for now I've connected the vertical pads by hand, which isn't too much fun :D I'll rename the issue to reflect it's new goal.

arlpetergadfort, 2022-09-20:
@Syndace this PR (https://github.com/The-OpenROAD-Project/OpenROAD/pull/2284) should solve the issue of not connecting on some edges.

closed 2022-09-20 by arlpetergadfort