The-OpenROAD-Project/OpenROAD #2253: padframe: rotating the corners moves them outside of the chip area

Syndace, 2022-09-08:
When I define out corner padcell using

```TCL
define_pad_cell \
    -name corner \
    -corner \
    -cell_name corner \
    -orient { ll R0 lr R90 ur R180 ul R270 } \
    -physical_only
```

it's in the correct position but rotated incorrectly. When I try to adjust the rotation by 180° as follows:

```TCL
define_pad_cell \
    -name corner \
    -corner \
    -cell_name corner \
    -orient { ll R180 lr R270 ur R0 ul R90 } \
    -physical_only
```

the cell is rotated correctly, but positioned outside of the chip area:

[image]

I guess the origin of the cell needs to be adjusted/is adjusted incorrectly after rotation.

maliberty, 2022-09-08:
Can you provide a test case or is this under NDA?

Syndace, 2022-09-08:
NDA, but I'll try to build an open-source test case. Will report back when I'm done.

Syndace, 2022-09-09:
Okay, this can be trivially reproduced with one of the tests. Change into `src/pad/test`, modify line 99 of `tcl_interface.tcl`:

```
@@ -96,7 +96,7 @@ define_pad_cell \
 define_pad_cell \
   -name PAD_CORNER \
   -corner \
-  -orient {ll R0 lr R90 ur R180 ul R270} \
+  -orient {ll R90 lr R180 ur R270 ul R0} \
   -physical_only
 
 define_pad_cell \
```

and run `openroad -gui tcl_interface.tcl`. The GUI will open, run the test and show the following:

[image]

I know this isn't really a "test case", just instructions for reproducing. Is that enough for you or do you want me to prepare something else?

maliberty, 2022-09-09:
I think that is enough; I'll give it a try.
@Colin-Holehouse any ideas?

maliberty, 2022-09-20:
Fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/2272

closed 2022-09-20 by maliberty

Syndace, 2022-11-23:
Can confirm this is fixed, thanks!