The-OpenROAD-Project/OpenROAD #2260: python API for global router

joamatab, 2022-09-10:
How could we expose OpenRoad's global router with a python API?

https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/grt/README.md 

https://github.com/gdsfactory/gdsfactory/issues/57

maliberty, 2022-09-10:
We are slowing proceeding with a python API for OR.  What is your use model that can't be served with tcl?

joamatab, 2022-09-10:
the use model is to support an analog/photonics flow where you can write PCells in python
See [gdsfatory docs](https://gdsfactory.github.io/gdsfactory/README.html)

A python API for OpenRoad will increase many other use cases as well and lower the entry barrier for new users

maliberty, 2022-09-10:
The value of a python api is clear and work has started as I said.  I'm just asking if this is blocking for you in some way.

maliberty, 2022-09-10:
We could prioritize global routing next in that process.

rovinski, 2022-09-10:
@joamatab I looked at your use case and I think you might want to consider a different approach.

FastRoute (the global router) generates a list of "routing guides" based on the goal of avoiding routing congestion. It is essentially a maze solver which operates at the granularity of a GCell (where GCell = a box of ~10 M3 tracks).

TritonRoute (the detailed router) uses these routing guides as inputs and generates actual metal routes which obey DRC rules (as best as possible).

The two are meant to work together to create routing solutions. It is possible to use just FastRoute, but then you would have to alter your routing tool to use FastRoute's route guides as an input.

On the other hand, I don't think it's known how well FastRoute/TritonRoute handle bus-like routes as shown in https://github.com/gdsfactory/gdsfactory/issues/57#issue-920645244. We don't have many (any?) test designs like this since OpenROAD is mainly a digital integrated circuit tool.

> minimize number of waveguide crossings
> keep path length match between groups of routes

The tools don't have any concept of these goals. The current cost functions generally optimize for 1) DRC avoidance 2) Layer-change avoidance 3) wire length minimization. It may inadvertently meet your goals if the solutions line up.

Also I looked at https://github.com/gdsfactory/gdsfactory a little bit and I don't fully understand the types of designs it's meant to be used for, but I saw (I think) routing solutions that have curves in them. Our tools definitely don't support this and have no plan to. We operate exclusively with manhattan-style routing.

maliberty, 2022-09-11:
Custom routing is quite a different thing from P&R routing.  It is unlikely that pcells are going to work the way you want

proppy, 2022-09-16:
related #1812 #1849

closed 2025-08-28 by maliberty