The-OpenROAD-Project/OpenROAD #2268: report_clock_skew_metric fails with float OverflowError

antonblanchard, 2022-09-14:
I starting hitting this after an update to ORFS - commit 90afa768f292 ("metrics: add clock skew report"). The attached test case fails with:

```
Error: floorplan.tcl, 101 OverflowError in method 'metric_float', argument 2 of type 'float'
```

Clock skew metrics before CTS seem of little use, but we probably shouldn't fall in a heap. I added:

```
puts [worst_clock_skew -setup]
puts [worst_clock_skew -hold]
```

and I get:

```
1.0000000190432691e+42
1.0000000190432691e+42
```

That's a lot of skew.

Test case: [floorplan_multiply_adder_asap7_base_2022-09-14_14-55.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9562953/floorplan_multiply_adder_asap7_base_2022-09-14_14-55.tar.gz)

maliberty, 2022-09-14:
@vvbandeira We should move clock_skew inside the include_erc block of report_metrics.

rovinski, 2022-09-23:
I am running into this as well. It's preventing me from running very simple designs for the upcoming tutorial.

maliberty, 2022-09-23:
I'll fix the flow to not report clock skew before cts is run.  I'll leave this for @jjcherry56 to give a more reasonable result if someone does run it manually.

jjcherry56, 2022-09-23:
It doesn't have anything to do with running cts and moving report_clock_skew will not help.
The issue is there are no capture registers to measure skew to, just output pins, which the command reports:
% report_clock_skew
Clock clk
No launch/capture paths found.

I updated opensta to return 0 in this case but did not update openroad.

jjcherry56, 2022-09-23:
@rovinski it looks like you don't have any clock constraints at all.

closed 2022-09-26 by tspyrou

rovinski, 2022-09-27:
@jjcherry56 My case was actually that there was a clock constraint, but there were no input/output constraints and no reg->reg paths, only in->reg and reg->out. https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/594 fixes `Error: floorplan.tcl, 101 OverflowError in method 'metric_float', argument 2 of type 'float'` and https://github.com/The-OpenROAD-Project/OpenSTA/commit/18f11a302958ea6814cd1d711a18ec3eab3c1184 fixes `[ERROR STA-0414] period '-9.500000411629289e+38' is not a positive float.`. Adding input/output constraints also fixes the second error.