The-OpenROAD-Project/OpenROAD #2270: STA output shows incorrect fan out and connectivity

antonblanchard, 2022-09-14:
I was looking through STA output for one of my designs, and noticed a few places where the fan out looks incorrect, and also some places where the connectivity appears wrong:

```
Startpoint: a[49] (input port clocked by clk)
Endpoint: _573_ (falling edge-triggered flip-flop clocked by clk')
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (propagated)
                         54.00   54.00 ^ input external delay
                  0.00    0.00   54.00 ^ a[49] (in)
     1    1.12                           a[49] (net)
                  0.40    0.13   54.13 ^ input44/A (BUFx2_ASAP7_75t_R)
                 13.46   16.81   70.93 ^ input44/Y (BUFx2_ASAP7_75t_R)
     1    2.60                           net44 (net)
                 13.48    0.28   71.21 ^ fanout925/A (BUFx12f_ASAP7_75t_R)
                 20.71   20.94   92.15 ^ fanout925/Y (BUFx12f_ASAP7_75t_R)
     8   20.50                           net925 (net)
                 27.95    6.33   98.48 ^ fanout922/A (BUFx16f_ASAP7_75t_R)
                 25.50   28.49  126.97 ^ fanout922/Y (BUFx16f_ASAP7_75t_R)
    16   35.34                           net922 (net)
                 29.62    4.91  131.88 ^ U$$3266/A (INVx1_ASAP7_75t_R)
                 10.49   10.30  142.17 v U$$3266/Y (INVx1_ASAP7_75t_R)
     1    0.58                           notblock$10211[2] (net)
                 10.49    0.04  142.21 v U$$3267/A1 (AO33x2_ASAP7_75t_R)
                 35.10   51.16  193.37 v U$$3267/Y (AO33x2_ASAP7_75t_R)
     2    7.21                           sel_0$10212 (net)
                 36.21    3.39  196.77 v fanout443/A (BUFx12f_ASAP7_75t_R)
                 21.00   29.01  225.78 v fanout443/Y (BUFx12f_ASAP7_75t_R)
    11   21.92                           net443 (net)
                 21.05    0.65  226.43 v fanout442/A (BUFx12f_ASAP7_75t_R)
                 17.43   21.67  248.10 v fanout442/Y (BUFx12f_ASAP7_75t_R)
    16   21.59                           net442 (net)
                 65.81   20.36  268.45 v U$$3283/A2 (AO22x1_ASAP7_75t_R)
                 17.70   41.38  309.84 v U$$3283/Y (AO22x1_ASAP7_75t_R)
     1    1.67                           t$10221 (net)
                 17.70    0.17  310.01 v U$$3284/A (XOR2x1_ASAP7_75t_R)
                 27.76   28.76  338.77 v U$$3284/Y (XOR2x1_ASAP7_75t_R)
     1    2.13                           booth_b48_m7 (net)
                 27.76    0.20  338.97 v dadda_fa_1_55_6/B (FAx1_ASAP7_75t_R)
                 33.87   26.06  365.02 ^ dadda_fa_1_55_6/CON (FAx1_ASAP7_75t_R)
     8    0.70                           con$451 (net)                         <-- fan out is actually 1
                 33.87    0.03  365.06 ^ U$$4938/A (INVx1_ASAP7_75t_R)
                 21.01   18.20  383.26 v U$$4938/Y (INVx1_ASAP7_75t_R)
     1    2.21                           c$1810 (net)
                 21.01    0.17  383.43 v dadda_fa_2_56_2/B (FAx1_ASAP7_75t_R)
                 96.37   54.65  438.08 ^ dadda_fa_2_56_2/CON (FAx1_ASAP7_75t_R) 
     8    4.62                           con$1807 (net)                         <-- fan out is actually 1
                 80.90   41.46  479.54 v dadda_fa_2_56_2/SN (FAx1_ASAP7_75t_R)  <-- SN is not connected to CON above
     1    2.78                           sn$1808 (net)
                 80.91    0.67  480.20 v U$$5891/A (INVx1_ASAP7_75t_R)
                 35.85   30.41  510.61 ^ U$$5891/Y (INVx1_ASAP7_75t_R)
     1    1.95                           s$3617 (net)
                 35.85    0.17  510.78 ^ dadda_fa_3_56_3/A (FAx1_ASAP7_75t_R)
                 27.05   44.35  555.13 v dadda_fa_3_56_3/SN (FAx1_ASAP7_75t_R)
     1    0.80                           sn$3616 (net)
                 27.05    0.07  555.20 v U$$6781/A (INVx1_ASAP7_75t_R)
                 20.39   17.46  572.66 ^ U$$6781/Y (INVx1_ASAP7_75t_R)
     1    1.83                           s$5266 (net)
                 20.40    0.18  572.84 ^ dadda_fa_4_56_2/CI (FAx1_ASAP7_75t_R)
                 31.35   35.50  608.34 ^ dadda_fa_4_56_2/SN (FAx1_ASAP7_75t_R)
     1    0.70                           sn$5263 (net)
                 31.35    0.02  608.36 ^ U$$7519/A (INVx1_ASAP7_75t_R)
                 15.84   14.60  622.96 v U$$7519/Y (INVx1_ASAP7_75t_R)
     1    1.53                           s$6650 (net)
                 15.84    0.07  623.03 v dadda_fa_5_56_1/CI (FAx1_ASAP7_75t_R)
                 31.36   19.88  642.91 ^ dadda_fa_5_56_1/CON (FAx1_ASAP7_75t_R)
     8    0.65                           con$6646 (net)                         <-- fan out is actually 1
                 31.36    0.03  642.94 ^ U$$8110/A (INVx1_ASAP7_75t_R)
                 19.73   17.21  660.15 v U$$8110/Y (INVx1_ASAP7_75t_R)
     1    2.12                           c$7580 (net)
                 19.73    0.17  660.32 v dadda_fa_6_57_0/B (FAx1_ASAP7_75t_R)
                 40.01   28.01  688.33 ^ dadda_fa_6_57_0/CON (FAx1_ASAP7_75t_R)
     8    1.40                           con$7577 (net)                         <-- fan out is actually 1
                 24.07   17.02  705.34 v dadda_fa_6_57_0/SN (FAx1_ASAP7_75t_R)  <-- SN is not connected to CON above
     1    0.70                           sn$7578 (net)
                 24.07    0.04  705.39 v U$$8499/A (INVx1_ASAP7_75t_R)
                 17.15   15.10  720.49 ^ U$$8499/Y (INVx1_ASAP7_75t_R)
     1    1.52                           s$8206 (net)
                 17.15    0.07  720.56 ^ dadda_fa_7_57_0/CI (FAx1_ASAP7_75t_R)
                 32.61   35.35  755.91 ^ dadda_fa_7_57_0/SN (FAx1_ASAP7_75t_R)
     1    0.75                           sn$8203 (net)
                 32.61    0.05  755.96 ^ U$$8755/A (INVx1_ASAP7_75t_R)
                 11.71   11.29  767.25 v U$$8755/Y (INVx1_ASAP7_75t_R)
     1    0.65                           s$10947 (net)
                 11.71    0.04  767.30 v _573_/D (DFFLQNx2_ASAP7_75t_R)
                                767.30   data arrival time

                        540.00  540.00   clock clk' (fall edge)
                          0.00  540.00   clock source latency
                  0.00    0.00  540.00 ^ clk (in)
     1    6.22                           clk (net)
                  7.69    2.43  542.43 ^ clkbuf_0_clk/A (BUFx8_ASAP7_75t_R)
                 12.15   22.56  564.99 ^ clkbuf_0_clk/Y (BUFx8_ASAP7_75t_R)
     2    4.35                           clknet_0_clk (net)
                 12.31    0.67  565.65 ^ clkbuf_1_0__f_clk/A (BUFx8_ASAP7_75t_R)
                 25.03   29.67  595.32 ^ clkbuf_1_0__f_clk/Y (BUFx8_ASAP7_75t_R)
     4   14.38                           clknet_1_0__leaf_clk (net)
                 26.55    3.28  598.60 ^ clkbuf_leaf_8_clk/A (BUFx8_ASAP7_75t_R)
                 42.01   40.27  638.87 ^ clkbuf_leaf_8_clk/Y (BUFx8_ASAP7_75t_R)
    16   27.61                           clknet_leaf_8_clk (net)
                 42.99    3.62  642.49 ^ _186_199/A (INVx3_ASAP7_75t_R)
                  9.36    8.42  650.91 v _186_199/Y (INVx3_ASAP7_75t_R)
     1    0.39                           net1528 (net)
                  9.36    0.03  650.94 v _573_/CLK (DFFLQNx2_ASAP7_75t_R)
                          0.00  650.94   clock reconvergence pessimism
                        -11.71  639.23   library setup time
                                639.23   data required time
-----------------------------------------------------------------------------
                                639.23   data required time
                               -767.30   data arrival time
-----------------------------------------------------------------------------
                               -128.07   slack (VIOLATED)
```

Test case: [sta-issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9565716/sta-issue.tar.gz)

jjcherry56, 2022-09-18:
You are going to have to be more specific about exactly what number in what line appears to be incorrect and why you think it is incorrect.

maliberty, 2022-09-18:
The path is annotated with exactly what is wrong

jjcherry56, 2022-09-19:
The issue is really a broken lib file.
Some of the cells have timing arcs between output pins.
What you are seeing is not incorrect connectivity but timing arcs from input to output to output on the same instance,
like  dadda_fa_6_57_0/B -> CON -> SN. The liberty should not have the output to output timing arcs.
They should be removed.

Pending OR PR 2279 (which may take a while to clear CI) fixes the fanout reporting to tolerate this situation and
generate the warnings shown below.

[WARNING STA-0164] asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib line 13177, timing group from output port.
[WARNING STA-0164] asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib line 13210, timing group from output port.
[WARNING STA-0164] asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib line 13243, timing group from output port.
[WARNING STA-0164] asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib line 13276, timing group from output port.
[WARNING STA-0164] asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib line 13309, timing group from output port.
[WARNING STA-0164] asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib line 13342, timing group from output port.
[WARNING STA-0164] asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib line 13375, timing group from output port.
[WARNING STA-0164] asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib line 14771, timing group from output port.
[WARNING STA-0164] asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib line 14804, timing group from output port.
[WARNING STA-0164] asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib line 14837, timing group from output port.

antonblanchard, 2022-09-19:
Thanks @jjcherry56 I submitted https://github.com/The-OpenROAD-Project/asap7/issues/38 to track the ASAP7 issue. I tested the fix, and it resolves all my issues. Thanks!

jjcherry56, 2022-09-19:
dc11f5970 fixes the fanout reporting and adds warnings for output to output timing arcs. but looks like you have to live with the goofy library

closed 2022-09-19 by jjcherry56