The-OpenROAD-Project/OpenROAD #2275: Use standard naming for variables (ie snake case) in LEF58 parsers

osamahammad21, 2022-09-15:
Use standard naming for variables (ie snake case). 

_Originally posted by @maliberty in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2267#discussion_r971275243_

vijayank88, 2023-06-22:
@osamahammad21 
Any update on this?

osamahammad21, 2023-06-25:
@vijayank88 This is a low priority issue at the moment. It's not done yet. It would be a good first issue for new comers.

titan73, 2024-08-28:
Do you mean camel case?
https://github.com/The-OpenROAD-Project/OpenROAD/pull/2267#discussion_r972247038 says EXCEPTEDGELENGTH should be translated exceptEdgeLength but it's camel case not snake case.
snake case would give except_edge_length.

osamahammad21, 2024-08-28:
Yeah sorry for misleading but we should use snake case not camel case:
except_edge_....

maliberty, 2024-08-28:
We roughly intend to follow the Google style (https://google.github.io/styleguide/cppguide.html#Type_Names) though have lots of code that doesn't.

titan73, 2024-08-28:
Ah ok:
type name => camel case
variable => snake case
constant => camel case mac style
Pretty mixed style ;)

Dhirenderchoudhary, 2026-03-10:
@maliberty can i work on this ?