The-OpenROAD-Project/OpenROAD #2276: GUI Unexpectedly Exits On Error

QuantamHD, 2022-09-15:
This script

```tcl
set clock_ports {}
foreach clock [all_clocks] {
    foreach pin [$clock sources] {
        lappend clock_ports [$pin port] 
    }
}

foreach input [all_inputs] {
    set is_clock 0
    foreach clock_port $clock_ports {
        if {[$clock_port bus_name] == [$input bus_name]} {
            set is_clock 1
        }
    }
    if $is_clock == 1{
        puts $is_clock
    }
}
```

Will cause openROAD GUI to immediately exit when run after reading in a design

arlpetergadfort, 2022-09-16:
@QuantamHD is this design dependent? I don't get this behavior.
But I do get:
```
>>> set clock_ports {}
... foreach clock [all_clocks] {
...     foreach pin [$clock sources] {
...         lappend clock_ports [$pin port] 
...     }
... }
... 
... foreach input [all_inputs] {
...     set is_clock 0
...     foreach clock_port $clock_ports {
...         if {[$clock_port bus_name] == [$input bus_name]} {
...             set is_clock 1
...         }
...     }
...     if $is_clock == 1{
...         puts $is_clock
...     }
... }
[ERROR GUI-0070] invalid command name "=="
```

QuantamHD, 2022-09-17:
I haven't tried it with other designs. I did have a database loaded at the time.

arlpetergadfort, 2022-09-18:
@QuantamHD could you upload a test case that replicates it for you? I can't seem to get it on my end.

QuantamHD, 2022-09-18:
Let me see what I can do.

jjcherry56, 2022-09-20:
This is not valid tcl:

    if $is_clock == 1{

it should be if { $is_clock == 1 }  {

maliberty, 2022-09-20:
It is hard to imagine how this tcl code would have any effect on the GUI

QuantamHD, 2022-09-20:
It's an uncaught syntax error from the looks of it. The code above has syntax errors in it, but those errors appear to not be caught

QuantamHD, 2022-09-20:
However, only I seem to be able to reproduce it

arlpetergadfort, 2022-09-20:
@QuantamHD if there is a stack trace that might be helpful.

arlpetergadfort, 2022-09-20:
@QuantamHD I'm able to get a stack trace if I add your code to a script at startup and run openroad with `-gui` `-exit`, are you adding the -exit? this should be fixed either way

arlpetergadfort, 2022-09-20:
@QuantamHD  does this prevent the issue you are seeing? https://github.com/The-OpenROAD-Project/OpenROAD/pull/2285

maliberty, 2022-12-15:
@QuantamHD still relevant?

QuantamHD, 2022-12-15:
I'll try again.

vijayank88, 2023-06-19:
@QuantamHD 
Waiting for your update to resolve it.

closed 2025-08-30 by maliberty