The-OpenROAD-Project/OpenROAD #2277: grt makeRouteParasitics() misses some vias

antonblanchard, 2022-09-16:
I'm looking at differences between grt parasitic resistance estimates and the final spef parasitics. An example in the attached test case is net `_065_`. `write_net_rc_script.tcl` creates a CSV file with this information:

```
_065_,2.279e+00,1.214e-17,4.689e+00,2.605e-17,3.707e+01,1.200e-17,M2,0.194,M3,0.0,M4,0.0,M5,0.0,M6,0.0,M7,0.0
```

Specifically:

```
grt:   4.689 Ohms
spef: 37.07 Ohms
```

The global routing guide looks like:

```
_065_
(
23760 10260 24300 10800 M1
23760 10260 24300 10800 M2
)
```

And the routed def matches this:
```
    - _065_ ( _690_ QN ) ( _326_ A ) + USE SIGNAL
      + ROUTED M2 ( 24093 10575 ) ( 24165 * )
      NEW M1 ( 24165 10575 ) VIA12
      NEW M1 ( 24093 10575 ) VIA12
      NEW M2 ( 24165 10575 ) RECT ( 0 -9 62 9 )  ;
```

So we have `VIA12 - M2 - VIA12`

If I enable some debugging:

```
set_debug_level GRT est_rc 10
```

```
[DEBUG GRT-0001] net _065_
(24.03, 10.53)  1 -> (24.03, 10.53)  2 l=0.00
[DEBUG GRT-0001] _065_:0 -> _065_:1 via 1-2 r=0.017
[DEBUG GRT-0001] _065_:1 -> _690_/QN (24.02, 10.56) 0.04u layer=2 r=0.001 c=0.006
[DEBUG GRT-0001] pin _690_/QN -> to grid 0u layer=2 r=0.001 c=0.006
[DEBUG GRT-0001] _065_:1 -> _326_/A (24.16, 10.56) 0.15u layer=2 r=0.004 c=0.021
[DEBUG GRT-0001] pin _326_/A -> to grid 0u layer=2 r=0.004 c=0.021
```

I only see one via (perhaps because this is a very short net, and both vias are in the same global route grid). On top of that issue, we seem to decide that the via is not in the path (perhaps because our min layer is defined as M2, but we still need to access standard cells with M1-M2 vias below it).

Enabling some debug in STA confirms that we ignored all the vias and ended up with 4.69 Ohms:

```
ReduceToPi::reduceToPi
 Pi model c2=1.08e-17 rpi=4.69 c1=1.97e-15 max_r=18.2
```

Test case: [rc-correlate.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9580760/rc-correlate.tar.gz)

maliberty, 2022-10-03:
@luis201420 any update on this?

antonblanchard, 2022-10-12:
This helps, however I have no idea if the pin access vias are meant to be accounted for in  `makeParasiticsToPin` or `makeRouteParasitics`. The second pin via is still missing:

```
diff --git a/src/grt/src/MakeWireParasitics.cpp b/src/grt/src/MakeWireParasitics.cpp
index 16429231c..19dc18b76 100644
--- a/src/grt/src/MakeWireParasitics.cpp
+++ b/src/grt/src/MakeWireParasitics.cpp
@@ -174,25 +174,25 @@ void MakeWireParasitics::makeParasiticsToPins(std::vector<Pin>& pins)
 void MakeWireParasitics::makeParasiticsToPin(Pin& pin)
 {
   sta::Pin* sta_pin = staPin(pin);
   sta::ParasiticNode* pin_node = parasitics_->ensureParasiticNode(parasitic_, sta_pin);
   const odb::Point& grid_pt = pin.getOnGridPosition();
   // Use the route layer above the pin layer if there is a via
   // to the pin.
   int layer = pin.getConnectionLayer() + 1;
   RoutePt grid_route(grid_pt.getX(), grid_pt.getY(), layer);
   sta::ParasiticNode* grid_node = node_map_[grid_route];
   
   // Use the pin layer for the connection.
-  if (grid_node == nullptr) {
+  if (1 || grid_node == nullptr) {
     layer--;
     grid_route = RoutePt(grid_pt.getX(), grid_pt.getY(), layer);
     grid_node = node_map_[grid_route];
   }
 
   if (grid_node) {
     // Make wire from pin to gcell center on pin layer.
     const odb::Point& pt = pin.getPosition();
     int wire_length_dbu
       = abs(pt.getX() - grid_pt.getX()) + abs(pt.getY() - grid_pt.getY());
     float res, cap;
     layerRC(wire_length_dbu, layer, res, cap);
```

```
[DEBUG GRT-0001] net _065_
(24.03, 10.53)  1 -> (24.03, 10.53)  2 l=0.00
[DEBUG GRT-0001] _065_:0 -> _065_:1 via 1-2 r=0.017
[DEBUG GRT-0001] _065_:0 -> _690_/QN (24.02, 10.56) 0.04u layer=1 r=0.006 c=0.005
[DEBUG GRT-0001] pin _690_/QN -> to grid 0u layer=1 r=0.006 c=0.005
[DEBUG GRT-0001] _065_:0 -> _326_/A (24.16, 10.56) 0.15u layer=1 r=0.021 c=0.017
[DEBUG GRT-0001] pin _326_/A -> to grid 0u layer=1 r=0.021 c=0.017
 Pi model c2=9.17e-18 rpi=26.9 c1=1.97e-15 max_r=26.9
```

maliberty, 2022-10-12:
I think you are pursuing a phantom problem.  In grt we just assume the cells are at the gcell center as we don't know the real routing topology below that granularity.  You can see the resistor is added in grt
```
[DEBUG GRT-0001] _065_:0 -> _065_:1 via 1-2 r=0.017
```
it just gets discarded by rc reduction as it doesn't lead to anything since the net is in a single gcell.  We could do something special for a net entirely in one gcell but it still wouldn't solve the problem of a net that has multiple sinks in one gcell while spanning more than one gcell.

What is the root problem you are after?

antonblanchard, 2022-10-12:
> I think you are pursuing a phantom problem

I might be. I'm trying to reduce the differences I see between STA post global routing and STA post detailed routing. I understand some differences are unavoidable, but when I went looking at R and C differences of individual nets in one of my designs, this issue came up many times.

Maybe it isn't a big deal in the scheme of things, but we are also using these nets to do our linear regression to create RC layer estimates. In the above example, the post global route net is  4.69 Ohms and the post detailed route net is 37.07 Ohms (most of that difference being 2 x 17.2 Ohm pin access vias).

maliberty, 2022-10-12:
The PR above is to address you concerns.  It will have to go through a full testing cycle but feel free to give it a try.  I now see:
```
[DEBUG GRT-0001] net _065_
(24.03, 10.53)  1 -> (24.03, 10.53)  2 l=0.00
[DEBUG GRT-0001] _065_:0 -> _690_/QN (24.02, 10.56) 0.04u layer=2 r=0.001 via_res=0.017 c=0.006
[DEBUG GRT-0001] pin _690_/QN -> to grid 0u layer=2 r=0.001 via_res=0.017 c=0.006
[DEBUG GRT-0001] _065_:0 -> _326_/A (24.16, 10.56) 0.15u layer=2 r=0.004 via_res=0.017 c=0.021
[DEBUG GRT-0001] pin _326_/A -> to grid 0u layer=2 r=0.004 via_res=0.017 c=0.021
```

maliberty, 2022-10-13:
Some issues in testing need a bit of time to validate

antonblanchard, 2022-10-13:
Thanks @maliberty I'm running some tests to see if it has improved things.

PR #2362 (2022-10-16): grt: include via to pin in estimating parasitics
Fixes #2277
commit 5ad890de (2022-10-16): grt: include via to pin in estimating parasitics

Fixes #2277
closed 2022-10-16 by maliberty