The-OpenROAD-Project/OpenROAD #2283: [WARNING STA-0201] should be error condition

openpowerwtf, 2022-09-20:
Macro I/O mismatches are detected early, but do not cause failure until LVS checking.  Seems like this is almost always a setup error.

Related: are there plans to allow log messages to be altered by tcl (info/warning/error, possibly custom text, etc.)?

From intial floorplan log (Verilog macro wrapper does not match actual macro):

```
[WARNING STA-0201] /data/projects/toy-sram-mpw7/openlane/toysram/runs/22_09_19_16_21/results/synthesis/toysram_site.v line 4666, instance \ra_0.genblk1.ra  port rd_dat_0 not found.
```

maliberty, 2022-09-20:
We did debate about user controls for error messages.  We would never allow a downgrade (ie error to warning) but could consider allowing an upgrade.  I have never seen custom text - how would that be used?

vijayank88, 2023-06-19:
@maliberty 
Is that required further support?

closed 2023-06-19 by maliberty