The-OpenROAD-Project/OpenROAD #2287: OpenROAD hangs forever when executing python scripts

proppy, 2022-09-21:
With #1812 (https://github.com/The-OpenROAD-Project/OpenROAD/commit/47e8b5f57e2f680172f99c0433c01ed91fcf7fe7), OpenROAD started hanging forever when executing external python scripts, see:
https://colab.research.google.com/gist/proppy/5a80fbc91ea8520839e0779126d96240/openroad-hanging-python-issue.ipynb

```
🧁 cat << EOF > foo.py
print('hello from external script')
EOF
🍉 openroad -python foo.py
OpenROAD 47e8b5f57e2f680172f99c0433c01ed91fcf7fe7 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING ORD-0039] .openroad ignored with -python
hello from external script
Python 3.7.13 (default, Mar 29 2022, 02:21:02) 
[GCC 7.5.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

Previous version like https://github.com/The-OpenROAD-Project/OpenROAD/commit/dc11f5970c23ff81770fcbf705737516dad90b23 are not affected:
```
🍙 openroad -python foo.py
OpenROAD dc11f5970c23ff81770fcbf705737516dad90b23 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
hello from external script
```

proppy, 2022-09-21:
/cc @QuantamHD

maliberty, 2022-09-21:
This is the same as with tcl.  You need to give -exit if you want it to end after the script otherwise it goes to the interactive prompt.

PR #2288 (2022-09-22): src: fix PyMain arguments forwarding
This copy all the arguments, instead of just forwarding `argv[0]`, reusing the same code that was previously in https://github.com/The-OpenROAD-Project/OpenROAD/commit/47e8b5f57e2f680172f99c0433c01ed91fcf7fe7#diff-7c2e2869b9670e340a14a2b2ed1cc45a9f85fb8b219a4c1f06fdd467f49375c1L216-L222

Fixes: #2287
closed 2022-09-21 by maliberty