The-OpenROAD-Project/OpenROAD #2308: package swig binding as shared libraries

proppy, 2022-09-27:
In order to be able to import them from `.py` script, the swig binding needs to be built as shared libraries.

Currently the `swig_library` helper forces static linking, see:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/cmake/swig_lib.cmake#L74

@maliberty is the `.a` actually needed by something else in the build? do we need to build both or can we get away with only building the `.so`?

/cc @QuantamHD

QuantamHD, 2022-09-27:
@proppy I have a demo working for one of them

maliberty, 2022-09-27:
The .a are used to build a statically linked binary which is much simpler for users as it doesn't require setting up an LD_LIBRARY_PATH entry.

proppy, 2022-09-27:
@maliberty understood, but curious if there a usecase for having the swig bindings `openroad_swig_py.a` as a static library? Is it meant for users that'd like to link and extend them?

maliberty, 2022-09-27:
openroad -python needs the swig bindings and is statically linked.

QuantamHD, 2022-09-27:
@proppy We need to set up the infrastructure to build and package the so's for pip.

QuantamHD, 2022-09-27:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/2311

maliberty, 2022-09-27:
@macd FYI

macd, 2022-09-27:
It would be nice to be able to import into an unmodified Python. I will take a look at @QuantamHD changes (I have PoC wrappers for about 10 of the tools/libs at present). How many versions of Python do you envision building against?