The-OpenROAD-Project/OpenROAD #2314: ppl: createMatrixForGroups accesses past the end of slots_

antonblanchard, 2022-09-28:
The attached test case fails with:

```
/usr/include/c++/12/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = ppl::Slot; _Alloc = std::allocator<ppl::Slot>; reference = ppl::Slot&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Stack trace:
 0# 0x0000000000C2540E in openroad
 1# 0x00007F75DF63EA70 in /lib64/libc.so.6
 2# 0x00007F75DF68EC4C in /lib64/libc.so.6
 3# raise in /lib64/libc.so.6
 4# abort in /lib64/libc.so.6
 5# __gnu_debug::_Safe_sequence_base::_M_get_mutex() in /lib64/libstdc++.so.6
 6# ppl::HungarianMatching::createMatrixForGroups() in openroad
 7# ppl::HungarianMatching::findAssignmentForGroups() in openroad
 8# ppl::IOPlacer::findPinAssignment(std::vector<ppl::Section, std::allocator<ppl::Section> >&) in openroad
 9# ppl::IOPlacer::run(bool) in openroad
10# 0x0000000000D830D1 in openroad
```

Test case: [io_placement_multiplier_asap7_base_2022-09-28_14-32.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9661556/io_placement_multiplier_asap7_base_2022-09-28_14-32.tar.gz)

antonblanchard, 2022-09-28:
It reproduced on the latest sha1 (c7ce441ce364). I wonder if your libstdc++ doesn't have the check. Does this patch trigger it?

[ppl.patch.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9668659/ppl.patch.gz)

maliberty, 2022-09-28:
@antonblanchard are you compiling debug?  The checks can differ with optimization level.

arlpetergadfort, 2022-09-28:
@maliberty this the same issue we kept running into as well with ppl.

QuantamHD, 2022-09-28:
Just looking at the code it seems like there's an outer loop iterator somewhere while the code in ppl::HungarianMatching::createMatrixForGroups is modifying that vector, which is not allowed.

eder-matheus, 2022-09-28:
@antonblanchard Could you check this PR: https://github.com/The-OpenROAD-Project/OpenROAD/pull/2319?

antonblanchard, 2022-09-29:
> @antonblanchard Could you check this PR: #2319?

@eder-matheus Thanks, this fixes the issue I was hitting

eder-matheus, 2022-09-29:
> > @antonblanchard Could you check this PR: #2319?
> 
> 
> 
> @eder-matheus Thanks, this fixes the issue I was hitting
> 
> 

Great! I've just merged the fix with master.

closed 2022-09-29 by eder-matheus